mobizt/Firebase-ESP8266 #1: Not able to set/update values to firebase when Streaming

chander-60905, 2019-03-20:
Everything works fine unless two things. i am able to update single node using one child but not using two or more childs as it returns as string not json. String updateData = "{\"firmware_version\":\"1.0.0\", \"device_online\":\"yes\"}"; and the second one is stream is working fine but i am not able to set/update string or int to firebase when streaming as esp8266 crashes.

mobizt, 2019-03-20:
Can you make the example sketch that causes the error or share some part of your sketch? This library version is port from ESP32 which may required some fixes to prevent crash.

chander-60905, 2019-03-20:
ok i will post an example tomorrow morning and thank you for quick reply.

chander-60905, 2019-03-21:
Please check every test case for 10-15 minutes to see if it crashes or not in NodeMcu esp8266 with 2.5.0 and noSpiffs.

//ino file

//FirebaseESP8266.h must be included before ESP8266WiFi.h
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "Firebase Host"
#define FIREBASE_AUTH "Firebase Auth"
#define WIFI_SSID "ssid"
#define WIFI_PASSWORD "password"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;
FirebaseData streamData; //If using single FirebaseData firebaseData; repeats stream loop every 2-3 seconds and using another FirebaseData streamData; works ok but crashes sometimes.

String path = "/LKV8zPLXYGXJ9gN75ecd8fJLwmE2/Device1"; // Setting path as "/User_id/Controller_1" crashes and Setting path as "/Controller_1" gives heap size approx 18800 in stream loop and Setting path as "/Firebase_User_id/Controller_1" gives free heap as 338080 in stream loop and crashes sometimes 

// In last case tested setting FirebaseData streamData; and using path = "/LKV8zPLXYGXJ9gN75ecd8fJLwmE2/Device1", where LKV8zPLXYGXJ9gN75ecd8fJLwmE2 is my firebase user id works with free heap size 338416 but if leaves sketch running for 10-15 minutes and then pinDetails changes then crashes.

void setup() {
  
  Serial.begin(115200);

  pinMode(D1, OUTPUT);
  pinMode(D2, OUTPUT);
  pinMode(D3, OUTPUT);
  pinMode(D5, OUTPUT);
  digitalWrite(D1, HIGH); 
  digitalWrite(D2, HIGH); 
  digitalWrite(D3, HIGH); 
  digitalWrite(D5, HIGH); 
  
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  
    if (Firebase.getInt(firebaseData, "/Controllers" + path + "/D1status"))
    {
      Serial.print("D1 Status: ");
      if (firebaseData.dataType() == "int")
        Serial.println(firebaseData.intData());
        if (firebaseData.intData() == 0)
        {
          digitalWrite(D1, LOW);
        }
        else
        {
          digitalWrite(D1, HIGH);
        }
        Serial.println(ESP.getFreeHeap());
    }
    delay(100);

    if (Firebase.getInt(firebaseData, "/Controllers" + path + "/D2status"))
    {
      Serial.print("D2 Status: ");
      if (firebaseData.dataType() == "int")
[... 131 lines cut ...]
          digitalWrite(D3, HIGH);
          Button_Status = "{\"D3status\":1}";
          Firebase.updateNodeSilent(firebaseData, "/Controllers" + path, Button_Status);
          break;

          case 50:
          digitalWrite(D5, LOW);
          Button_Status = "{\"D4status\":0}";
          Firebase.updateNodeSilent(firebaseData, "/Controllers" + path, Button_Status);
          break;

          case 51:
          digitalWrite(D5, HIGH);
          Button_Status = "{\"D4status\":1}";
          Firebase.updateNodeSilent(firebaseData, "/Controllers" + path, Button_Status);
          break;

          default:
          Button_Status = "";
          break;
      }
    Serial.println(ESP.getFreeHeap());
  }

}

chander-60905, 2019-03-21:
In last case tested after running for 10-15 minutes, Firebase.updateNodeSilent not works but o starting it works.

case 10:
      digitalWrite(D1, LOW);
      Button_Status = "{\"D1status\":0}"; //Sometimes Firebase.updateNodeSilent works in last case tested
      Firebase.updateNodeSilent(firebaseData, "/Controllers" + path, Button_Status);
      break;

mobizt, 2019-03-22:
Thank you I will investigate that and let you known.

mobizt, 2019-03-22:
As you know, esp8266 has too small memory.
I see you're using multiple Firebase Data Object which is not the approach that I wan't to be.
I'm not recommend to use multiple Firebase Data Object because every single object takes 15-17k memory included WiFi client instance which can lead to unhandled exception error.

Two objects may takes 30-40k ram which lead to unreliable situation, any system tasks may fail especially WiFi task.

Please use only one Firebase Data Object or move to ESP32 which has more ram and my library support this features.

mobizt, 2019-03-22:
Ok now I know the cause. 

Due to total header size [at line 368 of FirebaseESP8266.cpp](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.cpp#L368) to send is set to 320 bytes which is too small for wide range of absolute node path in your case.

To fix the error, at line 368 of FirebaseESP8266.cpp change from
```C++
size_t headerSize = 320;
```
to
```C++
size_t headerSize = 500; //600 or any larger value but not too much
```

Hope this helps.

Anyhow, be aware of using multiple Firebase Data Object in case of low free memory.

chander-60905, 2019-03-22:
ok i have tested and it is now working but if i use only one FirebaseData firebaseData; and then in loop if i update any node 

case 10:
      digitalWrite(D1, LOW);
      Button_Status = "{\"D1status\":0}"; //Sometimes Firebase.updateNodeSilent works in last case tested
      Firebase.updateNodeSilent(firebaseData, "/Controllers" + path, Button_Status);
      break;

like this then it will keep repeating stream to that node and it if i use comment out //Firebase.updateNodeSilent(firebaseData, "/Controllers" + path, Button_Status); then it works only when database changes and if i use another FirebaseData streamData; 
then it works only when database changes. can i do this with only one FirebaseData firebaseData; as free heap in this case is 32536 in stream loop and if i use FirebaseData streamData;  free heap size is 18548 only and thank you for the quick fix.

mobizt, 2019-03-22:
I may not clearly understand your post or misunderstand, please correct or re explain me again.

If you use only one Firebase Data object to lower mem usage and want to get stream together with read/update data  at some interval in loop, the stream will stop or disconnected from server after you read/update data because you use the same http or wifi client inside firebase data object to establish new connection.

In this usage way, you need to manage the time for readstream during the read/update cycle as in Stream.ino example.

For example your read/update data every 5 minutes and use the time to read/update data  1 minute, then you have 4 minutes of idle time to allow stream to work before next read/update cycle start again.

This means during 1 minute of read/update data, you will miss any data that being monitor in database that may change at that time.

This is the problem when you have only one communication channel which need to share to several tasks which this also occurred in other Firebase library i.e. developed by FirebaseExtend.

To work around, you may need to manually read json data at path being stream and keep in SD or SPIFS  before you read/update data at other database paths and later compare to the stream data that read from readStream when streamAvailable to determine any changes take place during you read/update data.

I personally move to ESP32 even time for establish new http connection is slower than ESP8266 but I use update function to save the time.

chander-60905, 2019-03-22:
ok so i think i am good at using two firebase data object as i have sufficient free heap memory 16928 after all the operations are done. 

As if now i was testing the whole sketch on Mobile internet as hotspot, it is working all ok. but today i used my broadband router wifi hotspot and esp connected and got ip address but neither firebase nor stream is working. can i know why

chander-60905, 2019-03-22:
i am getting errors like 
Read stream...
FAILED
REASON: connection lost

Read stream...
FAILED
REASON: connection refused

but i can use internet using router wifi hotspot on my laptop and mobile phones.

mobizt, 2019-03-22:
connection lost means WiFi connection is lost
connection refused means WiFi connected but no data return from that Access point or router.
All above means esp8266 could not connect to internet.

Try the following.

Reset your router or turn off (unplug) for 3 minutes and try again.
Make sure your router doesn't block port 443.
Try to setup soft AP on your computer and let esp8266 connect to that soft AP.
Try other routers.

chander-60905, 2019-03-22:
ok i will try an let you know but as for i tested Ping is working on router wifi using esp.

#include <ESP8266WiFi.h>
#include <ESP8266Ping.h>

const char* ssid     = "ssid";
const char* password = "passphrase";

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println("Connecting to WiFi");
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }

  Serial.println();
  Serial.print("WiFi connected with ip ");  
  Serial.println(WiFi.localIP());

  Serial.print("Pinging ip ");
  Serial.println(remote_ip);

  if(Ping.ping("www.google.com")) {
    Serial.println("Success!!");
  } else {
    Serial.println("Error :(");
  }
}

void loop() { }

i will also try with some other router

mobizt, 2019-03-22:
It doesn't mean this can access SSL port 443.

chander-60905, 2019-03-22:
ok esp connects and firebase runs for soft ap through laptop but not directly from router wifi.

mobizt, 2019-03-22:
This ssl port is used to connect to  Firebase server so we can't change.
I think that this may relate to low memory [as in this post](https://github.com/esp8266/Arduino/issues/4593) but I'm not sure. Please try to use only one Firebase Data Object for higher free memory and test.

chander-60905, 2019-03-23:
ok i have tested using single Firebase Data Object and setting lwIP variant to v2 lower memory or v2 Higher Bandwidth does not connect firebase to router wifi  but connects to normal mobile wifi hotspot and setting lwIP variant to v2 IPv6 lower memory or v2 IPv6 Higher bandwidth connects firebase to router wifi without changing any configuration on router and also to mobile wifi hotspot but firebase stops streaming after some time. so i think there is no problem with the router wifi.

mobizt, 2019-03-23:
Did you try fix ip, multi wifi AP. May be and may be not to confirm that router is not the problem depend on its function or algorithm based on its firewall.

chander-60905, 2019-03-23:
ok i will test and let you know

mobizt, 2019-03-23:
Thanks for your information. This actually relate to esp8266 core SDK or your router settings, not relate to this library.

chander-60905, 2019-03-23:
ok i have tested and after setting lwIP variant to v2 IPv6 lower memory works for both router wifi and mobile wifi hotspot. so is it okay to change nodemcu esp8266 from lwIP variant v2 lower memory to lwIP variant to v2 IPv6 lower memory.

mobizt, 2019-03-23:
Your info is good for everyone that work with esp8266 wifi
Thank you.

chander-60905, 2019-03-23:
Thank you for your help once again.

closed 2019-03-23 by chander-60905