mobizt/Firebase-ESP8266 #2: push json limits

blotfi, 2019-03-22:
Hi and thank you for this library
It works well but if I want to push this json it doesn't work
{
  "name": "iot4",
  "datetime": "2019/03/22 11:27:59",
  "accelX": 0.067871,
  "accelY": -0.026367,
  "accelZ": -0.928223,
  "temp": 26.65,
  "humid": 93248.09
}
where as if I put less numbers after the decimal point it works
Is there a buffer I can increase somewhere?
thanks

mobizt, 2019-03-22:
First, try to edit library in [this issue](https://github.com/mobizt/Firebase-ESP8266/issues/1#issuecomment-475478225).
Try to push this string

```C++

String json = "{\"name\":\"iot4\",\"datetime\": \"2019/03/22 11\:27\:59\",\"accelX\": 0.067871,\"accelY\": -0.026367,\"accelZ\":-0.928223,\"temp\": 26.65,\"humid\": 93248.09}";

```

It should ok. If it's ok, check how you create json string e.g. from ArduinoJson object using printTo or create as string as above string.

Not recommend to include colon (:) i.e. 11:27:59 in database because it used as key/value separator in json object.  I use escape character, back slash before colon (\:) to make it can parse as :

blotfi, 2019-03-22:
I use doc Arduinojson it is not a problem.
As I said I can send it if there is less characters
There is a limitation somewhere I need to bypass

mobizt, 2019-03-22:
My library send string as payload to server without any modification as I ask you to send json string  of above post. If it works, the problem is in json object you create. You need to check the value you add to root and read it back.

mobizt, 2019-03-22:
Can you show example code for me to test?

blotfi, 2019-03-22:
Ok but when I will be in front of my computer later

blotfi, 2019-03-22:
I mean instead of doing
`char output[128];
	serializeJson(doc, output);
  Firebase.pushJSON(firebaseData, "/data", output);
}`

mobizt, 2019-03-22:
You need to assign macho ARDUINOJSON_USE_DOUBLE  before include ArduinoJson.h like this

```C++

#define ARDUINOJSON_USE_DOUBLE 1
#include <ArduinoJson.h>

```
Here is the example to test

Try to push json data

```C++

DynamicJsonDocument doc(256);
double val1 = -0.00000001; //max 8 decimal places
double val2 = 321.12345678; //max 8 decimal places
doc["val1"] = val1;
doc["val2"] = val2;

char output[128];
memset(output, 0, 128);
serializeJson(doc, output);
Firebase.pushJSON(firebaseData, "/data", output);
```

Try to read json data back
```C++
Firebase.getJSON(firebaseData, "/data");

DynamicJsonDocument _doc(256);
deserializeJson(_doc, firebaseData.jsonData());
double _val1 = _doc["val1"];
double _val2 = _doc["val2"];
Serial.println(_val1,8);
Serial.println(_val2,8);
```

blotfi, 2019-03-22:
It seems that you did not understood me or that I didn't explain well.
Everything is working fine, it was my fault 
`char output[128];`
is not enough
`char output[150];`
is ok

I just asked if there is another way instead of going through a fixed buffer, perhaps using String

mobizt, 2019-03-22:
I know as much as information you provide, the only part of your incompleted code then I need to build the basic test and extend the possibility to handle maximum decimal places for ArduinoJSON. 

For String object it can because it has write function that ArduinoJson used for modify string.

```C++
String output="";
serializeJson(doc, output);

```

closed 2019-03-22 by blotfi

Proxcentaur, 2020-05-13:
I got this error when I was trying to put that code:
no matching function for call to 'FirebaseESP32::pushJSON(FirebaseData&, StringSumHelper&, ArduinoJson::JsonObject&)'

mobizt, 2020-05-13:
@Proxcentaur The library APIs were changed sofar. 

You need to provide the FirebaseJson object to the pushJSON function.

For example.

```C++
FirebaseJson json;

json.add("data1",100); //add (new) data
json.set("data2/new", "test"); //set (replace or create new) the data

Firebase.pushJSON(firebaseData, "your_node_path", json);
```

Proxcentaur, 2020-05-14:
> @Proxcentaur The library APIs were changed sofar.
>
> You need to provide the FirebaseJson object to the pushJSON function.
>
> For example.
>
> ```c++
> FirebaseJson json;
>
> json.add("data1",100); //add (new) [...]

thank you, it works!