mobizt/Firebase-ESP8266 #3: Firebase error: Can't get data - Path not exist

StefanVerresen, 2019-03-23:
When trying to receive data from my firebase database I always get the error message "Path not exist".
This happens even when I am using one of the examples (Basic) to test my project settings
The rules (.read and .write) are both true and I checked that it's possible to sent / receive data from it by using the simulator.

### **ESP8266 Code:**

`#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>

#define FIREBASE_HOST "edubox-49528.firebaseio.com"
#define FIREBASE_AUTH "xxx"

WiFiManager wifiManager;
FirebaseData firebaseData;

void setup()
{
Serial.begin(115200);
  wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
  wifiManager.autoConnect("EduBox-Wifi", "EduBox");

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  
  if (Firebase.getInt(firebaseData, "/edubox-49528/value"))
    {
      Serial.println("----------Get result-----------");
      Serial.println("PATH: " + firebaseData.dataPath());
      Serial.println("TYPE: " + firebaseData.dataType());
      Serial.print("VALUE: ");
      if (firebaseData.dataType() == "int")
        Serial.println(firebaseData.intData());
      else if (firebaseData.dataType() == "float")
        Serial.println(firebaseData.floatData());
      else if (firebaseData.dataType() == "string")
        Serial.println(firebaseData.stringData());
      else if (firebaseData.dataType() == "json")
        Serial.println(firebaseData.jsonData());
      Serial.println("--------------------------------");
      Serial.println();
    }
    else
    {
      Serial.println("----------Can't get data--------");
      Serial.println("REASON: " + firebaseData.errorReason());
      Serial.println("--------------------------------");
      Serial.println();
    }
}`

Here below is the error message displayed and my firebase information.

### **Error message:**
[image]

### **Firebase Information:**
[image]

mobizt, 2019-03-23:
How about to use `WiFi.begin(your_wifi_ssid,your_wifi_psw)` insead of WiFiManager library?

I mean use normal wifi connection without WiFiManager.

mobizt, 2019-03-23:
Ok I guess your problem is you include **edubox-49528** which is your project ID in the path.
Just exclude  **edubox-49528** from `Firebase.getInt(firebaseData, "/edubox-49528/value1")` to `Firebase.getInt(firebaseData, "/value1")`.

WiFiManager woks in my test.

StefanVerresen, 2019-03-23:
Damn, feel stupid :)
Works perfectly!
Can be closed :)

mobizt, 2019-03-23:
To prevent long path error, please edit FirebaseESP8266.cpp as in [this issue](https://github.com/mobizt/Firebase-ESP8266/issues/1#issuecomment-475478225)

closed 2019-03-23 by mobizt