mobizt/Firebase-ESP8266 #5: firebaseData.errorReason() returns "Connection refused"

frederik0312, 2019-04-02:
I have a problem. When i try to read from the firebase realtime database, all it returnes is connection refused.

`

//-----------------------------
//Wifi Manager
#include <ESP8266WiFi.h>          //ESP8266 Core WiFi Library (you most likely already have this in your sketch)
#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
//-----------------------------
//Firebase
#include <FirebaseESP8266.h>
#include <FirebaseESP8266HTTPClient.h>
//-----------------------------

#define relayPin 5
#define statusLED 15
#define FIREBASE_HOST "This is the right one"
#define FIREBASE_AUTH "This is the right one"

FirebaseData firebaseData;

int SMID = 123;

void setup() {

//digitalWrite(WifiLED, HIGH);
pinMode(relayPin, OUTPUT);
pinMode(statusLED, OUTPUT);
Serial.begin(115200);

WiFiManager wifiManager;
wifiManager.autoConnect("SmartHomie");

digitalWrite(statusLED, HIGH);
Serial.println("Det virker");
Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
//wifiManager.autoConnect("SmartHomie");
}
void loop() {

// Check her på MCU'ens respektive path til databasen

  
digitalWrite(relayPin, HIGH);
delay(3000);
digitalWrite(relayPin, LOW);
delay(3000);

  int val = 0;
Serial.println(Firebase.getInt(firebaseData, "number"));
  if (Firebase.getInt(firebaseData, "number")) {

    //Success, then read the payload value
Serial.println("DET FUCKINGER VIRKER!!");
    

  }else {
    //Failed, then print out the error detail
    Serial.println("Error: "+firebaseData.errorReason());
  }
}`
[image]

mobizt, 2019-04-02:
Due to WiFi client library in ESP8266 Core SDK version 5.x is different and change from old version. 

I need to notify or warn user to include file as state in[ this document](https://github.com/mobizt/Firebase-ESP8266#declaration-and-initialization) and also in example sketch comment.
FirebaseESP8266.h must be included before ESP8266WiFi.h to let it works under ESP8266 Core SDK version 5.x. 

For your case, include these 3 files as following.

```C++
#include <FirebaseESP8266.h> //include first
#include <ESP8266WiFi.h> //the second
#include <WiFiManager.h> 
```

Please let me know if you still get the error.

frederik0312, 2019-04-03:
I did what you wrote and I still get the error :(.

mobizt, 2019-04-03:
From your code, the format of **path** parameter you pass to function is not valid. Path must begin with slash (**/**)
Becase it is a relative path from your root to defined node.

```C++
Firebase.getInt(firebaseData, "number");
```
Should be
```C++
Firebase.getInt(firebaseData, "/number");
```

frederik0312, 2019-04-03:
thanks but I still get the same error.

mobizt, 2019-04-03:
Please explain the error. Connection refuse?

mobizt, 2019-04-03:
I try your code, everything is ok.

```C++
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>

#define relayPin 5
#define statusLED 15
#define FIREBASE_HOST "xxxxxxxx.firebaseio.com"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
FirebaseData firebaseData;
int SMID = 123;
void setup()
{
  //digitalWrite(WifiLED, HIGH);
  pinMode(relayPin, OUTPUT);
  pinMode(statusLED, OUTPUT);
  Serial.begin(115200);
  WiFiManager wifiManager;
  wifiManager.autoConnect("SSID", "PSWXXXXXXXXXX");
  digitalWrite(statusLED, HIGH);
  Serial.println("Det virker");

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  //wifiManager.autoConnect("SmartHomie");
}
void loop()
{
  // Check her på MCU'ens respektive path til databasen
  digitalWrite(relayPin, HIGH);
  delay(3000);
  digitalWrite(relayPin, LOW);
  delay(3000);
  int val = 0;
  Serial.println(Firebase.getInt(firebaseData, "/number"));
  if (Firebase.getInt(firebaseData, "/number"))
  {
    //Success, then read the payload value
    Serial.println("DET FUCKINGER VIRKER!!");
  }
  else
  {
    //Failed, then print out the error detail
    Serial.println("Error: " + firebaseData.errorReason());
  }
}

```

frederik0312, 2019-04-03:
I use 2.5

frederik0312, 2019-04-03:
when I try to run the demo program it doesnt work either

frederik0312, 2019-04-03:
[image]

mobizt, 2019-04-03:
Thank you for your information.

Connection refused return directly from esp8266 core SDK which means it cannot establish network connection or cannot connect to internet even your AP is connected.

Try to reset the router, use different router, set soft AP or WiFi hotspot using your computer or mobile phone.

I'm not sure this related to [this issue](https://github.com/mobizt/Firebase-ESP8266/issues/1#issuecomment-475661185)?

frederik0312, 2019-04-03:
I already use a mobile hotspot but il try another :)

mobizt, 2019-04-03:
Please try to change lwIP variant.
From Arduino IDE, go to menu **Tools** > **lwIP Variant** > **v2 IPv6 Lower Memory** or other options.

frederik0312, 2019-04-03:
i havenow tried that, and it dosent work with any of the variants... i also tried switching to other networks and using a new NodeMCU

mobizt, 2019-04-03:
How about try to install Esp8266 Arduino core v 2.4.2?

frederik0312, 2019-04-03:
that dosent work either... :(

frederik0312, 2019-04-03:
i have been trying to get this to work for days now, i really hope you can help me find a solution

mobizt, 2019-04-03:
I will reply you back within an hour.

mobizt, 2019-04-03:
Ok, I try to install esp8266 Arduino core back to v 2.5.0 and test with this example
Go to menu File > **Examples** > **ESP8266WiFi** > **HTTPSRequestAxTLS** 
And post the serial printed text from it.

mobizt, 2019-04-03:
This is  how I clean install everything to test the library.

- Clean install Arduino 1.8.9 by uninstall old version and delete folder \AppData\Local\Arduino15 to delete old sdk.
- Create new libraries folder.
- Install ESP8266 Arduino core SDK by provide this url in Settings http://arduino.esp8266.com/stable/package_esp8266com_index.json
- Install ESP8266 Arduino Core version 2.5 from Boards Manager.
- Install Firebase ESP8266 Arduino library v 1.0.3 from Library Manager.
- Use your code to test.

Everything works fine.

frederik0312, 2019-04-03:
thanks, this is the serial print from HTTPSRequestAxTLS:
13:57:44.017 -> certificate matches
13:57:44.017 -> requesting URL: /repos/esp8266/Arduino/commits/master/status
13:57:44.017 -> request sent
13:57:44.256 -> headers received
13:57:59.694 -> esp8266/Arduino CI successfull!
13:57:59.694 -> reply was:
13:57:59.694 -> ==========
13:57:59.694 -> {"state":"success","statuses":[{"url":"https://api.github.com/repos/esp8266/Arduino/statuses/ca79f2ce39724527052319717a731b3501d5f0b2","avatar_url":"https://avatars2.githubusercontent.com/oa/1508?v=4","id":6534478199,"node_id":"MDEzOlN0YXR1c0NvbnRleHQ2NTM0NDc4MTk5","state":"success","description":"The Travis CI build passed","target_url":"https://travis-ci.org/esp8266/Arduino/builds/514415197?utm_source=github_status&utm_medium=notification","context":"continuous-integration/travis-ci/push","created_at":"2019-04-02T01:16:52Z","updated_at":"2019-04-02T01:16:52Z"}],"sha":"ca79f2ce39724527052319717a731b3501d5f0b2","total_count":1,"repository":{"id":32969220,"node_id":"MDEwOlJlcG9zaXRvcnkzMjk2OTIyMA==","name":"Arduino","full_name":"esp8266/Arduino","private":false,"owner":{"login":"esp8266","id":8943775,"node_id":"MDEyOk9yZ2FuaXphdGlvbjg5NDM3NzU=","avatar_url":"https://avatars1.githubusercontent.com/u/8943775?v=4","gravatar_id":"","url":"https://api.github.com/users/esp8266","html_url":"https://github.com/esp8266","followers_url":"https://api.github.com/users/esp8266/followers","following_url":"https://api.github.com/users/esp8266/following{/other_user}","gists_url":"https://api.github.com/users/esp8266/gists{/gist_id}","starred_url":"https://api.github.com/users/esp8266/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/esp8266/subscriptions","organizations_url":"https://api.github.com/users/esp8266/orgs","repos_url":"https://api.github.com/users/esp8266/repos","events_url":"https://api.github.com/users/esp8266/events{/privacy}","received_events_url":"https://api.github.com/users/esp8266/received_events","type":"Organization","site_admin":false},"html_url":"https://github.com/esp8266/Arduino","description":"ESP8266 core for Arduino","fork":false,"url":"https://api.github.com/repos/esp8266/Arduino","forks_url":"https://api.github.com/repos/esp8266/Arduino/forks","keys_url":"https://api.github.com/repos/esp8266/Arduino/keys{/key_id}","collaborators_url":"https://api.github.com/repos/esp8266/Arduino/collaborators{/collaborator}","teams_url":"https://api.github.com/repos/esp8266/Arduino/teams","hooks_url":"https://api.github.com/repos/esp8266/Arduino/hooks","issue_events_url":"https://api.github.com/repos/esp8266/Arduino/issues/events{/number}","events_url":"https://api.github.com/repos/esp8266/Arduino/events","assignees_url":"https://api.github.com/repos/esp8266/Arduino/assignees{/user}","branches_url":"https://api.github.com/repos/esp8266/Arduino/branches{/branch}","tags_url":"https://api.github.com/repos/esp8266/Arduino/tags","blobs_url":"https://api.github.com/repos/esp8266/Arduino/git/blobs{/sha}","git_tags_url":"https://api.github.com/repos/esp8266/Arduino/git/tags{/sha}","git_refs_url":"https://api.github.com/repos/esp8266/Arduino/git/refs{/sha}","trees_url":"https://api.github.com/repos/esp8266/Arduino/git/trees{/sha}","statuses_url":"https://api.github.com/repos/esp8266/Arduino/statuses/{sha}","languages_url":"https://api.github.com/repos/esp8266/Arduino/languages","stargazers_url":"https://api.github.com/repos/esp8266/Arduino/stargazers","contributors_url":"https://api.github.com/repos/esp8266/Arduino/contributors","subscribers_url":"https://api.github.com/repos/esp8266/Arduino/subscribers","subscription_url":"https://api.github.com/repos/esp8266/Arduino/subscription","commits_url":"https://api.github.com/repos/esp8266/Arduino/commits{/sha}","git_commits_url":"https://api.github.com/repos/esp8266/Arduino/git/commits{/sha}","comments_url":"https://api.github.com/repos/esp8266/Arduino/comments{/number}","issue_comment_url":"https://api.github.com/repos/esp8266/Arduino/issues/comments{/number}","contents_url":"https://api.github.com/repos/esp8266/Arduino/contents/{+path}","compare_url":"https://api.github.com/repos/esp8266/Arduino/compare/{base}...{head}","merges_url":"https://api.github.com/repos/esp8266/Arduino/merges","archive_url":"https://api.github.com/repos/esp8266/Arduino/{archive_format}{/ref}","downloads_url":"https://api.github.com/repos/esp8266/Arduino/downloads","issues_url":"https://api.github.com/repos/esp8266/Arduino/issues{/number}","pulls_url":"https://api.github.com/repos/esp8266/Arduino/pulls{/number}","milestones_url":"https://api.github.com/repos/esp8266/Arduino/milestones{/number}","notifications_url":"https://api.github.com/repos/esp8266/Arduino/notifications{?since,all,participating}","labels_url":"https://api.github.com/repos/esp8266/Arduino/labels{/name}","releases_url":"https://api.github.com/repos/esp8266/Arduino/releases{/id}","deployments_url":"https://api.github.com/repos/esp8266/Arduino/deployments"},"commit_url":"https://api.github.com/repos/esp8266/Arduino/commits/ca79f2ce39724527052319717a731b3501d5f0b2","url":"https://api.github.com/repos/esp8266/Arduino/commits/ca79f2ce39724527052319717a731b3501d5f0b2/status"}
13:58:00.113 -> ==========
13:58:00.113 -> closing connection

frederik0312, 2019-04-03:
[image]

This is the log from HTTPSRequestAxTLS

mobizt, 2019-04-03:
Please try this

```C++

#include <WiFiClientSecureAxTLS.h>
using namespace axTLS;
WiFiClientSecure client;
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "xxxxxxxxxxxxxxxx.firebaseio.com"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to WiFi");
  WiFi.mode(WIFI_STA);
  WiFi.begin("SSID", "PSWxxxxxxxxx");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(200);
  }

  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.println();

  if (client.connect(FIREBASE_HOST, 443)) {

    Serial.println("Connect to Firebase OK");
    client.print("GET /number.json?auth=");
    client.print(FIREBASE_AUTH);
    client.print(" HTTP/1.1\r\n");
    client.print("Host: ");
    client.print(FIREBASE_HOST);
    client.print("\r\n");
    client.print("Connection: close\r\n\r\n");

    while (client.connected() && !client.available())
      delay(1);

    if (client.connected() && client.available())
      while (client.available())
        Serial.print((char)client.read());
  } else {
    Serial.println("Connect to Firebase failed!");
  }
}

void loop() {

}

```

And post the printed serial.

frederik0312, 2019-04-03:
ok, i tried this and it still dosent work :(

Could you try to use my database to  see if thats the problem?

Its just a test database so its fine:

#define FIREBASE_HOST "smarthome2-be8a1.firebaseio.com/"
#define FIREBASE_AUTH "ucHoY5ssFUmTld7VTAq71RDouQvxfjkYjJJdvfgY"

frederik0312, 2019-04-03:
..............
WiFi connected.
IP address: 
192.168.43.141

Connect to Firebase failed!

mobizt, 2019-04-03:
OK it works. I think I found the error in your FIREBASE_HOST you wrote "........firebaseio.**com/**".
Slash behind **.com** is not valid, it should be "????????.firebaseio.com"

frederik0312, 2019-04-03:
ok now it dosent return a error!

hwhat code should i use to read something from a path at the database

mobizt, 2019-04-03:
What do you mean? 

If you want to khow how to use, the examples or read me on the main page will provide complete usages.

For path parameter, you need to add slash at its beginning e.g. "/Room/Node1/Senor1". 

For FIREBASE_HOST that pass to Firebase.begin function should be valid in form "**xxxx.firebaseio.com**", no http:// or https:// or slash at beginning or at the end.

closed 2019-04-04 by mobizt

lc998, 2019-09-30:
> OK it works. I think I found the error in your FIREBASE_HOST you wrote "........firebaseio.**com/**".
> Slash behind **.com** is not valid, it should be "????????.firebaseio.com"

This was the problem for me :P

lc998, 2019-09-30:
> What do you mean?
>
> If you want to khow how to use, the examples or read me on the main page will provide complete usages.
>
> For path parameter, you need to add slash at its beginning e.g. [...]

And this was useful!

xpertcrush, 2020-06-17:
I have the same issue, did as you suggested. still having the connection refused error. I am just setting up a sensor value.

mobizt, 2020-06-17:
@xpertcrush

Please post your code here. 
The issue is too old and the library updates so far.

xpertcrush, 2020-06-17:
Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
Firebase.reconnectWiFi(true);
  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(1024, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(1024);

  //Set database read timeout to 1 minute (max 15 minutes)
  Firebase.setReadTimeout(firebaseData, 1000 * 60);
  //tiny, small, medium, large and unlimited.
  //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
  Firebase.setwriteSizeLimit(firebaseData, "tiny");

 /*
  This option allows get and delete functions (PUT and DELETE HTTP requests) works for device connected behind the
  Firewall that allows only GET and POST requests.
  
  Firebase.enableClassicRequest(firebaseData, true);
  */

  String path = "/";
 
  Serial.println("------------------------------------");
  Serial.println("Set double test...");

    Serial.println ("beginning to send data to firebase");
    //Also can use Firebase.set instead of Firebase.setDouble
    if (Firebase.setString(firebaseData, path + "/roof/level" , WaterLevel().c_str()))
    {
      Serial.println("PASSED");
      Serial.println("PATH: " + firebaseData.dataPath());
      Serial.println("TYPE: " + firebaseData.dataType());
      Serial.println("ETag: " + firebaseData.ETag());
      Serial.print("VALUE: ");
      Serial.println("------------------------------------");
      Serial.println();
    }
    else
    {
      Serial.println("FAILED");
      Serial.println("REASON: " + firebaseData.errorReason());
      Serial.println("------------------------------------");
      Serial.println();
    }

 
very basic code

mobizt, 2020-06-17:
@xpertcrush

Can't see WiFi connection.

xpertcrush, 2020-06-17:
> 
#include <FirebaseESP8266.h> //include first
#include <ESP8266WiFi.h> //the second
//#include <WiFiManager.h> 
#include <ArduinoOTA.h>
#include <hcsr04.h>
#include "FirebaseESP8266.h"

//DECLARE THE CREDENTIALS
#ifndef STASSID
#define STASSID "Virut"
#define STAPSK  "Wro23456"
#endif

#define FIREBASE_HOST "AC?????.firebaseio.com"
#define FIREBASE_AUTH "_hidden_"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;

FirebaseJson json;

bool ota_flag = true;
uint16_t timer_start = 0;
// Time to sleep (in seconds):
const int sleepTimeS = 60;
int analogInput = A0;
float correctionFactor = 0;
float Vout = 0.0;
float Vin = 0.0;
float R1 = 98190;
float R2 = 10150;
int value = 0;

const char* ssid = STASSID;
const char* password = STAPSK;
const char* deviceName = "findmylevel.com";

//ATTACH THE TRIGGER AND ECHO TO PINS
#define TRIG_PIN 5
#define ECHO_PIN 4

//Static IP address configuration
IPAddress staticIP(192, 168, 1, 150); //ESP static ip
IPAddress gateway(192, 168, 1, 1);   //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0);  //Subnet mask
IPAddress dns(8, 8, 8, 8);  //DNS

void setup() {

  pinMode(2,OUTPUT);
  digitalWrite(2,HIGH);
  pinMode(TRIG_PIN, OUTPUT);
  pinMode(ECHO_PIN, INPUT);
  Serial.begin(115200);
  Serial.println("Booting");
//WIFI MODE SETTINGS
  WiFi.disconnect();  //Prevent connecting to wifi based on previous configuration
  WiFi.hostname(deviceName);      // DHCP Hostname (useful for finding device for static lease)
  WiFi.config(staticIP, subnet, gateway, dns);  
  WiFi.mode(WIFI_STA);

//CONNECTION INITIALIZATION  
  
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(100);
  }

here they are

mobizt, 2020-06-17:
@xpertcrush

The problem is your device can't connect to internet.

You need to test the basic WiFi Client or HTTPClient connection to some server e.g. google.com.

The issue is not related to library.

xpertcrush, 2020-06-17:
I see, its connected to router which has connectivity. and dns is set to google. let me check with HTTPSrequest example and report back here.

mobizt, 2020-06-17:
Ok, that's the way you should do before anything.

xpertcrush, 2020-06-17:
14:02:48.066 -> .
14:02:48.066 -> WiFi connected
14:02:48.066 -> IP address: 
14:02:48.066 -> 192.168.1.180
14:02:48.066 -> connecting to api.github.com
14:02:48.492 -> cert FP: 59 74 61 88 13 CA 12 34 15 4D 11 0A C1 7F E6 67 07 69 42 F5 
14:02:48.532 -> test FP: 5F F1 60 31 09 04 3E F2 90 D2 B0 8A 50 38 04 E8 37 9F BC 76 
14:02:48.532 -> certificate doesn't match
14:02:48.532 -> requesting URL: /repos/esp8266/Arduino/commits/master/status
14:02:48.532 -> request sent
14:02:48.836 -> headers received

its fine

mobizt, 2020-06-17:
@xpertcrush
May be you provide something wrong with FIREBASE_HOST and FIREBASE_AUTH.
You should check or recreate the new database secret.

xpertcrush, 2020-06-17:
even uncommented   Firebase.enableClassicRequest(firebaseData, true);

still same error, let me try new database secret and see what happens

xpertcrush, 2020-06-17:
is it related to new feature of firebase?

Firebase Admin SDK
Your Firebase service account can be used to authenticate multiple Firebase features, such as Database, Storage and Auth, programmatically via the unified Admin SDK. Learn more

Firebase service account
firec@accuam.gserviceaccount.com

mobizt, 2020-06-17:
@xpertcrush 
No, it's not related to the SDK.

This is the REST API client, internet capable device can communicate to Firebase without SDK installed.

[Just follow this for the Firebase project setup](https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp8266-9937d98d1ae0)

xpertcrush, 2020-06-17:
tried with completely new account and database still connection refused error, it freezes serial data for 12-16 seconds and through that error.

mobizt, 2020-06-17:
@xpertcrush

Tell me about your device or board model, ESP8266 Core SDK version, IDE version, board, flash and lwIp setting in IDE.

Post your complete sketch not just some code portion like this.

Complete serial debug message.

xpertcrush, 2020-06-17:
[image]
see i am getting connection on firebase database

xpertcrush, 2020-06-17:
device is esp8266 12f core skd version is latest, everything is latest already

here is the flash setting screenshot

[image]

xpertcrush, 2020-06-17:
[image]

I tried pinging outside my network and its fine as well.

mobizt, 2020-06-17:
You don't provide what exactly I asked then I don't know you are understand my request.

Don't conclude with latest. No code nothing to verify.

I can't provide any further help

mobizt, 2020-06-17:
This is not the Q&A forum which no one care about your questions and problems.

This is the place where I take my time to support you.

xpertcrush, 2020-06-17:
/*
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2020 mobizt
 *
*/

//This example shows how to set and push timestamp (server time) which is the server variable that suopported by Firebase

//FirebaseESP8266.h must be included before ESP8266WiFi.h
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "rooftoplevel.firebaseio.com"
#define FIREBASE_AUTH "KgHSpodbojToDDTXarIwOdtupbR6NPjcKDfdknz2"
#define WIFI_SSID "VirusAlert"
#define WIFI_PASSWORD "WrongEnd123456"

//Define Firebase Data object
FirebaseData firebaseData;

void setup()
{

  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(1024, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(1024);

  /*
  This option allows get and delete functions (PUT and DELETE HTTP requests) works for device connected behind the
  Firewall that allows only GET and POST requests.
  
  Firebase.enableClassicRequest(firebaseData, true);
  */

  String path = "/Test";

  Serial.println("------------------------------------");
  Serial.println("Set Timestamp test...");

  if (Firebase.setTimestamp(firebaseData, path + "/Set/Timestamp"))
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + firebaseData.dataPath());
    Serial.println("TYPE: " + firebaseData.dataType());

    //Timestamp saved in millisecond, get its seconds from intData()
    Serial.print("TIMESTAMP (Seconds): ");
    Serial.println(firebaseData.intData());

    //Or print the total milliseconds from doubleData()
    //Due to bugs in Serial.print in Arduino library, use printf to print double instead.
    printf("TIMESTAMP (milliSeconds): %.0lf\n", firebaseData.doubleData());

    //Or print it from payload directly
    Serial.print("TIMESTAMP (milliSeconds): ");
    Serial.println(firebaseData.payload());

    //Due to some internal server error, ETag cannot get from setTimestamp
    //Try to get ETag manually
    Serial.println("ETag: " + Firebase.getETag(firebaseData, path + "/Set/Timestamp"));
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  Serial.println("------------------------------------");
  Serial.println("Get Timestamp (double of milliseconds) test...");

  if (Firebase.getDouble(firebaseData, path + "/Set/Timestamp"))
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + firebaseData.dataPath());
    Serial.println("TYPE: " + firebaseData.dataType());

    printf("TIMESTAMP: %.0lf\n", firebaseData.doubleData());
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  Serial.println("------------------------------------");
  Serial.println("Push Timestamp test...");

  if (Firebase.pushTimestamp(firebaseData, path + "/Push/Timestamp"))
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + firebaseData.dataPath());
    Serial.print("PUSH NAME: ");
    Serial.println(firebaseData.pushName());

    //Due to some internal server error, ETag cannot get from pushTimestamp
    //Try to get ETag manually
    Serial.println("ETag: " + Firebase.getETag(firebaseData, path + "/Push/Timestamp/" + firebaseData.pushName()));
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
}

this is the example which is included in library and here is the output.

Sorry about not posting code earlier, totally overlooked. Sorry again. 

[image]

surprisingly it worked. Thank you for your time.

xpertcrush, 2020-06-19:
Dear mobizt

[image]

everything is fine if I do not set IP address as static. here is the code.

if I setup static IP address it refuses connection with firebase.

xpertcrush, 2020-06-19:
if i setup dns to 192.168.1.1 it works fine but then it doesn't setup static Ip and choose one from DHCP.

mobizt, 2020-06-19:
It's not the library issue.
It's your router and network configuration issue.

mobizt, 2020-06-19:
Library does not touch the network stacks, it plays with requests and responses.

xpertcrush, 2020-06-19:
> Library does not touch the network stacks, it plays with requests and responses.

you are right. it is a router issue, I will try with another router or a hotspot. thank you once again for the response. you did brilliant work on this library. Thank you so much.

N3utral03, 2020-12-14:
Hello, I have a problem with "read Timeout"
[image]

The value in my Firebase Database changed just fine.

`#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST ".firebaseio.com"
#define FIREBASE_AUTH "x"
#define WIFI_SSID "x"
#define WIFI_PASSWORD "x"
#define LED D13
FirebaseData firebaseData;
String c;

void setup()
{
  Serial.begin(115200);
  pinMode(LED, OUTPUT);
  
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }

  Serial.println();
  Serial.print("Connected ...");
  //Serial.println(WiFi.localIP());
  //Serial.println();
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (Firebase.setString(firebaseData, "LED", "0"))
  {
    //Success
    Serial.println("Set int data success");

  } else {
    //Failed?, get the error reason from firebaseData
    Serial.print("Error in setString, ");
    Serial.println(firebaseData.errorReason());
  }
}

void loop()
{
  if (Firebase.get(firebaseData, "/LED"))
  {
    //Success
    Serial.print("Get int data success, int = ");
    Serial.println(firebaseData.stringData());
    c = firebaseData.stringData();
    digitalWrite(LED, (c.toInt()));

  } else {
    //Failed?, get the error reason from firebaseData
    Serial.print("Error in getInt, ");
    Serial.println(firebaseData.errorReason());
  }
  delay(2000);
}`

Thank You in advance

mobizt, 2020-12-14:
@N3utral03  your router or AP that your device connected is not response, no internet or unstable internet or connect trough some login portal of ISP or your router hang. 

You should test with the Arduino examples to make https request using BearSSL WiFiClientSecure and running repeatedly in the loop, you will get the similar results.

N3utral03, 2020-12-15:
> @N3utral03 your router or AP that your device connected is not response, no internet or unstable internet or connect trough some login portal of ISP or your router hang.
>
> You should test with the [...]

right. turns out it's my unstable connection. try using another connection today and works just fine.
thank you!

nuraisa29, 2021-06-30:
I have the same problem. When i try to read from the firebase realtime database, all it returnes is connection refused.

this is my code, anyone can help me?

#include <SoftwareSerial.h>

// for WiFi
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#define WIFI_SSID "xxxxx"
#define WIFI_PASSWORD "xxxxx"

//for FireBase
#define FIREBASE_HOST "xxxxx"
#define FIREBASE_AUTH "xxxxx"

FirebaseData firebaseData;

//buat variabel untuk software serial (Rx, Tx)
SoftwareSerial DataSerial(12, 13);

//millis sebagai pengganti delay
unsigned long previousMillis = 0;
const long interval = 1000;

//variabel array untuk data parsing
String arrData[3];

void setup() 
{
  // connect to wifi.
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("connecting");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected: ");
  Serial.println(WiFi.localIP());
  
  //ke firebase
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Serial.begin(9600);
  DataSerial.begin(9600);
}

void loop() 
{
 //konfigurasi millis
 unsigned long currentMillis = millis(); //baca waktu millis saat ini
 if(currentMillis - previousMillis >= interval)
 { 
  //update previousMillis
  previousMillis = currentMillis;

  //prioritaskan pembacaan data dari arduino uno (hasil kiriman data)
  //baca data serial
  String Data = ""; //variabel yang menampung pembacaan DataSerial
  while(DataSerial.available()>0)
  {
    Data += char (DataSerial.read());
  }
  //buang spasi datanya
  Data.trim();

  //uji data
  if(Data != "")
  {
    //format data "10#200#1" = array
    //parsing data (pecah data)
    int index = 0;
    for(int i=0; i<= Data.length(); i++)
    {
      char delimiter = '#';
      if(Data[i] != delimiter)
        arrData[index] += Data[i];
      else
        index++; //variabel index bertambah 1
    }

    //pastikan bahwa data yang dikirim lengkap (LDR, HUJAN, PJEMURAN)
    //urutannya 0 = LDR, 1 = Hujan, 2 = PJemuran
    if(index == 2)
    {
      //tampilkan nilai sensor ke serial monitor 
      Serial.println("LDR : " + arrData[0]);                                                //LDR                                                                       
      Serial.println("Hujan : " + arrData[1]);                                           //Hujan                                     
      Serial.println("Posisi Jemuran [1]Di dalam / [0]Di luar : " + arrData[2]);        //PJemuran
      Serial.println();
      delay(2000);

    }

    arrData[0] = "";
    arrData[1] = "";
    arrData[2] = "";
  }
  
  //minta data ke arduino uno
  DataSerial.println("Ya");
  
 }
 
 // set value LDR di FireBase
  if (Firebase.setString(firebaseData,"/Hasil_Pembacaan/LDR", arrData[0]))
  {
  Serial.println("Data LDR terkirim");
   } 
   else
   {
      Serial.println("Data LDR tidak terkirim");
      Serial.println("Karena: " + firebaseData.errorReason());
      return;
   }
   
  if (Firebase.setString(firebaseData,"/Hasil_Pembacaan/Hujan", arrData[1]))
  {
  Serial.println("Data Hujan terkirim");
   } 
   else
   {
      Serial.println("Data Hujan tidak terkirim");
      Serial.println("Karena: " + firebaseData.errorReason());
      return;
   }
  
  if (Firebase.setString(firebaseData,"/Hasil_Pembacaan/Posisi Jemuran : ", arrData[2]))
  {
  Serial.println("Data Posisi Jemuran terkirim");
   } 
   else
   {
      Serial.println("Data LDR tidak terkirim");
      Serial.println("Karena: " + firebaseData.errorReason());
      return;
   }
    delay(1000);
}

mobizt, 2021-06-30:
@nuraisa29  

Change the AP or router. Connection refused error is the internet connection problem.

[14 more replies]