mobizt/Firebase-ESP8266 #7: Connection being refused by firebaseData.errorReason()

AMAN-KHATRI, 2019-04-17:
I have already read the issue opened previously but it doesn't seem to work out for me.

So, I am trying to connect to firebase for a stream connection but I get a message "connection refused". Here's my code:-

#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

FirebaseData firebaseData;

String PATH = "/LEDStatus/LED1Status";

void setup(){
  Serial.begin(115200);

  // Connecting to wifi
  Serial.println("Connecting to WiFi");
  WiFi.begin("SSID", "Password");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }
  Serial.println("Connected to wifi");

  Firebase.begin("xxxxxxxxxxxxxx.com", "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx");
  Serial.println("Firebase connected");

  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData, PATH)){
    //Could not begin stream connection, then print out the error detail
    Serial.println(firebaseData.errorReason());
  }
  Serial.println("Stream connected");

}

void loop(){
  if (!Firebase.readStream(firebaseData))
  {
    //If read stream was failed, print the error detail.
    Serial.println(firebaseData.errorReason());
  }

  if (firebaseData.streamTimeout())
  {
     //If stream timeout, just notify
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable())
  {
   
   //Print out the value
   //Stream data can be many types which can be determined from function dataType

    if (firebaseData.dataType() == "int")
      Serial.println(firebaseData.intData());
    else if (firebaseData.dataType() == "float")
      Serial.println(firebaseData.floatData());
    else if (firebaseData.dataType() == "string")
      Serial.println(firebaseData.stringData());
    else if (firebaseData.dataType() == "json")
      Serial.println(firebaseData.jsonData());
    
  }
}

[image]

mobizt, 2019-04-17:
This is because device cannot connect to internet even the WiFi was connected. 

The error connection refuse is directly return by esp8266 core when connection to server was failed.

Please try the following.
 
- Reset your router.
- Connect to another WiFi hotspot.
- Connect to soft AP created with your computer.
- Connect to your mobile phone WiFi hotspot.

AMAN-KHATRI, 2019-04-17:
I have already tried connecting with my mobile phone hotspot and also tried connecting with my laptop hotspot. I cross checked and the Internet connection was properly working, actually I have also developed an app for updating values on firebase database and that app is working properly which ensures that Internet connection was proper and working.
Already tried connecting to 3 to 4 different WiFi hotspots, but the problem still retains.

mobizt, 2019-04-17:
Please try this

```C++

#include <WiFiClientSecureAxTLS.h>
using namespace axTLS;
WiFiClientSecure client;
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "xxxxxxxxxxxxxxxx.firebaseio.com"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to WiFi");
  WiFi.mode(WIFI_STA);
  WiFi.begin("SSID", "PSWxxxxxxxxx");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(200);
  }

  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.println();

  if (client.connect(FIREBASE_HOST, 443)) {

    Serial.println("Connect to Firebase OK");
    client.print("GET /number.json?auth=");
    client.print(FIREBASE_AUTH);
    client.print(" HTTP/1.1\r\n");
    client.print("Host: ");
    client.print(FIREBASE_HOST);
    client.print("\r\n");
    client.print("Connection: close\r\n\r\n");

    while (client.connected() && !client.available())
      delay(1);

    if (client.connected() && client.available())
      while (client.available())
        Serial.print((char)client.read());
  } else {
    Serial.println("Connect to Firebase failed!");
  }
}

void loop() {

}

```

And post the printed serial.

mobizt, 2019-04-18:
If problem still existed, please update ESP8266 Arduino Core.

From Arduino IDE, go to **Tools** > **Board: "xxxxx"** > **Boards Manager..**, search for esp8266 and select version 2.5.0 and install.

AMAN-KHATRI, 2019-04-18:
Thank you for the response.
I have already installed the correct version i.e. 2.5.0 of the ESP8266 Arduino Core library.

Here's the printed serial output.

[image]

mobizt, 2019-04-18:
Ok now you're ready to connect.
The library should work now.

The problem i guess is the version of Arduino core version that checked in my library are 2.4.0, 2.4.1, 2.4.2 and 2.5.0. If user install beta or rc version or older version, some files are not included properly due to WiFi client changes since v 2.4.2 and 2.5.0.

mobizt, 2019-04-18:
Ok now you're ready to connect.
The library should work now.

The problem I guess is the versions of Arduino core that checked in my library are 2.4.0, 2.4.1, 2.4.2 and 2.5.0. If user installs beta or rc version or older version, some files are not included properly due to WiFi client changes since v 2.4.2 and 2.5.0.

AMAN-KHATRI, 2019-04-21:
Okay, so I have already updated the library and it worked for 2-3 times properly, but again it started to cause the same trouble. Please help to solve this issue ASAP. Here's a screenshot-

[image]

AMAN-KHATRI, 2019-04-21:
To ensure Internet connectivity I have also performed ping test.

[image]

mobizt, 2019-04-22:
From Arduino IDE go to menu 
**Tools** > **Debug port:**, choose "**Serial**"
**Tools** > **Debug Level:**, choose "**WIFI**"
**Tools** > **lwIP Variant:**, choose "**v2 Lower Memory**"

Upload below sketch.

Copy all serial and post it when it shows Connect to Firebase failed!

```C++

#include <WiFiClientSecureAxTLS.h>
using namespace axTLS;
WiFiClientSecure client;
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com"
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"
#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"

unsigned long lastTime = 0;
uint32_t count = 0;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to WiFi");
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(200);
  }
  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();
}

void loop() {

  if (millis() - lastTime > 5000) {
    lastTime = millis();
    if ( WiFi.status() != WL_CONNECTED) {
      Serial.println("WIFi connection lost");
      Serial.println();
      return;
    }
    if (client.connect(FIREBASE_HOST, 443)) {
      Serial.println();
      Serial.println(">> " + String(count));
      client.print("PUT /----test----.json?auth=");
      client.print(FIREBASE_AUTH);
      client.print(" HTTP/1.1\r\n");
      client.print("Host: ");
      client.print(FIREBASE_HOST);
      client.print("\r\n");
      client.print("Connection: keep-alive\r\n");
      client.print("Keep-Alive: timeout=30, max=100\r\n");
      client.print("Content-Length: 1\r\n\r\n");
      client.print(0);
      while (client.connected() && !client.available())
        delay(1);
      if (client.connected() && client.available())
        while (client.available())
          Serial.print((char)client.read());
      Serial.println();
      Serial.println("<<");
      client.print("GET /----test----.json?auth=");
      client.print(FIREBASE_AUTH);
      client.print(" HTTP/1.1\r\n");
      client.print("Host: ");
      client.print(FIREBASE_HOST);
      client.print("\r\n");
      client.print("Connection: close\r\n\r\n");
      while (client.connected() && !client.available())
        delay(1);
      if (client.connected() && client.available())
        while (client.available())
          Serial.print((char)client.read());
      Serial.println();
    } else {
      Serial.println("Connect to Firebase failed!");
    }
  }
}

```

AMAN-KHATRI, 2019-04-22:
It's connecting to firebase successfully, the connection only fails when I turn off the mobile data from my smartphone...otherwise everything works just fine.

[image]

mobizt, 2019-04-22:
If this works, the library should work too, because this is the way I use in the library.

AMAN-KHATRI, 2019-04-22:
But still it is showing connection refused, and my Internet connection is working properly as you can see in the image. Both the device my laptop and esp8266 are connected to the same WiFi network.

[image]

mobizt, 2019-04-22:
Please show your serial debug and your sketch.

mobizt, 2019-04-22:
Make sure **FIREBASE_HOST** should **not ending with any character after .com** and **not leading with https://**
#define FIREBASE_HOST "???????.firebaseio.**com**" 
#define FIREBASE_AUTH "????????????????"

mobizt, 2019-04-22:
And update Firebase-ESP8266 library to v 1.0.5.

AMAN-KHATRI, 2019-04-22:
FIREBASE_HOST and FIREBASE_AUTH are configured correctly and I have updated the Firebase-ESP8266 library recently. 

[image]

mobizt, 2019-04-22:
Please paste your code (delete your host and auth).

mobizt, 2019-04-22:
Didi you missed 

```C++
  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
```

AMAN-KHATRI, 2019-04-22:
Here's a link to my code-

https://github.com/AMAN-KHATRI/ESP8266/blob/master/Clikofy.ino

mobizt, 2019-04-22:
The problem is due to Firebase server connection was reset when use ping  and my library was not detect any change through WiFi core and  assumed that the connection to Firebase is still alive and not need to restart server server connection.

To work around this issue at this time, please call ping before call begin stream or avoid to use ping.

This issue will be fixed in the next update soon.

AMAN-KHATRI, 2019-04-22:
Okay, now I have removed ping completely from the code but the issue still persists. I really don't have any idea why this is happening, I have already tried restarting the router, connecting with my mobile phone hotspot, also tried connecting from one of my friend's mobile hotspot but yet no visible change.
Please help me to solve this issue. Thankyou.

[image]

mobizt, 2019-04-22:
Ok I finally [fixed](https://github.com/mobizt/Firebase-ESP8266/commit/7a237dde000fc8ed3b775e4a5dd0d77e588826c0) this issue and some bugs in mode switching. The version is still 1.0.5, please update the library and check.

affibox, 2019-04-23:
> Ok I finally [fixed](https://github.com/mobizt/Firebase-ESP8266/commit/7a237dde000fc8ed3b775e4a5dd0d77e588826c0) this issue and some bugs in mode switching. The version is still 1.0.5, please update the library and check.

still not working on my end and I just downloaded this library on a brand new arduino IDE installation.

mobizt, 2019-04-23:
@affibox Which ESP8266 Arduino Core version you installed? 

Please update ESP8266 Arduino Core to version 2.5.0 (or older 2.4.2).

@affibox and @AMAN-KHATRI 
Please delete the current version of Firebase-ESP8266 from library folder.

Down load and install the new one of v 1.0.5 (I just fixed the issue and update [core](https://github.com/mobizt/Firebase-ESP8266/commit/4631324ed01564851eebe38bd452c25aab0ad2ea)   and  [its header](https://github.com/mobizt/Firebase-ESP8266/commit/1955b9bac74a3ef0a3745402d0733be02f571153) few monent ago but not change the version number)

AMAN-KHATRI, 2019-04-24:
Thank you so much. Now the library is working perfectly fine. One more thing, when I update the firebase using setInt or updateNode, it takes about 2 seconds to reflect changes, and at that particular time no other code is functional, everything just stops working. So, is there any faster and more efficient way to do so?
One more thing, can I use ping now in between the code?

mobizt, 2019-04-24:
It depends on size of json data passed to updateNodeSilent, time for WiFi client to start new SSL connection after disconnected from stream because you use the same Firebase Data object.

Another way is use difference Firebase Data object for stream and for normal operations (get/set/push/update/delete) as in examples. But beawre each Firebase Data Object takes about 18-20k memory.

affibox, 2019-04-24:
> ESP8266 Arduino Core

Like I said... it doesn't work when I try using 
` if (!Firebase.readStream(firebaseData))
  {
    //If read stream was failed, print the error detail.
    Serial.println(firebaseData.errorReason());
  }`

The output is this:
`
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687

Connecting to MOTOROLA-7B4AC
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 9
cnt 

connected with MOTOROLA-7B4AC, channel 6
dhcp client start...
wifi evt: 0
.....ip:192.168.0.12,mask:255.255.255.0,gw:192.168.0.1
wifi evt: 3
.
WiFi connected
IP address: 
192.168.0.12
connection refused
pm open,type:2 0
connection refused
connection refused
`

affibox, 2019-04-24:
@mobizt 

Writing and reading one time works but I need to get updates from the database to the module ro my project won't be useful.

affibox, 2019-04-24:
@mobizt thanks...!!!

AMAN-KHATRI, 2019-04-24:
Thanks, @mobizt. I'll give it a try.

closed 2019-04-24 by AMAN-KHATRI