mobizt/Firebase-ESP8266 #8: value is "Bool" but detected as "Int" when streaming

maisonsmd, 2019-04-17:
I have data structure like this: {"test": true} in database.
When I stream path "/test", it return value type is "int" instead of "bool", and the value is always 0.

I tried the code like this:
```C++
if (streamData.streamAvailable()) {
		Serial.println(F("-------Stream Data available-------"));
		Serial.println("STREAM PATH: " + streamData.streamPath());
		Serial.println("PATH: " + streamData.dataPath());
		Serial.println("TYPE: " + streamData.dataType());
		Serial.print("VALUE: ");
		if (streamData.dataType() == "bool")
			Serial.println(streamData.boolData());
		else if (streamData.dataType() == "int")
			Serial.println(streamData.intData());
		else if (streamData.dataType() == "float")
			Serial.println(streamData.floatData());
		else if (streamData.dataType() == "boolean")
			Serial.println(streamData.boolData());
		else if (streamData.dataType() == "string")
			Serial.println(streamData.stringData());
		else if (streamData.dataType() == "json")
			Serial.println(streamData.jsonData());
		Serial.println();
	}
```
the log goes like this:
```
-------Stream Data available-------
STREAM PATH: /test
PATH: /
TYPE: int
VALUE: 0
```
It triggers whenever I change data in database, so I think it cannot distinguish int from bool.

mobizt, 2019-04-17:
Boolean is not support by Firebase. Only integer, float, string and json object are supported data types.

My library was implemented boolean by store its value (via setBool, pushBool) as string with prefixed **bool,** i.e. **"bool,0"** for false and **"bool,1"**  for true.  And this special prefixed string data will be converted to boolean when read with getBool function.

When you store data as JSON via push or update, Firebase will evaluate boolean value as its support data type, may be integer or string.

If you want to work with JSON data and your data contains boolean value, you need to handle boolean yourself as Firebase not support it.

maisonsmd, 2019-04-17:
So if I write something like:
```C++
FirebaseData fb;
...
Firebase.getBool(fb, "/test");
Serial.println(fb.boolData();
```
it won't work neither?

mobizt, 2019-04-17:
You need to use setBool or pushBool to store Boolean value then you can get it back by getBool because this is special implementation by this client only.

Or you can use integer 1 for true and 0 for false and manage it by your self.

At console, you can see what is exact data store when you call setBool or pushBool

maisonsmd, 2019-04-17:
Ok now I understand, with setBool() the data in database is a String, with "bool" prefix.
I will use Int instead. 
Thank you!

closed 2019-04-17 by maisonsmd

mobizt, 2019-04-19:
Hi maisonsmd,

I would like to inform you that Boolean is support by Firebase as it is also a valid JSON value which I misunderstand. 

Now this issue was fixed in version 1.0.5 which you can store actual Boolean value instead of prefixed string in previous version.