mobizt/Firebase-ESP8266 #9: Reset issues

iotproductions, 2019-04-20:
Hi,
I just try to use the Basic example sketch.
After compile the sketch and program to my board successfully.  
My board can only run the SetInt function success once time, after my board reset repeatedly.
Please let me know what is wrong with the Basic example ?
Below is my logs:
```
Connecting to Wi-Fi.
17:21:20.858 -> Connected with IP: 192.168.1.9
17:21:20.858 -> 
17:21:20.858 -> -----------------------------------
17:21:20.858 -> ----------Begin Set Test-----------
17:21:20.858 -> -----------------------------------
17:21:20.858 -> 

17:21:22.150 -> Exception (3):
17:21:22.150 -> epc1=0x4020f167 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024cce0 depc=0x00000000
17:21:22.193 -> 
17:21:22.193 -> ctx: cont 
17:21:22.193 -> sp: 3ffffc10 end: 3fffffd0 offset: 01a0
17:21:22.193 -> 
17:21:22.193 -> >>>stack>>>
17:21:22.193 -> 3ffffdb0:  3ffffe08 3ffef710 00000160 3fff0030  
17:21:22.193 -> 3ffffdc0:  3fff34f4 00000002 3ffef6bc 4021489a  
17:21:22.193 -> 3ffffdd0:  00000000 3fff0024 3fff1224 3ffef73c  
17:21:22.193 -> 3ffffde0:  3fff46cc 00000030 00000020 401004e4  
17:21:22.193 -> 3ffffdf0:  3fff0c7c 00000030 00000020 4020c0e4  
17:21:22.193 -> 3ffffe00:  3fff34f4 3fff46cc 3ffef680 4020489c  
17:21:22.193 -> 3ffffe10:  3fff0c7c 00000030 3ffef680 3fff46cc  
17:21:22.236 -> 3ffffe20:  3fff0c7c 00000030 3ffef680 40206a17  
17:21:22.236 -> 3ffffe30:  4020cb50 00000000 00003a98 40205ec1  
17:21:22.236 -> 3ffffe40:  3ffef710 3fff10ec 3fff1194 3fff093c  
17:21:22.236 -> 3ffffe50:  fffffffa 3fff08d0 3fff0030 401006dc  
17:21:22.236 -> 3ffffe60:  3fff34f4 3ffef73c 00000001 00000000  
17:21:22.266 -> 3ffffe70:  3fff0c3c 00000002 3ffef6bc 00000000  
17:21:22.266 -> 3ffffe80:  3ffef6c8 0000000a 00000000 00000000  
17:21:22.266 -> 3ffffe90:  00000000 00000001 0000208b 00000001  
17:21:22.266 -> 3ffffea0:  00000031 6e6d6c6b 00000002 3ffef680  
17:21:22.266 -> 3ffffeb0:  fffffffa 3ffef73c 3ffef6b4 3ffef680  
17:21:22.266 -> 3ffffec0:  fffffffa 3ffef73c 3ffef6b4 4020726c  
17:21:22.298 -> 3ffffed0:  00000000 3ffef710 3fffff14 3fffff10  
17:21:22.298 -> 3ffffee0:  00000000 3fffff10 3fff083c 40214d46  
17:21:22.298 -> 3ffffef0:  3fffff18 00000001 3fff083c 3fffff50  
17:21:22.298 -> 3fffff00:  3ffef73c 3ffef680 3fff083c 40207427  
17:21:22.332 -> 3fffff10:  3fff08f8 3fff08d0 3fffff50 4020bfda  
17:21:22.332 -> 3fffff20:  0000000a 3ffef73c 3fffff50 4020c02a  
17:21:22.332 -> 3fffff30:  3ffe8b7c 00ffffff 3fffff50 3ffe8b34  
17:21:22.332 -> 3fffff40:  3ffef680 3ffef680 3ffefa58 40202bbd  
17:21:22.332 -> 3fffff50:  3fff0814 0000001f 00000017 00000000  
17:21:22.332 -> 3fffff60:  00000000 00000000 feefeffe feefeffe  
17:21:22.366 -> 3fffff70:  feefeffe 3fff0abc 0000000f 00000000  
17:21:22.366 -> 3fffff80:  3fff0aa4 0000000f 0000000d feefeffe  
17:21:22.366 -> 3fffff90:  00000001 feefeffe feefeffe feefeffe  
17:21:22.366 -> 3fffffa0:  feefeffe feefeffe feefeffe 3ffefb18  
17:21:22.399 -> 3fffffb0:  3fffdad0 00000000 3ffefb10 4020c2a8  
17:21:22.399 -> 3fffffc0:  feefeffe feefeffe 3ffe8624 40100739  
17:21:22.399 -> <<<stack<<<
17:21:22.399 -> 
17:21:22.399 ->  ets Jan  8 2013,rst cause:2, boot mode:(1,6)
17:21:22.433 -> 

17:21:30.534 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,6)
17:21:30.534 -> 
17:21:30.534 -> wdt reset
```

mobizt, 2019-04-20:
Hi,
Please update ESP8266 Arduino Core to v 2.5.0.

From Arduino IDE, go to menu **Tools** > **Board: "xxxxx"** > B**oards Manager...**, search for **esp8266**, click **Update** button or choose version **2.5.0** and click **Install**.

If errors still exist, please let me know.

iotproductions, 2019-04-20:
> Hi,
> Please update ESP8266 Arduino Core to v 2.5.0.
> 
> From Arduino IDE, go to menu **Tools** > **Board: "xxxxx"** > B**oards Manager...**, search for **esp8266**, click **Update** button or choose version **2.5.0** and click **Install**.
> 
> If errors still exist, please let me know.

Hi,
I just updated the Arduino Core to 2.5.0 and it working well. 
My problem has been solved.
Thank you so much !

closed 2019-04-20 by iotproductions