mobizt/Firebase-ESP8266 #11: getInt returning  incorrect values

jibin2706, 2019-04-21:
I tried retrieving integer values using 
getInt but it is returning 1 no matter what the value is in the database.

long int test2 = Firebase.getInt(firebaseData, "/parking/1000/spot1/validTill");

mobizt, 2019-04-21:
My library has difference API interfaces from other Firebase libraries or from FirebaseExtended user. 
Please see examples in examples folder and read me page.

You need to modify your existing code to match this library's API.

mobizt, 2019-04-21:
You can start from here https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino

jibin2706, 2019-04-21:
> You can start from here https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino
Solved
Thanks a lot for this awesome library

closed 2019-04-22 by mobizt