mobizt/Firebase-ESP8266 #13: Booleans are handled as strings

ratnick, 2019-04-22:
Hi,

I like this library, and I switched to it from another Arduino library a few days ago mainly because the error handling is way better.
However, this library seems to save booleans as strings in Firebase. Is there any way to avoid or change that?
The problem is that other applications (such as Android SDK, which I use) use real booleans, and this obviously is incompatible (unless handled specially). 
I am no Firebase expert, but I think it would be wise to use the method provided by Firebase: https://firebase.google.com/docs/firestore/manage-data/data-types 

Meanwhile, I will make a work-around somehow, but as far as I can tell in your library, it's not straight forward to modify your central function FirebaseESP8266::sendRequest() since it already in the parameters takes a string as input.

Anyway - I already modified your getBool to make it possible to read boolean values. Not the most elegant and bullit-proof way (but it does read "real" booleans off Firebase):

bool FirebaseESP8266::getBool(FirebaseData &dataObj, const String &path)
{
    dataObj.queryFilter.clear();
    bool flag = sendRequest(dataObj, path.c_str(), FirebaseMethod::GET, FirebaseDataType::BOOLEAN, "");
	if (flag && dataObj._dataType == FirebaseDataType::BOOLEAN) {
		flag = true;
	} else if (flag && dataObj._dataType == FirebaseDataType::INTEGER) {
		flag = true;
	}
	else {
		flag = false;
	}
	return flag;
}

mobizt, 2019-04-22:
Sorry for this issue due to my misunderstanding at the first time I implement it.
Now the boolean issue was solved in version 1.0.5 which you can update.

ratnick, 2019-04-22:
Seems to work in 1.0.5 - thanks for quick reply!!!

closed 2019-04-22 by ratnick