mobizt/Firebase-ESP8266 #15: Shows esp8266 core outdated, even though I have 2.4.2 installed

nikrath, 2019-04-22:
According to the prerequisites for the library, esp8266 v2.4.0 or above must be installed. Recommended versions are 2.4.2 or 2.5.0.
I have 2.4.2 version installed. But compiling the code gives the error:

```
Arduino: 1.8.6 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Multiple libraries were found for "SD.h"
In file included from C:\Users\nikhil\Desktop\PHOTON\PHOTON\PHOTON.ino:13:0:

 Used: C:\Users\nikhil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\SD
C:\Users\nikhil\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h:58:2: error: #error Your ESP8266 Arduino Core SDK is outdated, please update. From Arduino IDE go to Boards Manager and search 'esp8266' then select version 2.4.0 or above.

 Not used: D:\Arduino\libraries\SD
 #error Your ESP8266 Arduino Core SDK is outdated, please update. From Arduino IDE go to Boards Manager and search 'esp8266' then select version 2.4.0 or above.

  ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

What is it that I am doing wrong?
Any help appreciated.

mobizt, 2019-04-22:
How about SD library at D:\Arduino\libraries\SD?
ESP8266 Arduino core lib has SD library too.

You need to remove or delete another SD library at SD library at D:\Arduino\libraries\SD

nikrath, 2019-04-22:
@mobizt Thanks for the quick response. 
Tried that already, still get the following error

```
Arduino: 1.8.6 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\nikhil\Desktop\CODES\CODES\CODES.ino:13:0:

C:\Users\nikhil\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h:58:2: error: #error Your ESP8266 Arduino Core SDK is outdated, please update. From Arduino IDE go to Boards Manager and search 'esp8266' then select version 2.4.0 or above.

 #error Your ESP8266 Arduino Core SDK is outdated, please update. From Arduino IDE go to Boards Manager and search 'esp8266' then select version 2.4.0 or above.

  ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

mobizt, 2019-04-22:
You need to install ESP8266 Arduino Core SDK from Boards Manager. 
Your core version may not released git version.

These are released git version that this library supported and also available in Boards Manager
2.5.0, 2.5.0-beta3, 2.5.0-beta2, 2.5.0-beta1, 2.4.2, 2.4.1, 2.4.0, 2.4.0-rc2, 2.4.0-rc1

nikrath, 2019-04-22:
I already have version 2.4.2, installed through the Boards Manager

mobizt, 2019-04-23:
I just fixed the issue and update.

Please delete the current version from library folder.
Down load the new one of v 1.0.5.

I don't change the version number due to I'm working on the new version.

nikrath, 2019-04-24:
@mobizt I just did that and it works perfectly. Thanks!

closed 2019-04-24 by mobizt