mobizt/Firebase-ESP8266 #16: SD Card mount failed

Ishara-Lakshan, 2019-04-24:
I mounted my SD card module as follows
Gnd > gnd
Vcc > vcc
Cs > D2
Sck > D5
Mosi > D6
Miso > D7
When i uplaod the code and open the serial monitor an error message appears saying "SD Card mounted failed". Is there anything wrong in my connection with SD card module and NodeMcu? 
Thank you.

mobizt, 2019-04-24:
Here is the default configuration

NodeMCU > SD

D5---------> SCK
D6---------> MISO
D7---------> MOSI
D8---------> CS

closed 2019-04-26 by mobizt

Ishara-Lakshan, 2019-05-05:
Tried connecting this way...but still the error message appears.

mobizt, 2019-05-05:
My library use core ESP8266 SPI library.
This may be SD card, SD module and wrong connection problems.

Try basic example in Arduino IDE  to test read or write simple text file to SD card.

Ishara-Lakshan, 2019-05-05:
Ok I'll try it. Thank you

mobizt, 2019-05-05:
Or try to include SPI.h in your sketch.

Ishara-Lakshan, 2019-05-05:
SPI.h is already included