mobizt/Firebase-ESP8266 #18: Could not parse auth token

LarryReader, 2019-05-12:
Hello
Using the Basic example in PlatformIO 
I think I have Firebase setup correctly 
Serial output
------------------------------------
Set double test...
FAILED
REASON: unauthorized, Could not parse auth token.
------------------------------------

Thanks for any help

mobizt, 2019-05-12:
Please make sure you provide valid database secret for **FIREBASE_AUTH** and valid Project ID for **FIREBASE_HOST**
```C++
#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com"
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"
```
[image]

LarryReader, 2019-05-12:
Thanks for the quick reply. I will try that later. Really love the potential for this library. Are there plans for supporting a non deprecated authentication method? Or suggestions for dealing with that?

mobizt, 2019-05-12:
Yes I also try to use access token in my library instead.

LarryReader, 2019-05-12:
Whoo Hooo !
The Database Secret works!
For anyone else this may help the database secret is found at:
Firebase Console > Project Overview > Settings > Service Accounts >
Database Secrets
Thanks

closed 2019-05-13 by mobizt