mobizt/Firebase-ESP8266 #19: Not getting data from firebase

tiagosk81989, 2019-05-13:
Hello,
I'm experiencing problems getting data from firebase database, to change state of relays.
No problem setting data on database, only getting. All other functions work well.
This code work well with other firebase library.
Can anyone help me?

#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include "DHT.h"

// Set these to run example.
#define FIREBASE_HOST "..."
#define FIREBASE_AUTH "..."
#define WIFI_SSID "..."
#define WIFI_PASSWORD "..."

FirebaseData firebaseData;

#define LAMP_PIN D1
#define LAMP_PIN2 D2
#define LAMP_PIN3 D3
#define LAMP_PIN4 D4
#define DHT_PIN D5
#define PRESENCE_PIN D6
#define LDR_PIN D7
#define DHTTYPE DHT11

#define PUBLISH_INTERVAL 1000*60*5

DHT dht(DHT_PIN, DHTTYPE);
Ticker ticker;
bool publishNewState = true;

void publish() {
  publishNewState = true;
}

void setupPins() {
  
  pinMode(LAMP_PIN, OUTPUT);
  digitalWrite(LAMP_PIN, HIGH);
  
  pinMode(LAMP_PIN2, OUTPUT);
  digitalWrite(LAMP_PIN2, HIGH);
  
  pinMode(LAMP_PIN3, OUTPUT);
  digitalWrite(LAMP_PIN3, HIGH);

  pinMode(LAMP_PIN4, OUTPUT);
  digitalWrite(LAMP_PIN4, HIGH);
  
  pinMode(PRESENCE_PIN, INPUT);

  pinMode(LDR_PIN, INPUT);

  dht.begin();
}

void setupWifi() {
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("A ligar");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("ligado: ");
  Serial.println(WiFi.localIP());
  digitalWrite(LAMP_PIN, LOW);
  delay(300);
  digitalWrite(LAMP_PIN, HIGH);
}

void setupFirebase() {

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.setBool(firebaseData, "lampada", false);
  Firebase.setBool(firebaseData, "lampada2", "false");
  Firebase.setBool(firebaseData, "lampada3", false);
  Firebase.setBool(firebaseData, "lampada4", false);
  Firebase.setBool(firebaseData, "presenca", false);
  Firebase.setBool(firebaseData, "luminosidade", false);
  Firebase.reconnectWiFi(true);
  Firebase.setMaxRetry(firebaseData, 5);
}

void setup() {
  Serial.begin(9600);

  setupPins();
  setupWifi();
  setupFirebase();

  ticker.attach_ms(PUBLISH_INTERVAL, publish);
}

void firebasereconnect() {
  Serial.println("a reconectar");
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
}

void loop() {

  if (publishNewState) {
    Serial.println("A publicar novo estado");
    // Obtem os dados do sensor DHT
    float humidity = dht.readHumidity();
    float temperature = dht.readTemperature();
   
 if (!isnan(humidity) && !isnan(temperature)) {
      Firebase.pushFloat(firebaseData, "temperatura", temperature);
      Firebase.pushFloat(firebaseData, "humidade", humidity);
      publishNewState = false;
    } else {
      Serial.println("Erro ao publicar");
    }
  }

  int presence = digitalRead(PRESENCE_PIN);
  Firebase.setBool(firebaseData, "presenca", presence == HIGH);

  bool lampValue2 = Firebase.getBool(firebaseData, "/lampada2");
  digitalWrite(LAMP_PIN2, lampValue2 ? LOW : HIGH);
  
  bool lampValue3 = Firebase.getBool(firebaseData, "/lampada3");
  digitalWrite(LAMP_PIN3, lampValue3 ? LOW : HIGH);

  bool lampValue4 = Firebase.getBool(firebaseData, "/lampada4");
  digitalWrite(LAMP_PIN4, lampValue4 ? LOW : HIGH);

  int ldrValor = digitalRead(LDR_PIN);
  if (ldrValor == HIGH) {
    digitalWrite(LAMP_PIN, LOW);
    Firebase.setBool(firebaseData, "lampada", HIGH);
    Firebase.setBool(firebaseData, "luminosidade", LOW); 
  }
  if (ldrValor == LOW) {
    digitalWrite(LAMP_PIN, HIGH);
    Firebase.setBool(firebaseData, "lampada", LOW); 
    Firebase.setBool(firebaseData, "luminosidade", HIGH);
  }
  
  delay(200);
}

mobizt, 2019-05-13:
Did you see the [Basic.ino](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino)  example? 

This Firebase ESP8266 library APIs are different from other Arduino Firebase library (s).

To get the value from get functions, you need to call  one of following functions with Firebase Data object which depending on the data type returned from server which can check from `firebaseData.dataType()`

```C++
int myInt = firebaseData.intData();
float myFloat = firebaseData.floatData();
double myDouble = firebaseData.doubletData();
bool myBool = firebaseData.boolData();
String myString = firebaseData.stringData();
...

```

**From part of your code below, it should change from**

```C++

 bool lampValue2 = Firebase.getBool(firebaseData, "/lampada2");

```
**to**

```C++

Firebase.getBool(firebaseData, "/lampada2");
bool lampValue2 = firebaseData.boolData();
```

tiagosk81989, 2019-05-13:
thank you for your prompt reply mobizt!
And now everything works well. 
I previously checked all your sample codes, but didn't understand the dynamics, and how to adapt my code to your library. I'm only a enthusiast.
Thank you again!
Problem solved

closed 2019-05-14 by mobizt