mobizt/Firebase-ESP8266 #20: Streaming + Firebase.sendMessage

UltimatPronin, 2019-05-14:
Is there a way to monitoring data changes with Streaming and paralelly send FCM messages?
IF yes, please provide simple example how to implement this.

Now if I use same firebaseData object - after 1-st FCM message I have message about connection refused, and can't receive stream a lot of time, but still can send FMC messages. But if I use one firebaseData object for Streaming and other firebaseData object fom FMC messages - I get STACK error after 1-st FMC send try.

Is there workaround or I need to use the other way?

mobizt, 2019-05-15:
Please post your sketch here that I will try and investigate the problem.

UltimatPronin, 2019-05-16:
Here it is:
`#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "XXXXXX.firebaseio.com"
#define FIREBASE_AUTH "YYYYYYYYY"
#define FIREBASE_FCM_SERVER_KEY "ZZZZZZZZZZZZZZZZ"
#define FIREBASE_FCM_DEVICE_TOKEN_1 "XXXXXXXXXXXXXXX"
#define WIFI_SSID "RRRRRRRRRRRRRRR"
#define WIFI_PASSWORD "PPPPPPPPPPPPP"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;

unsigned long sendDataPrevMillis = 0;
unsigned long sendMSGPrevMillis = 0;

String path = "/Devices";

uint16_t count = 0;

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  firebaseData.fcm.begin(FIREBASE_FCM_SERVER_KEY);
  firebaseData.fcm.addDeviceToken(FIREBASE_FCM_DEVICE_TOKEN_1);
  //firebaseData.fcm.addDeviceToken(FIREBASE_FCM_DEVICE_TOKEN_2);
  firebaseData.fcm.setPriority("high");
  firebaseData.fcm.setTimeToLive(1000);

  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
  //This example uses the same Firebase Data object to read/store data (get, set, update, push and delete) and stream.
  //This causes some delay (for start new SSL connection) for swiching between read/store and stream operation.
  //For no delay, see Different_objects_stream.ino example which uses different Firebase Data object for read/store and stream data.

  if (millis() - sendMSGPrevMillis > 10000){
    sendMSGPrevMillis = millis();

    firebaseData.fcm.setNotifyMessage("Notification"+String(random(0xffff)), "Hello World! " + String(random(0xffff)));

    if (Firebase.sendMessage(firebaseData, 0))//send message to recipient index 0
    {
[... 121 lines cut ...]
3ffffee0:  3ffe895f 3ffeed30 ffffffff 40202294  
3ffffef0:  3fffff10 3ffeedc8 3fffff30 40210658  
3fffff00:  3ffe8765 3fffff50 3ffeec00 40204a2a  
3fffff10:  3ffef38c 3ffeed30 3ffeedc8 4020ffbc  
3fffff20:  3ffe8765 3ffeedc8 3ffeec00 40201885  
3fffff30:  53414552 203a4e4f 00203a00 402255e2  
3fffff40:  3fffff64 3ffeec00 3fffff80 3ffeee38  
3fffff50:  00000000 00000000 fffeec00 00000000  
3fffff60:  00000000 fffeec00 00000000 00000000  
3fffff70:  fffeed30 3ffeec00 3ffeedc8 402011cd  
3fffff80:  00000000 00000000 ffefef00 00000000  
3fffff90:  00000000 00000000 00000001 3ffeee38  
3fffffa0:  3fffdad0 00000000 3ffeee04 40210cd4  
3fffffb0:  feefeffe feefeffe 3ffe854c 401004e9  
<<<stack<<<

Decoding stack results
0x40202294: FirebaseESP8266::errorToString(int, std::string&) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3612
0x40210658: String::copy(char const*, unsigned int) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 211
0x40204a2a: FirebaseData::errorReason() at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4847
0x4020ffbc: Print::println(char const*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Print.cpp line 190
0x40201885: loop() at D:\Profile\User\Documents\Arduino\Tests\Stream_FCM/Stream_FCM.ino line 167
0x402255e2: std::basic_string  , std::allocator  >::~basic_string() at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 541
0x402011cd: setup() at D:\Profile\User\Documents\Arduino\Tests\Stream_FCM/Stream_FCM.ino line 61
0x40210cd4: loop_wrapper() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_main.cpp line 125

mobizt, 2019-05-16:
If two issues you open are the same can be close one?

UltimatPronin, 2019-05-16:
Really, I think issues are similar. You can close one of them.

mobizt, 2019-05-16:
Multiple .ino files may lead to problems. 
Try to declare global variables in one header file (.h file on any tab). 
Then include this header file at the top of main sketch but not sure it help or not.

For big project and library developt I switch to Platformio on VS Code.

UltimatPronin, 2019-05-16:
Thank you for advice.

mobizt, 2019-05-16:
As I test your posted sketch. No error for stack.

For stream, when FCM need to send new message,  it will stop the stream connection and start new http connection to different uri due to they use the same WiFi client in Firebase Data object, and the FCM response from server is a bit slower than RTDB connection.  During this time stream can't be connected and after FCM message complete it work, stream will be resumed and use time to re-connected to the stream path, you may miss stream data during this as normal behavior of multi-tasking with only one WiFi client.

closed 2019-05-16 by mobizt