mobizt/Firebase-ESP8266 #21: Core:2.5.1 STACK Error On firebaseData.errorReason()

UltimatPronin, 2019-05-14:
On Core:2.5.1 I get STACK Error On firebaseData.errorReason().

Please check this issue.

mobizt, 2019-05-15:
Please post your sketch here that I will try and investigate the problem.

UltimatPronin, 2019-05-16:
Real sketck is big. 
Decoding stack results
0x40242270: alloc at crypto/bigint.c line 1113
0x402235e1: optimistic_yield(uint32_t) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 115
0x402430a7: bi_barrett at crypto/bigint.c line 1289
0x40243386: bi_mod_power at crypto/bigint.c line 1323
0x40242100: trim at crypto/bigint.c line 1197
0x40244529: RSA_encrypt at crypto/rsa.c line 254
0x4023f030: send_client_key_xchg at ssl/tls1_clnt.c line 435
0x4023f529: do_clnt_handshake at ssl/tls1_clnt.c line 123
0x4023c6b0: mem_free at core/mem.c line 237
0x4023732e: pbuf_free_LWIP2 at core/pbuf.c line 786
0x40211672: ClientContext::read(char*, unsigned int) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/ClientContext.h line 553
0x4022359e: __yield() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 100
0x4023ee38: basic_read at ssl/tls1.c line 1607
0x4023efa0: ssl_read at ssl/tls1.c line 314
0x40211d80: axTLS::SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/SSLContext.h line 137
0x40231dd8: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x40211dbc: std::__shared_ptr ::operator=(std::__shared_ptr &&) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/shared_ptr_base.h line 862
0x4021202f: axTLS::WiFiClientSecure::_connectSSL(char const*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/SSLContext.h line 341
0x402120be: axTLS::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\WiFiClientSecureAxTLS.cpp line 117
0x4020fbd8: FirebaseHTTPClient::http_connected() at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266HTTPClient.cpp line 49
0x4020fc53: FirebaseHTTPClient::http_connect() at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266HTTPClient.cpp line 81
0x4020fc8c: FirebaseHTTPClient::http_sendRequest(char const*, char const*) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266HTTPClient.cpp line 63
0x40231dd8: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x40210414: ESP8266WiFiSTAClass::status() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 563
0x4020bb14: FCMObject::fcm_send(FirebaseHTTPClient&, int&, unsigned char) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 5499
0x4020b430: FCMObject::fcm_connect(FirebaseHTTPClient&) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 5212
0x4020bcaa: FirebaseESP8266::sendFCMMessage(FirebaseData&, unsigned char) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3805
0x40101778: free at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc\umm_malloc.c line 1760
0x4020bd13: FirebaseESP8266::sendMessage(FirebaseData&, unsigned short) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3813
0x40207ed5: DSCm_FB_sendMessage() at D:\Profile\User\Documents\Arduino\DSC_Module\DSC_Module/Google_FireBase.ino line 83
0x4020819c: loop() at D:\Profile\User\Documents\Arduino\DSC_Module\DSC_Module/DSC_Module.ino line 99
0x4022356c: loop_wrapper() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 125

UltimatPronin, 2019-05-16:
Decoding stack results
0x402042f0: FirebaseESP8266::errorToString(int, std::string&) at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3612
0x4021ecdd: Print::write(char const*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/Print.h line 60
0x402067e2: FirebaseData::errorReason() at D:\Profile\User\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4847
0x4021edb4: Print::println(char const*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Print.cpp line 190
0x4020160a: loop() at D:\Profile\User\Documents\Arduino\DSC_Module\DSC_Module/DSC_Module.ino line 110
0x4022089c: loop_wrapper() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_main.cpp line 125

mobizt, 2019-05-16:
Ok let me investigate.

UltimatPronin, 2019-05-16:
I determinate, if I copy all my sketch parts from many tabs in Arduino IDE to 1 big sketch tab - all warking VERY GOOD. But If I split my sketck by many tabs - then I get ERRORS. Any Ideas? It is IDE BUG?

mobizt, 2019-05-16:
Look like something is calling external function inside while loop and that function hold something in stack which make the stack growth until it  causes the corrupted the heap.

This issue is due to design of your program.

UltimatPronin, 2019-05-16:
OK, I will try redesign it. Thank you!

closed 2019-05-16 by mobizt

UltimatPronin, 2019-05-17:
I find an issue.

Please see on it. There are sketch example:

#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "XXX.firebaseio.com"
#define FIREBASE_AUTH "XXX"
#define WIFI_SSID "XXX"
#define WIFI_PASSWORD "XXX"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;

unsigned long sendDataPrevMillis = 0;

String path = "/Devices";

uint16_t count = 0;

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: ");
    if (firebaseData.dataType() == "int")
      Serial.println(firebaseData.intData());
    else if (firebaseData.dataType() == "float")
      Serial.println(firebaseData.floatData());
    else if (firebaseData.dataType() == "boolean")
      Serial.println(firebaseData.boolData() == 1 ? "true" : "false");
    else if (firebaseData.dataType() == "string")
      Serial.println(firebaseData.stringData());
    else if (firebaseData.dataType() == "json")
      Serial.println(firebaseData.jsonData());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

Please run it and wait for error. In my case it is HTTPC_ERROR_NOT_CONNECTED
When I get it, then STACK Error reboot my nodemcu. It hapenn only on 2.5.1., on 2.5.0 all OK.

Error is on  
void FirebaseESP8266::errorToString(int httpCode, std::string &buff)

case HTTPC_ERROR_NOT_CONNECTED:
        p_memCopy(buff, ESP8266_FIREBASE_STR_42);
        return;

IF I change it (for test) to
case HTTPC_ERROR_NOT_CONNECTED:
        //p_memCopy(buff, ESP8266_FIREBASE_STR_42);
		p_memCopy(buff, ESP8266_FIREBASE_STR_39);
        return;

THERE ARE NO MORE TROUBLES AND STACH ERRORs for me.

mobizt, 2019-05-17:
That's not cause or avoid error and not related to stack error because it's give the same result. I also test your posted sketch long enough on esp core 2.5.1 but there's nothing error.

If you still get error from something like this, it may depend on your hardware and wifi situation.

UltimatPronin, 2019-05-17:
Thank you, for invitations.