mobizt/Firebase-ESP8266 #22: ets Jan  8 2013,rst cause:4, boot mode:(1,6)  wdt reset

arushgupta2007, 2019-05-16:
Hi,

I am trying to control my tubelight/plug with my esp8266. I am storing my status and other data in firebase  realtime database.
I want to stream the data because I know if my data has changed then it means I have to change the state of my relay.

Here is my Serial error:

---------------------------------------------

`ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
`

---------------------------------------------

And my code :

---------------------------------------------

`
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

#define FIREBASE_HOST "____"
#define FIREBASE_AUTH "____"
#define WIFI_SSID "____"
#define WIFI_PASSWORD "____"
#define relayswitch 7

String path = "";

FirebaseData firebaseData;
DynamicJsonBuffer json_buffer;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  pinMode(relayswitch, OUTPUT);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  QueryFilter query;
  query.equalTo("esp.2615");
  query.limitToLast(1);

  if (Firebase.getJSON(firebaseData, "/arushpass/cool/chill", query)) {
    Serial.println(firebaseData.dataPath());
    path = firebaseData.dataPath();
  } else {
    Serial.println(firebaseData.errorReason());
  }
  query.clear();
    
  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop() {
  Serial.println("asddhhfkslfghlksfa");
  // put your main code here, to run repeatedly:
  if (!Firebase.readStream(firebaseData)) {
    Serial.println("------------------------------------");
    Serial.println(firebaseData.errorReason());
    Serial.println("------------------------------------");
  }
  
  if (firebaseData.streamTimeout()) {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  
  if (firebaseData.streamAvailable()){
    JsonObject& json = json_buffer.parseObject(firebaseData.jsonData()); 
    String type = json["type"];
    if (type.equals("TubeLight") || type.equals("Plug")) {
      digitalWrite(relayswitch, !digitalRead(relayswitch));
    }
  }
}
`

---------------------------------------------

My database structure is :
    
    database:
        user_key:
            room1:
                device_name:
                      id: esp.2615
                      on_or_off: false
           room2:
                device_name:
                      id: esp.3526
                      on_or_off: false

etc

Any clue of what I am doing wrong?

arushgupta2007, 2019-05-16:
I have tried updating esp8266 Arduino core to 2.5.0

mobizt, 2019-05-16:
From this
```C++
#define relayswitch 7
```
You can't use GPIO7 because it uses as MISO of SPI bus that connect with Flash memory. Your device can't run and program because no data to/from flash memory.

For query, **you must and always assign orderBy** as states in Firebase document, like this.
 ```C++
QueryFilter query;
  query.orderBy("id");
  query.equalTo("esp.2615");
  query.limitToLast(1);
```
You may get error like this  return from server.
**connection refused, Index not defined, add \".indexOn\": \"id\", for path \"/arushpass/cool/chill\", to the rules**

You also set the database rules like this by use wild card $ to any dynamic child keys of user_key which can be room1, room2,... and their child key $device. The name that use with $ can by any what you want.
```C++
 "rules": {
    ...
    ...
    ,
    "user_key":{
      "$room":{
           "$device":{
                ".indexOn":"id"
           } 
       }
    }
  }
```

or use only one wild card  $ to refer to any child keys under user_key included  room1, room2, device_name and so on.

```C++
 "rules": {
    ...
    ...
    ,
    "user_key":{
        "$device":{
             ".indexOn":"id"
       }
    }
  }
```

arushgupta2007, 2019-05-16:
Thanks for the quick reply

> You also set the database rules like this by use wild card $ to any dynamic child keys of user_key which can be room1, room2,... and their child key $device. The name that use with $ can by any what [...]

My user_key, device, room can be changed or deleted so do I have to edit the rules every time it  is deleted or updated ?

mobizt, 2019-05-16:
Yes but it better to have permanent parent key.

arushgupta2007, 2019-05-16:
Oh

So can I set my rules like this:

`"rules" : {
    ....
    ....
    ....
    ,
    ".indexOn": "id"
}`

mobizt, 2019-05-16:
This means id is the parent key at the root of database and sort or select is no effect, indexOn should set to childs.

arushgupta2007, 2019-05-16:
Thanks

One more question:

Is it normal that when I am streaming I cannot execute Serial.println()

mobizt, 2019-05-16:
Don't understand. But don't serial print continuously like that, mcu need idle time to feed the watchdog timer or handle WiFi task too, if you do that wdt reset may occur (only when connected to PC).

arushgupta2007, 2019-05-16:
Oh thanks

BTW can I use GPIO4. In the pinout it tells GPIO4 is not used

mobizt, 2019-05-16:
Yes, you can see any esp8266 dev board like this what the pin you can use
https://wiki.wemos.cc/products:d1:d1_mini
Some can be multifunction as basic GPIO and communication port as SPI or I2C.

arushgupta2007, 2019-05-16:
Thanks man
I'll try it now and tell you

arushgupta2007, 2019-05-16:
I am getting:

------------------------------------
not connected
------------------------------------
Stream timeout, resume streaming...

and after I edit my id the relay does not change it's state

mobizt, 2019-05-16:
Your internet problem or something.  Begin stream may fail in setup. What's about code in loop?

arushgupta2007, 2019-05-16:
Did not understand, did you ask code in loop?
I have not changed it a bit?

mobizt, 2019-05-16:
Try the examples 
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Stream/Stream.ino
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Different_objects_stream/Different_objects_stream.ino
It should work, if any failed, it's your internet or AP problem.

closed 2019-05-16 by mobizt