mobizt/Firebase-ESP8266 #25: No device token provided on Firebase.sendTopic

arushgupta2007, 2019-05-19:
Hi,

I am trying to use cloud messaging in esp8266 to give notification to android phone
My android phone has subscribed to the topic "general"

My esp8266 code :  

`

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);

  Firebase.reconnectWiFi(true);

  firebaseData.fcm.begin("SECRET_KEY_IS_COPIED");

  firebaseData.fcm.setPriority("high");

  firebaseData.fcm.setTimeToLive(60);

  firebaseData.fcm.setNotifyMessage("Notification", "Hello World!", "firebase-logo.png", 
"http://www.google.com");
  firebaseData.fcm.setTopic("general");

  if (Firebase.sendTopic(firebaseData))
  {
    //Success, print the result returned from server
    Serial.println(firebaseData.fcm.getSendResult());
  }
  else
  {
    //Failed, print the error reason
    Serial.println(firebaseData.errorReason());
  }

`

And Error log:

`
No device token provided

`

Why do I need to give device token for topics?

mobizt, 2019-05-19:
This is a bug and I jus fix this issue and update to v 2.1.2. 
Thank for your report.

arushgupta2007, 2019-05-19:
Is it updated? Can I re-download the library?

mobizt, 2019-05-19:
Yes, you need to delete the old one and download new zip file from this GitHub. Arduino library manager was not the latest version.

arushgupta2007, 2019-05-19:
Ok

arushgupta2007, 2019-05-19:
It worked!
Thanks man, your library has saved my life 👍  :)

closed 2019-05-19 by arushgupta2007

afung2468, 2020-04-13:
Hi Mobizt. I am using Firebase ESP8266 Client Version 2.8.5. but I am still getting this "No device token provided" message. Which version should I use to get rid of this error? Thanks.

mobizt, 2020-04-13:
@afung2468
The issue was fixed since v 2.1.2.

See the fcm [example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Cloud_Messaging/Cloud_Messaging.ino)

afung2468, 2020-04-13:
Thanks for the reply. I used the wrong function to send the topic. Your example does not show how to send a topic. After some experiments, it works now. Thank you.

mobizt, 2020-04-14:
@afung2468

You need to read the doc rather than ask for everthing

https://github.com/mobizt/Firebase-ESP8266/tree/master/src#firebase-cloud-messaging-object-functions