mobizt/Firebase-ESP8266 #28: BUG - warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated

Gerriko, 2019-05-30:
I've just updated my ESP8266 core via Arduino boards manager and we are now getting stronger warnings "'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated" and recommends "Upgrade to BearSSL is advised"

mobizt, 2019-05-30:
It's not bug. I intend to use axTTLS  instead of BearSSL.

Gerriko, 2019-05-31:
Not sure why you want to keep using something that has proven
vulnerabilities and it is very easy to switch over to BearSSL.

mobizt, 2019-05-31:
Did you don't understand backward compatibility?

Gerriko, 2019-05-31:
Yes of course I do. But I cannot see how changing to BearSSL will cause the
library functions to stop working from an application point of view. At
some stage you would need to change that is all.

Your library and in particular the examples you give are very good by the
way. They's been very helpful.

mobizt, 2019-05-31:
You still don't get what I said.
Bearssl was introduced in ESP8266 Arduino core since 2.4.2 and still need to improve to be more stable. As I state on readme that this library support v 2.4.0 and newer. If I switch to use Bearssl immediately I will break supports for user that still use old core sdk.

closed 2019-05-31 by mobizt

Gerriko, 2019-05-31:
Yes, I was using ESP8266 Core SDK version 2.4.0 and above and all was good.
Then when I updated to ESP8266 Core SDK version *2.5.2* they strengthened
their warning message, as before it was just a suggestion. Now with this
new core I receive a stronger warning with a new deprecated message in my
console. So what I gather is that the developers of the ESP8266 Core now
reckon BearSSL must be stable.

mobizt, 2019-05-31:
No warning is jus warning and nothing special, it 's up to library maintainer to use compiler attritute to warn or even rais an error and it comes from [here](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h#L35) from class consructor. 

If you install esp8266 core v < 2.4.0, I also raised the [error](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.h#L58) instead of user warning.

Please stop mention about this. I will definitely change to use BearSSL if core lib does not support axTLS anymore.