mobizt/Firebase-ESP8266 #29: Freeze after internet conection goes down

alvaroaguero55, 2019-06-03:
I can connect and post/get from Firebase Database ok, but if internet connection goes down (ie nodemcu Keeps connected to Wifi all the time) then, when internet comes up again, Firebase doesn´t reconnect again automaticly. I tested that using different smartphones as Mobile Hotspot (Samung J7, S10, S8) in this way: 1) turn on Mobile Hotspot (Nodemcu connects right to wifi and get/post data to Firebase). 2) Disable internet data from smartphone (Nodemcu keeps connected to wifi but doesn't post/get data to Firebase, which is ok because no internet) 3)Enable again internet data from smartphone (Nodemcu doesn´t connect again automaticly to Firebase, so this is the issue).

Any idea of how to fix it? I also test it with linksys routers with same result.

mobizt, 2019-06-03:
Yu can debug the WiFi status by select debug level  and see the serial message.
 
The issue is not from Firebase library but from somewhere in your ESP8266, WiFi library or Hotspot. 

If you call **Firebase.reconnectWiFi(true)** then library will check the WiFi connection status for every Firebase call included stream, and if the WiFi connection failed, then call **WiFi.reconnect()** and wait for 3 seconds and check the WiFi status again before return false with **connection lost** reason if connection failed.

If you not call **Firebase.reconnectWiFi(true)**, WiFi connection will not reconnect and Firebase call function returns false with **connection lost** reason.

Function **WiFi.reconnect()** of WiFi client library will handle AP reconnection and no guarantee for success.

No packets return from AP is difference case even WiFi connection status is OK. 

I also test in the same way like this before even 5GHz to 2.4GHz repeater and know issues that unresolvable then I don't handle anything more than check status > try to connect  then check status again for every Firebase call.

Other connection conflicts that may come from AP hangs and IP/DHCP restriction, no packet, WiFi radio interference and all possible things are unknown and are impossible to handle with the program.

alvaroaguero55, 2019-06-03:
Thanks Mobizt, there is any flunction/flag (like Firebase.failed() of official library) to realize if Firebase connection is successfull? 

I.e. with the official library you can do this if connection goes down:

if (Firebase.failed())                     
      {
      Serial.print("setting number failed:");
      Serial.println(Firebase.error());
      ESP.reset();
      }

mobizt, 2019-06-03:
You can track all error of all firebase calls status using **firebaseData.errorReason()** or **firebaseData.httpCode()**.

You can check on readme on main page, src page and all the example sketches.

closed 2019-06-06 by mobizt