mobizt/Firebase-ESP8266 #30: Updating values + streaming, streaming ignored

arushgupta2007, 2019-06-09:
Hello,
I am updating a value in my Firebase realtime Database every second, and also streaming another value. For some reason when I start updating the value, the streaming stops and when I stop updating, the stream starts to work. 
Do you know what is the issue?!

Thanks in advance

mobizt, 2019-06-09:
This is not issue this is the normal behavior when you use only one Firebase Data object that hold one instance of WiFi client. 

The stream will disconnected to start new connection with different node path when you call firebase functions e.g get, set, update, push and delete, and reconnect the stream again when you call readStream.

You can use different Firebase Data object to handle only stream task and it uses additional 20k of ram.

arushgupta2007, 2019-06-10:
Oh,

So there is no way I can update value and continue stream at the same time?

mobizt, 2019-06-10:
As I said **"use different Firebase Data object"**, see [this example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Different_objects_stream/Different_objects_stream.ino).

arushgupta2007, 2019-06-10:
It worked :) 

Thanks for the support and help

closed 2019-06-10 by arushgupta2007