mobizt/Firebase-ESP8266 #32: Streaming stops after a while ..!

MrMakaVeLLi, 2019-06-22:
Hi,
i'm using Firebase-ESP8266 Library, basically i'm using a mobile app to send and receive data from my firebase .
everything works fine but after several hours , streaming stops but i can send data ..!
so i can't receive a new data from the streaming path , so what is the problem ? 
thanks in advance .

mobizt, 2019-06-23:
Read this issue https://github.com/mobizt/Firebase-ESP8266/issues/30

MrMakaVeLLi, 2019-06-23:
i think my problem is very different 
my problem is that streaming stops without any condition and in random times !
it could be working for hours without problems and it could be work for only 2 minutes !

MrMakaVeLLi, 2019-06-23:
> Read this issue #30

i think my problem is very different
my problem is that streaming stops without any condition and in random times !
it could be working for hours without problems and it could be work for only 2 minutes !

mobizt, 2019-06-23:
Please debug WiFi and check the printed serial. Did you try #330 solution?. I didn't see your sketch, serial printed result and what's your WiFi type, I cannot tell anything.

MrMakaVeLLi, 2019-06-23:
> Please debug WiFi and check the printed serial. Did you try #330 solution?. I didn't see your sketch, serial printed result and what's your WiFi type, I cannot tell anything.

that's the problem , when i debugging using serial when the streaming is lost, i got nothing !
no error message appeared like everything is running good !
here's my sketch for that part:
`if (WiFi.status() != WL_CONNECTED)
  {
    WiFi_Connect();
  }
  else
  {
    if (!Firebase.readStream(firebaseData))
    {
      Serial.println("Can't read stream data");
      Serial.println("REASON: " + firebaseData.errorReason());
      Serial.println();
    }
    if (firebaseData.streamTimeout())
    {
      Serial.println("Stream timeout, resume streaming...");
      Serial.println();
    }
    if ( !Firebase.beginStream(firebaseData, path + "/State") )
    {
      Serial.println("Can't read stream data");
      Serial.println("REASON: " + firebaseData.errorReason());
      Serial.println();
      ESP.restart();
    }
    if (firebaseData.streamAvailable())
    {
      val = firebaseData.stringData().toInt();
    }`

mobizt, 2019-06-23:
This part is not need, the library will check the internet connection and report the ploblem for you.

```C++
if (WiFi.status() != WL_CONNECTED)
{
    WiFi_Connect();
}

```

MrMakaVeLLi, 2019-06-23:
> This part is not need, the library will check the internet connection and report the ploblem for you.
> 
> ```c++
> if (WiFi.status() != WL_CONNECTED)
> {
>     WiFi_Connect();
> }
> ```

so , that's what causing the problem ?!

mobizt, 2019-06-23:
Where is this part of your code at, setup or loop?

MrMakaVeLLi, 2019-06-23:
> Where is this part of your code at, setup or loop?

both ,
void WiFi_Connect()
{
 while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
  }
  Serial.println("Connected !");
}
void setup()
{
  Serial.begin(115200);
  pinMode(latchPin, OUTPUT);
  pinMode(dataPin, OUTPUT);
  pinMode(clockPin, OUTPUT);
  EEPROM.begin(512);
  wifi_set_sleep_type(NONE_SLEEP_T);
  delay(100);
  for ( int i = 1; i <= 16; i++ )
  {
    Turn(i, EEPROM.read(i));
  }
  WiFi_Connect();
  Serial.println();
  Serial.println(WiFi.localIP());
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  if (!Firebase.beginStream(firebaseData, path + "/State"))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("PASSED");
    Serial.println("------------------------------------");
    Serial.println();
  }
}

MrMakaVeLLi, 2019-06-23:
> Where is this part of your code at, setup or loop?

both

void WiFi_Connect()
{
 while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
  }
  Serial.println("Connected !");
  Serial.println();
  Serial.println(WiFi.localIP());
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
}
void setup()
{
  Serial.begin(115200);
  pinMode(latchPin, OUTPUT);
  pinMode(dataPin, OUTPUT);
  pinMode(clockPin, OUTPUT);
  EEPROM.begin(512);
  wifi_set_sleep_type(NONE_SLEEP_T);
  delay(100);
  for ( int i = 1; i <= 16; i++ )
  {
    Turn(i, EEPROM.read(i));
  }
  WiFi_Connect();
  if (!Firebase.beginStream(firebaseData, path + "/State"))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("PASSED");
    Serial.println("------------------------------------");
    Serial.println();
  }
}

mobizt, 2019-06-23:
**Firebase.readStream** must be in loop and try to avoid to use blocking code (time consumed work) or delay in that loop.

**Firebase.beginStream** must be call once in setup and should not call it again for the same streaming path. 

If you use the same Firebase Data object to normal firebase calls and stream. This will interrupts stream connection and you not need to call **beginStream** again to reume, it resumed automatically with **Firebase.readStream** as long as r**eadStream** runs in the loop.

If you want to stream multiple streaming path using **Firebase.beginStream**  somewhare in your loop, it's not efficient because it breaks your last streaming and reconnect to new streaming with difference streaming paths.

MrMakaVeLLi, 2019-06-23:
i will try to delete **Firebase.beginStream** in the loop and try again
i'm using only one streaming object , i used Firebase.pushInt() function only for testing the connectivity to firebase and the result was firebase is still connected and i can send but i can't receive ( after problem happened )

mobizt, 2019-06-23:
Keep **Firebase.readStream** in loop, it will check the Firebase connection for you without to test by set any value to Firebase and if you use the same Firebase Data object, call any Firebase function (get/set/push/update/delete) will break the stream and resumed automatically.

closed 2019-06-24 by mobizt

MrMakaVeLLi, 2019-06-26:
thanks a lot, problem is fixed after removing 
**if ( !Firebase.beginStream(firebaseData, path + "/State") )**
from the loop .
if wifi is dropped , the default of the library is reconnecting ?

mobizt, 2019-06-26:
Yes, if you callFirebase.reconnectWiFi(true) once in set up.

asifakhtar98, 2021-12-25:
Firebase Firestore Stream Subscription automatically stopping in Flutter Apps. I have got this issue with all state managements packages. I think this problem in Firebase itself, not in client apps.

amy1school, 2022-11-24:
Streaming stop after a while, I tried in multiple ways but did not worked.  Please help me 

const int led = LED_BUILTIN;
const int led2 = 4;   //d2
const int led3 = 14;  //d6

#include <EEPROM.h>
#include <FirebaseESP8266.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

#include <WebSocketsServer.h>
#include <ESP8266WiFiMulti.h>
#include <ArduinoJson.h>
ESP8266WiFiMulti wifiMulti;
WebSocketsServer websockets(81);
const uint32_t connectTimeoutMs = 5000;

String esid; 
String epass = "";
String temp_path;
String temp_value;
String temp_code;

String fbdo_err ="";

#define USER_EMAIL "  "
#define USER_PASSWORD "  "

#define API_KEY "    "
#define DATABASE_URL "   "

FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

String uid;
String databasePath;
String fbdoPath;

unsigned long fb_uid_new_millis=0;
unsigned long fb_uid_old_millis=0;
unsigned long fb_uid_wait =5000;

unsigned long internet_new_millis=0;
unsigned long internet_old_millis=0;
unsigned long internet_wait=5000;

//unsigned long wifi_new_millis=0;
unsigned long wifi_old_millis=0;
const long wifi_wait=15000;

unsigned long i_new_millis=0;
unsigned long i_old_millis=0;
unsigned long i_wait =12000;

int b=0;
int c=0;
int internet=0;
int internet2=0;
int fb=0;
int tb=0;
int ws_s=0;
int j=0;
int k=0;
int Timeout=2;
int feed=0;
String feed_v="";
String path_i="";
String value_i="";
[... 246 lines cut ...]
  }
  data[len]='\0';
  return String(data);
}

void output()
{
  if(path_i == "/hl1")
  {
      if(value_i == "1")
      {
        digitalWrite(led3, HIGH);
      }
      else
      {
         digitalWrite(led3, LOW);  
      }  
  }
  if(path_i == "/feed")
  {
    feed=1;
    feed_v=value_i;
    delay(500);
  }  
}

mobizt, 2022-11-24:
@amy1school 

Check the examples and read all comments in the examples to use the library properly, what you should do and don't.