mobizt/Firebase-ESP8266 #33: Cloud Messaging no sound

tomworms, 2019-06-23:
Hi,
I am using Firebase-ESP8266 Library,  example of Cloud_Messaging. My question is how to start the sound in push notification. I tried eg
`firebaseData1.fcm.setDataMessage ("{\" sound \ ": \" ON \ "}");`
I am asking for a suggestion. thanks.

mobizt, 2019-06-23:
For Android

From this [ref](https://firebase.google.com/docs/reference/fcm/rest/v1/projects.messages#androidnotification). The sound to play when the device receives the notification. Supports "default" or the filename of a sound resource bundled in the app. Sound files must reside in /res/raw/

Try this if you don't know or not prepared sound file.

"{\"sound\":\"default\"}"

tomworms, 2019-06-23:
exactly, the problem is about the Android system.
Unfortunately, the result is a mistake:
in code
firebaseData1.fcm.setDataMessage ("{" sound "," default "}");
compilation error in ARDUINO:
unable to find string literal operator 'operator'"sound"

Generation from firebase.google.com triggers the default sound on your phone so I guess it's all ok

mobizt, 2019-06-23:
See above comment again I forgot the escape char \ for quotes "

tomworms, 2019-06-23:
it happens, there is still no sound.
if this call is correct

firebaseData1.fcm.setNotifyMessage or firebaseData1.fcm.setDataMessage ??
....
```
    firebaseData1.fcm.setNotifyMessage("Notification", "Hello World! " + String(count));
    firebaseData1.fcm.setDataMessage("{\"myData\":" + String(count) + "}"); 
    firebaseData1.fcm.setDataMessage("{\"sound\":\"default\"}");
```
....

mobizt, 2019-06-23:
Actually I'm not prepare all parameters for notification message, you should edit library at line 5475 of [FirebaseESP8266.cpp](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.cpp#L5475)

From 
```C++

strcat_P(msg, ESP8266_FIREBASE_STR_127);

```
to

```C++

strcat_P(msg, F(",\"sound\":\"default\""));

strcat_P(msg, ESP8266_FIREBASE_STR_127);
```

tomworms, 2019-06-23:
thank you, but it did not help.
Arduino 1 .8.7
  Error compiling for board LOLIN (WEMOS) D1 R2 & mini.

PlatformIO - compilation after changes library, it's ok but still no sound

mobizt, 2019-06-23:
The modification is not cause the error on Arduino IDE except for your message is too large (400 bytes are reserved for build the message payload) or you make something wrong. 

Sound is depends on your mobile device settings whether you muted the notification or not, check your phone settings. I can hear the notification sound with this modified payload.

closed 2019-06-23 by mobizt

tomworms, 2019-06-23:
I changed to :
`        strcat_P(msg, PSTR(",\"sound\":\"default\""));`
instead of F, I used PSTR.
it looks like I have a problem with __FlashStringHelper.

with PSTR it works and it's OK thank you!

ahmedejaz22, 2020-12-11:
> Hi,
> I am using Firebase-ESP8266 Library, example of Cloud_Messaging. My question is how to start the sound in push notification. I tried eg
> `firebaseData1.fcm.setDataMessage ("{\" sound \ ": \" ON \ "}");`
> I am asking for a suggestion. thanks.

Hi,
Sir I'm trying to send notification message to my flutter app using topic. The message is working fine with google developer console with topic "all" but arduino says message passed but i don't get any notification on phone all. Can you kindly give any idea. If possible plz share code that works.

    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
    Firebase.reconnectWiFi(true);

    //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
    firebaseData1.setBSSLBufferSize(1024, 1024);

    //Set the size of HTTP response buffers in the case where we want to work with large data.
    firebaseData1.setResponseSize(1024);

    firebaseData1.fcm.begin(FIREBASE_FCM_SERVER_KEY);

    firebaseData1.fcm.setPriority("high");

    firebaseData1.fcm.setTimeToLive(1000);

    firebaseData1.fcm.setTopic("all");

    sendMessage();
}

void loop()
{
    if (millis() - lastTime > 60 * 1000)
    {
        lastTime = millis();
        sendMessage();
    }
}

void sendMessage()
{

Serial.println("------------------------------------");
Serial.println("Send Firebase Cloud Messaging...");

 firebaseData1.fcm.setNotifyMessage("Notification", "Hello World!");

    //firebaseData1.fcm.setDataMessage("{\"myData\":" + String(count) + "}");

//firebaseData1.fcm.setTopic("all");

//if (Firebase.broadcastMessage(firebaseData1))
//Firebase.sendTopic(firebaseData1);
if (Firebase.sendTopic(firebaseData1))
//if (Firebase.sendMessage(firebaseData1, 0))//send message to recipient index 0
{

    Serial.println("PASSED");
    Serial.println(firebaseData1.fcm.getSendResult());
    Serial.println("------------------------------------");
    Serial.println();
}
else
{
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData1.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
}

count++;
}

mobizt, 2020-12-11:
@ahmedejaz22  Is this work for you? https://github.com/mobizt/Firebase-ESP8266/issues/157

ahmedejaz22, 2020-12-13:
sadly beyond 2.9.5 library doesn't work on nodemcu - gives error compiling
errors

thanks for remembering

Best Regards,

Ahmed

mobizt, 2020-12-13:
@ahmedejaz22

You may need to update the ESP8266 Arduino Core SDK to latest.

If you include some WiFi UART bridge library in the sketch and see the compiler error like this
**WiFi UART bridge was not supported** 

You can't use the Firebase library in your sketch with that library included. 

If the errors are about multiple libraries found, you 're installing the ambiguous libraries with the same class name or file name in many locations which used by this Firebase library but these libraries come from different repos and  are not compatible to each other e.g. SD.h installed in **libraries** folder which you may install and SD.h in the core library which used by Firebase library will cause the compilation errors.