mobizt/Firebase-ESP8266 #34: wifi.reconnect(true); problem

MrMakaVeLLi, 2019-07-05:
Hi,
i have many problems :

**First** problem with wifi.reconnect(true);  : 
i tried the wifi.reconnect(true); and replaced it with my manual reconnect, but actually it didn't work!
what i did is to connect my nodemcu to my router then restart my router, the result no reconnect is happened !

**Second** problem with firebaseData.errorReason() :
if the wifi is connected and there's no streaming or any change in firebase, the errorReason is " not connected "
if internet is being down ( wifi connected ), it has the same result " not connected " !
only if wifi is off the reason is " connection refused " 
so, how can i differ between normal situation and connection being lost with the two states having the same error message !?

**Third** problem : 
if internet connection is lost ( wifi still connected ), and internet is back then, streaming is lost !
and the only way to get streaming back is to reset the nodemcu or reset wifi ( but using manual wifi reconnect ) .
my code: 

```
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <EEPROM.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <ShiftRegister74HC595.h>
#include "WiFiManager.h"
WiFiServer server(80);
WiFiClient client;
#define FIREBASE_HOST "xxxxxxx"                     //Your Firebase Project URL goes here without "http:" , "\" and "/"
#define FIREBASE_AUTH "xxxxxxx"       //Your Firebase Database Secret goes here
FirebaseData firebaseData;
ShiftRegister74HC595 Turn (2, 5, 16, 4);
String path = "";
int val, prev = 0;
void WiFi_Connect()
{
  server.stop();
  server.setNoDelay(false);
  WiFiManager wifiManager;
  //wifiManager.resetSettings();
  wifiManager.setTimeout(30);
  if (!wifiManager.autoConnect("NAME", "PASSWORD"))
  {
    Serial.println("failed to connect and hit timeout");
    ESP.reset();
  }
  Serial.println("Connected !");
  server.begin();
  server.setNoDelay(true);
}
void setup()
{
  Serial.begin(115200);
  EEPROM.begin(512);
  wifi_set_sleep_type(NONE_SLEEP_T);
  delay(100);
  for ( int i = 1; i <= 16; i++ )
  {
    Turn.set(i - 1, EEPROM.read(i));
  }
  WiFi_Connect();
  Serial.println();
  Serial.print("connected:");
  Serial.println(WiFi.localIP());
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  if (!Firebase.beginStream(firebaseData, path + "/State"))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("PASSED");
    Serial.println("------------------------------------");
    Serial.println();
  }
}
void loop()
{
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("Can't read stream data");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println();
  }
  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  if (firebaseData.streamAvailable())
  {
    val = firebaseData.stringData().toInt();
  }
  if (server.hasClient())
  {
    client = server.available();
  }
  if (client && client.connected()) {
    if (client.available()) {
      String readString = "";
      while (client.available())
      {
        char c = ((char)client.read());
        readString += c;
      }
      val = readString.toInt();
    }
  }
  if ( prev != val )
  {
      Serial.println(val);
  }
  prev = val;
}
```
thanks in advance !

mobizt, 2019-07-06:
You trying to use your device ESP8266 as server while connected to internet.

Which you need to set the WiFi mode to AP_STA to connect to both client as server and Firebase server via internet as client.

You need to start with

```WiFi.mode(WIFI_AP_STA);```

Then connect to WiFi with

```WiFi.begin("WiFi_SSID", "WiFi_PSW");```

Then set your device as AP

```WiFi.softAP("Your_AP_Name", "Your_AP_PSW");```

You may face some unreliable issues which you can search from google or Github about WiFi in WiFi_AP_STA mode.

The following blocking code from handle client data will block the Firebase stream connection too as it runs in the loop.
```C++

if (server.hasClient())
  {
    client = server.available();
  }
  if (client && client.connected()) {
    if (client.available()) {
      String readString = "";
      while (client.available())
      {
        char c = ((char)client.read());
        readString += c;
      }
      val = readString.toInt();
    }
  }
```

I'm not recommend to use as server and client at the same time because the library need to connect to internet and assume that your device was well configured and internet ready. 

When some blocking operation that cases the STA mode connection problem occurred, library will report the error based on WiFi client status after trying to reconnect the WiFi or timed out when trying to connect to server.

MrMakaVeLLi, 2019-07-06:
> You're trying to use your device ESP8266 as server while connected to internet.
> 
> Which you need to set the WiFi mode to AP_STA to connect to both client as server and Firebase server via internet as client.
> 
> You need to start with
> 
> `WiFi.mode(WIFI_AP_STA);`
> 
> Then connect to WiFi with
> 
> `WiFi.begin("WiFi_SSID", "WiFi_PSW");`
> 
> Then set your device as AP
> 
> `WiFi.softAP("Your_AP_Name", "Your_AP_PSW");`
> 
> You may face some unreliable issues which you can search from google or Github about WiFi in WiFi_AP_STA mode.
> 
> The following blocking code from handle client data will block the Firebase stream connection too as it runs in the loop.
> 
> ```c++
> if (server.hasClient())
>   {
>     client = server.available();
>   }
>   if (client && client.connected()) {
>     if (client.available()) {
>       String readString = "";
>       while (client.available())
>       {
>         char c = ((char)client.read());
>         readString += c;
>       }
>       val = readString.toInt();
>     }
>   }
> ```
> 
> I'm not recommend to use device as server and client at the same time because the library need to connect to internet and assume that your device was well configured and internet ready.
> 
> When some blocking operation that cases the STA mode connection problem occurred, library will report the error based on WiFi client status after trying to reconnect the WiFi or timed out when trying to connect to server.

after removing this part the result still the same , if internet connection is lost and back again , streaming is stopped and the error is " not connected " 
is it a bug ?

mobizt, 2019-07-06:
You need to read what I menttion above carefully.

You're trying to use WiFi in AP + STA  mode without proper setup. I advice for neccessary setup.

Eventhough the  AP + STA  mode was setup, some issue about WiFi may be occurred (WiFi connection was unstable under this mode) which may cause the internet connection lost. 

The library never detect the root cause of the WiFi core library problem. 

Finally I'm not recommend you to use library in this WiFi mode due to unstable operation.

mobizt, 2019-07-06:
I also don't understand that why you start device as WiFi server (AP mode) instead of normal client (STA mode) or you want to response some web request but I never seen you send http response in your code.

If you copy some part of code from example somewhere on internet, please study how the code works and what is the purpose of each line of code.

mobizt, 2019-07-06:
My mistake, the above comment about WiFi AP_STA mode is not neccessary, you can start the device as server under the local network which I still not understand what you're trying to do with server.

The use as WiFi server and client together is not recommend. 

You need to debug the WiFi is only way to investigate the problem. The library report the status of WiFi client directly but need to debug for the reason.

MrMakaVeLLi, 2019-07-06:
> My mistake, the above comment about WiFi AP_STA mode is not neccessary, you can start the device as server under the local network which I still not understand what you're trying to do with server. [...]

i already did that , here's what i got from the debug :

> Can't read stream data 
> REASON: not connected
> Stream timeout, resume streaming...

Note that this result is the same as the internet alive and streaming is connected !
what i did for debugging and testing : 
connect the device to the wifi 
wait the firebase streaming to successfully connect
read a change successfully 
unplug the telephone cable from my router ( no internet )
check the serial monitor still the same as above 
plug the cable again ( internet is back ) 
still the same result and no response 
what i did is to simulate what happening if i have a bad internet connection or connection interrupt , and what i need to do is a self reconnect to the internet if internet is back .
note that if restart my router , the device is automatically reconnect and streaming is successfully resumed again .
thanks for your time .

mobizt, 2019-07-07:
I mean WiFi debug in Arduino IDE,  from Tools > Debug Level.
You still not tell me why you start the sever with  `server.begin();`. 

I saw you use by include WiFiManager, DNSServer, ESP8266WebServer, Is it your code or you copy from internet?.

Your may have many idea how to build your system but the real implementation, you need to test everything separately that it works and add that to your code step by step, test it with your existing system that it works properly under the system. 

You need to always debug the heap and stack memory which are consumed by every library you used and your own code because low memory lead to unexpected fault.

No matter what you're trying, ESP8266 itself trying to reconnect WiFi when it lost. 

The library try it best to recover the internet connection as I post this https://github.com/mobizt/Firebase-ESP8266/issues/29#issuecomment-498272286.

You need to investigate and debug yourself.  Try to connect WiFi with normal and clean method without any library except for Firebase and ESP8266WiFi.

MrMakaVeLLi, 2019-07-07:
here's the wifi debug after connecting the wifi and unplug then plug the telephone cable :
[image]

server.begin(); is used for local server 
WiFiManager, DNSServer, ESP8266WebServer
are libraries to manage wifi settings so that i don't need to reconfigure wifi if wifi network name changed, and all i need is to create an access point and connect to , then open a specific local web page and reconfigue the wifi settings.

> You need to investigate and debug yourself. Try to connect WiFi with normal and clean method without any library except for Firebase and ESP8266WiFi.

actually i did that and result is the same !
streaming lost and never back again if internet interrupted !
streaming back again if i resetart my router ( not my NodeMCU ) so that it;s NOT a heap or memory problem .

mobizt, 2019-07-07:
Try the following.
 
- Reset your router.
- Connect to another WiFi hotspot.
- Connect to soft AP created with your computer.
- Connect to your mobile phone WiFi hotspot.
- Try the [following post](https://github.com/mobizt/Firebase-ESP8266/issues/7#issuecomment-485303352)

mobizt, 2019-07-07:
I understand your edit message about WiFiManager, know it and ever use it. Lets try my above solution. 
The problem may be the WiFi core lib or router which you need to try yourself. 

Sorry I will close the issue because it's not relate to library anymore.

closed 2019-07-07 by mobizt