mobizt/Firebase-ESP8266 #35: getInt and setInt

MarijoVidovic, 2019-07-07:
Hi can someone help me to get int from firebase.
This is code i am using for this 

`R = Firebase.getInt(firebaseData, "DomeLED/Color/r");`

Where R is my global variable of Int type, firebaseData is object, and after is the path for reading the value from 

it work perfectly for boolean type, but not for integer or float

mobizt, 2019-07-07:
All Firebase call functions return the operation status (boolean), true for success and false for failed.

To get data, you can use `firebaseData.intData()`.

Check [this example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino), for basic useage.

MarijoVidovic, 2019-07-07:
So i have to use something like this for every int i read? 

`if (Firebase.getInt(firebaseData, path + "/Double/Data" + (i + 1)))
    {
      Serial.println("PASSED");
      Serial.println("PATH: " + firebaseData.dataPath());
      Serial.println("TYPE: " + firebaseData.dataType());
      Serial.println("ETag: " + firebaseData.ETag());
      Serial.print("VALUE: ");
      if (firebaseData.dataType() == "int")
        Serial.println(firebaseData.intData());
      else if (firebaseData.dataType() == "float")
        Serial.println(firebaseData.floatData(), 5);
      else if (firebaseData.dataType() == "double")
        printf("%.9lf\n", firebaseData.doubleData());
      else if (firebaseData.dataType() == "boolean")
        Serial.println(firebaseData.boolData() == 1 ? "true" : "false");
      else if (firebaseData.dataType() == "string")
        Serial.println(firebaseData.stringData());
      else if (firebaseData.dataType() == "json")
        Serial.println(firebaseData.jsonData());
      Serial.println("------------------------------------");
      Serial.println();
    }`

for my usage would be 
```
if (Firebase.getInt(firebaseData, "DomeLED/Color/r" ){
if (firebaseData.dataType() == "int")
        R = (firebaseData.intData());
}
```

I have a bit hard time understanding this library :)
Sorry for obviously dumb question

mobizt, 2019-07-07:
Yes, you can check the data type before get its value  which ensure that you will not mis the resolution of numeric data in case of int, float and double.

MarijoVidovic, 2019-07-07:
> Yes, you can check the data type before get its value which ensure that you will not mis the resolution of numeric data in case of int, float and double.
> 
> In addition, the usual function e.g. myValue = MyFunction(); as in other Firebase library that returns the data directly is not good practice when it processes the http request and response tasks inside the function which can be failed, unreachable, invalid request or many error reasons which user will get undesired and questionable value from that function.

Thank you i think i figure it out. It need a little effort but its good now. I wanted to avoid fingerprint so i used this library. Nice work and keep it up :)

closed 2019-07-07 by MarijoVidovic