mobizt/Firebase-ESP8266 #36: HELP

mesut25, 2019-07-08:
merhaba mobizt
sevgili arkadaşım 
bizlere basit bir örnek yazabilirmisiniz?
bizler daha çok örnek konusu elektronik bir led  açmak (role çalıştırmak)  veya  butona basmak firebase yazmak akıcı ve basit bir örnek paylaşırsan çok sevinirim.

mobizt, 2019-07-09:
English please.

I'm already created a lot of examples at https://github.com/mobizt/Firebase-ESP8266/tree/master/examples that show you how to use it.
 
If you want to know the basic setup for Firebase to use with ESP8266, I also write the story here.
 
Part 1 
https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp32-2d86eda06ff1
 
Part 2 (For ESP32 but similar to ESP8266)
https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp32-def049181b57

mesut25, 2019-07-09:
I am watching you with admiration.
I have installed your library but I get a card error I have installed the card version (2.4.1) current version (2.5.2) error continues.
I can't work with current versions, if you let me share the picture and tell you why.

mobizt, 2019-07-09:
Try the following.

- Update or install Firebase ESP8266 library v2.1.6 (delete the old version from libraries folder).
- Update Arduino IDE.
- From menu **Tool**s > **Flash Size**, choose from **1M SPIFFS** to **3M SPIFFS** based on your device.

Normally you should not get any error from code compilation.

Please copy all errors that show on Arduino IDE status window and paste it in your post.

mesut25, 2019-07-09:
Properly successful programming.
but all my steps could not find the same error where.
the result is successful. Many thanks for the help.

closed 2019-07-09 by mesut25