mobizt/Firebase-ESP8266 #37: Random "SOFT WDT reset" crashes + Feature request

arushgupta2007, 2019-07-10:
Hi,

I am experiencing soft wdt reset after some time of Firebase streaming and edit some values in Firebase console. I opened an issue [here](https://github.com/esp8266/Arduino/issues/6277) where he said it is most likely caused by the Firebase library using axtls which is deprecated.

Code:

```cpp
#include <ACS712.h>
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

#define FIREBASE_HOST "..."
#define FIREBASE_AUTH "..."
#define WIFI_SSID "..."
#define WIFI_PASSWORD "..."

ACS712 sensor(ACS712_05B, A0);
int delayto = 2000;
bool has_changed = false;
String path = "";
unsigned long previousMillis = 0;
const long interval = 1000;
int till = 0;
//int debug = 0;

FirebaseData firebaseData;
FirebaseData firebaseDatadiffrent_obj;
DynamicJsonBuffer json_buffer;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);

  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  pinMode(D2, INPUT);
  pinMode(D7, OUTPUT);
  pinMode(D8, OUTPUT);

  sensor.calibrate();

  attachInterrupt(digitalPinToInterrupt(D2), zcd, FALLING);

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (Firebase.getString(firebaseData, "/arushpass/device_path/esp{2615}")) {
    path = String("/arushpass" + firebaseData.stringData());
    Serial.println(path);
  }

  if (!Firebase.beginStream(firebaseData, path + "/controls"))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
[... 147 lines cut ...]
0x401001dd: __digitalWrite at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring_digital.c line 82
0x40207382: zcd() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 143
0x40100349: interrupt_handler at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring_digital.c line 166
0x40100338: interrupt_handler at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring_digital.c line 162
0x4020d7c4: ESP8266WiFiSTAClass::status() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 563
0x4020b878: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1718
0x4020ce18: FirebaseESP8266::sendRequest(FirebaseData&, unsigned char, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1828
0x4020d02a: FirebaseESP8266::buildRequest(FirebaseData&, unsigned char, unsigned char, std::string const&, char const*, bool, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 199
0x402118e1: Print::write(char const*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Print.h line 60
0x4021cef0: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4020d1ab: FirebaseESP8266::setBool(FirebaseData&, std::string const&, bool, bool, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 728
0x4021d7f2: std::basic_string  , std::allocator  >::basic_string(char const*, std::allocator  const&) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 218
0x4010014c: millis at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring.c line 183
0x4020d21b: FirebaseESP8266::setBool(FirebaseData&, String const&, bool) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 688
0x40211eb7: String::reserve(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 146
0x40211f00: String::copy(char const*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 182
0x40211fec: String::operator=(String const&) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 220
0x40207795: loop() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 85
0x40212400: loop_wrapper() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 125

```

Where do you think I am going wrong? Is it because of axtls? How can I fix it?

He reccomends using **bssl**. Could you use bssl?

mobizt, 2019-07-10:
Actually axTLS was obsoleted and not cause any problem as long as it's set as optional. BearSSL may be the next to update but not this time.

The possible error causes can be ISR.

First, you must not use delay in Interrupt Service Routine (ISR), i.e **zcd**.

If the incomming interrup take place and the operation from last interrupt event is not finished, the error can be occured.

The best practice is do all operations inside ISR as fast as possible by set the volatile variable e.g. Boolean flag and check that flag outside ISR.

mobizt, 2019-07-10:
Please try to remove delay from zcd().
Or try to use volatile variable e.g. declare it as **volatile int delayto**

arushgupta2007, 2019-07-10:
Hi, 

I am trying to control the status and the dimness of a bulb for which I am using 2 triacs and zero cross detection circuit, so removing the delay would not be possible as it is needed for light dimming, I try volitile int in a bit.

arushgupta2007, 2019-07-10:
I tried adding detachInterrupt(digitalPinToInterrupt(D2)); and   attachInterrupt(digitalPinToInterrupt(D2), zcd, FALLING); on the start and end of zcd, and it worked for most of the part. I ran it for ~10 mins, and after that I got this error:

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4021ff26
EXCVADDR: 0x00000100
Memory allocation of 520 bytes failed at 0x4022f962: more_comps at crypto/bigint.c line 1072

Decoding stack results
0x4022fb5c: alloc at crypto/bigint.c line 1106
0x402303e0: bi_square at crypto/bigint.c line 958
0x4022fa08: trim at crypto/bigint.c line 1197
0x402304e1: bi_square at crypto/bigint.c line 1016
0x4022ff92: bi_clone at crypto/bigint.c line 263
0x40230989: bi_barrett at crypto/bigint.c line 1285
0x40230bf0: bi_mod_power at crypto/bigint.c line 1414
0x4022fa08: trim at crypto/bigint.c line 1197
0x40231e0d: RSA_encrypt at crypto/rsa.c line 254
0x4022c968: send_client_key_xchg at ssl/tls1_clnt.c line 435
0x4022ce61: do_clnt_handshake at ssl/tls1_clnt.c line 123
0x4022a04c: mem_free at core/mem.c line 237
0x40224c6a: pbuf_free_LWIP2 at core/pbuf.c line 786
0x4020874a: ClientContext::read(char*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 554
0x4020e956: __yield() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 100
0x4022c770: basic_read at ssl/tls1.c line 1607
0x402202c6: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4022c8d8: ssl_read at ssl/tls1.c line 314
0x40208db9: axTLS::SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/SSLContext.h line 137
0x4020e874: esp_yield() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4020ee53: delay(unsigned long) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40209007: axTLS::WiFiClientSecure::_connectSSL(char const*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/SSLContext.h line 341
0x402090ad: axTLS::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureAxTLS.cpp line 117
0x40207345: FirebaseHTTPClient::http_connect() at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266HTTPClient.cpp line 82
0x40207380: FirebaseHTTPClient::http_sendRequest(char const*, char const*) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266HTTPClient.cpp line 63
0x40207714: ESP8266WiFiSTAClass::status() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x402202c6: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x40203130: FirebaseESP8266::sendFirebaseRequest(FirebaseData&, char const*, unsigned char, unsigned char, char const*, char const*, unsigned int) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 3362
0x40211034: free(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x40204c21: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1480
0x4022c403: basic_read at ssl/tls1.c line 1374
0x40100532: __attachInterrupt(uint8_t, voidFuncPtr, int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 206
0x401000f1: zcd() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 147
0x40100439: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 169
0x40100428: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 165
0x40207714: ESP8266WiFiSTAClass::status() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x402052cc: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1718
0x40206ab0: FirebaseESP8266::sendRequest(FirebaseData&, unsigned char, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1828
[... cut ...]
0x4020e25c: String::copy(char const*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 211
0x4020e3f8: String::operator=(String const&) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 259
0x40201391: loop() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 85
0x4020e924: loop_wrapper() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125

mobizt, 2019-07-10:
You **NEED** to remove delay from ISR, **you don't have choices**, the resource was pushed in stack every new interrupt occurred causes stack growed bigger and interfered with heap memory then corrupted the heap and finally system error.  Any way, nested interrupt must be prevented.

For timing, you can use non-block code as millis() or micros() to get the milliseconds and microseconds counts. Or you can use esp os timer and schedule task library [Ticker](https://github.com/esp8266/Arduino/tree/master/libraries/Ticker).

Redesign your program flow, try to not use blocking code (delay), use the ms/us counter comparation, timer or schedule tasks.

arushgupta2007, 2019-07-10:
Hey thanks for your suggestion, I try Ticker in a bit. Through volatile int did not work for me

closed 2019-07-11 by mobizt

arushgupta2007, 2019-07-12:
🎉 It worked! 🎉

Thanks @mobizt  for your time and support!