mobizt/Firebase-ESP8266 #39: Issue about "Not Required Fingerprint and Certificate" features

lhdxb69, 2019-07-12:
Hello @mobizt , thank for your great project!
I have a few questions, look forward to your answers.
1) Without fingerprint, is there a possibility of man in the middle attack?
2) Is your "Not Required Fingerprint and Certificate" features as the same way:
    bool WiFiClientSecure::verify(const char* fp, const char* domain_name)
    {
        return trure;
Have a nice day!

mobizt, 2019-07-12:
For your questions.

- It's SSL encryption (128 or 256 bit) data included your URL parameters, don't worry, fingerprint was used in handshake to verify the identity of each other that all are match and not involved in data encryption.

  Google convinced us to use OAuth2 token or Firebase ID tokens which is the standard method and better.

- If you use WiFiClientSecure library directly, verify is upon you to code, you can connect to server directly or verify the fingerprint. 

  If you use HTTPClient library, it calls [verify](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L76) to WiFiClientSecure.

closed 2019-07-12 by mobizt

lhdxb69, 2019-07-12:
Thank you very much!