mobizt/Firebase-ESP8266 #40: HELP: How to push timestamp with data

l0ginp, 2019-07-17:
Hi, I see in timestamp example function like "Firebase.pushTimestamp(firebaseData, path)" has push only timestamp but i want to push like this { "data":"data1", "timestamp":"1563339831099"  }
thanks

mobizt, 2019-07-17:
This is the feature provided by Firebase RTDB which is the server value of timestamp in millisecond as stated in [this document](https://firebase.google.com/docs/reference/rest/database#section-server-values).

Library was not take or handle any timestamp from device, it sends payload **{".sv": "timestamp"}** instead.

If you wish use timestamp in json data, you need to get your timestamp value from NTP server or RTC chip .

l0ginp, 2019-07-17:
Many thanks.

closed 2019-07-17 by l0ginp