mobizt/Firebase-ESP8266 #41: Help with compiler error

kbssa, 2019-07-17:
When I try to compile the Basic example I get the following error:

> In file included from C:\Users\SuporteProduto1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD\src/SD.h:25:0,

                 from C:\Users\SuporteProduto1\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h:72,

                 from C:\Users\SuporteProduto1\Documents\Arduino\libraries\Firebase_ESP8266_Client\examples\Basic\Basic.ino:15:

C:\Users\SuporteProduto1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:190:5: error: 'sdfat' does not name a type

     sdfat::SdFat* getFs()

>

mobizt, 2019-07-18:
Please make sure you don't have or installed SdFat or SD library in libraries folder, try to remove it from libraries folder because it's not consistent with core library that used by this Firebase library.

kbssa, 2019-07-18:
> Please make sure you don't have or installed SdFat or SD library in libraries folder, try to remove it from libraries folder because it's not consistent with core library that used by this Firebase library.

Removing SdFat library solved.

Thank you, now I can test your library and finish my project.

closed 2019-07-18 by kbssa