mobizt/Firebase-ESP8266 #42: Check for internet connection when only using stream

arlucio, 2019-07-22:
In my project I just create a stream to the database and keep checking if available and reading this stream forever, depending of the stream response, I do read/set values. I never read/set values without a stream response.

Everything works really well, only problem I'm having is when I lose internet connection. Program stays in the loop and there is no sign that I lost internet connection. The way to do it in the other Firebase library was checking Firebase.failed() on the loop and restarting ESP when it is true. Is there anything like this in this library?

I tried some possibilities, httpCode() goes from -4 to -5 if lost connection, but only when I try to set/read value somewhere or use pathExists(), just on stream it doesn't change. Tried also to read httpConnected() but it stayed always on false. Tried to check if there is errors in streamAvaiable() or readStream() when without internet, but also no errors appear.

I made a solution, but don't know if there is a better one. On the loop I check for the path from time to time using pathExists()and when internet connection is off it goes false, also httpCode goes from 200 to -5.

Of course I can just set/read a variable from time to time and check the errors/use pathExists() so not a big problem, but wanted to know if there is something I'm not seeing here.

Btw, thank you @mobizt for all the efforts you put in this, and congratulations to make such a great library, I used to work with the other Firebase library and moved to this when I saw it. So much better, stable and fixes lots of stuff. Really great library 😁

mobizt, 2019-07-22:
The negative value returned from httpCode() is actually the client side error code which can be referred to [this](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266HTTPClient.h#L43).

For streaming and other firebase calls that WiFi  reconnection flag was set to true once in setup with `Firebase.reconnectWiFi (true);`, the library will try to reconnect the WiFi if it was disconnected by call [WiFi.reconnect();](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.cpp#L1742) internally and wait while the status was not connected for 3 seconds then check WiFi status again before exit with the error or continue.

Then you can bypass this blocking code and exit the function immediately  (if the WiFi connection status was disconnected) by call **Firebase.reconnectWiFi (false)** in setup or somewhere that you want to toggle this option.  

This is not include the 5 seconds [timeout](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266HTTPClient.h#L104) in case of WiFi was OK but no server response.

Another approach is to pause/uncaused the Firebase calls by call [firebaseData.pauseFirebase(true) or firebaseData.pauseFirebase(false)](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.h#L2031) to bypass (or resume) Firebase calls upon your decision from WiFi connection status.

arlucio, 2019-07-22:
Ok, important information I forgot to mention is that I'm using tzapu WiFiManager, I believe because of this I'm getting wrong http codes, even when I'm connected and everything is working httpCode() returns -4.

Using WiFi.reconnect(true) really works great, when I lose connection and it comes back, it just starts working again perfectly. My problem is just when it loses connections and the connection never comes back again. Doing some tests here and using this option (on my code) makes httpCode() always stay on -4 no matter the internet connection status.

So I tried WiFi.reconnect(false) and found out that this way httpCode goes to -5 when with no internet connection. So I made a workaround for this and it works great, gonna leave it here for anyone if looking for it in the future. I'm just starting it with reconnect(false) and when I get a httpCode -5 I put it on true until it reconnects, and then put it back to false. 

This is probably just because I'm using WiFiManager and it seems like it breaks the httpCodes.

//on Setup:
errorCount=0;
WiFi.reconnect(false)

//on loop:
    httpCode = firebaseData.httpCode();
    if (httpCode == -5)
    {
      Firebase.reconnectWiFi(true);
      while (httpCode == -5)
      {
        errorCount++;
        Serial.print("Trying to reconnect. Error count: ");
        Serial.println(errorCount);
        delay(2000);
        Firebase.pathExist(firebaseData, path); //necessary to update httpCode
        httpCode = firebaseData.httpCode();
        if (errorCount > 30) // restart in one minute if it don't come back (get on WiFiManager AP mode)
        {
          Serial.println("Restarting...");
          ESP.restart();
        }
      }
    }
    else
    {
      Firebase.reconnectWiFi(false);
      errorCount = 0;
    }

Btw, thanks again for the really quick and helpful response.

mobizt, 2019-07-23:
HTTPC_ERROR_NOT_CONNECTED or -4  for situation where no WiFi connection is OK (ESP8266 was successfully connected to router or AP) but no response from server, which means no internet available, server connection was timed out, something wrong in HTTP request.

HTTPC_ERROR_CONNECTION_LOST or -5  for situation where no WiFi connection which WiFi status was the first thing to check before do anything by using basic condition checking where WiFi.status() != WL_CONNECTED.

WiFiManager does not cause any effect to Firebase library but you should check WiFi status yourself to take control the WiFi, don't let WiFiManager control this by do something that you can't track or don't know.

closed 2019-07-25 by mobizt