mobizt/Firebase-ESP8266 #43: BUG

komputerboy, 2019-07-25:
**Describe the bug**
When i try your Firebase Files example i got this error:
```
Set file data test...
FAILED
REASON: SD card was not available
```
When i try to use SD (esp8266) default example, i am able to write file in SD Card.

And when i try your Firebase File Example, the file1.txt is written in my SD card though, with some weird character. But not uploading in Firebase Server.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Arduino IDE
2. Click on Example Firebase ESP8266>File
3. Upload

**IDE and its version:**
 - Arduino
 - Version 1.8.9

**ESP8266 Arduino Core SDK version**
 - Version 2.5.1

mobizt, 2019-07-25:
Please post your test code and your SD pins connection with ESP8266 or module.

komputerboy, 2019-07-25:
okay, i edited my question

mobizt, 2019-07-25:
D2 or GPIO4 was assigned to the data pin of I2C bus by default (SDK lib), now you use it as CS pin for SPI bus.

The default CS pin for SPI is D8 or GPIO15.

You can change your CS to use GPIO15 or D8 or change the assigned macro in library [here](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.h#L78) to 
```C++
#define SD_CS_PIN 4
```

Also please remove BUG in your issue title, this is the different config issue.

komputerboy, 2019-07-25:
okay, now the error gone, i will close this issue. 
But i still can't upload files, the error i got right now is:

```
Set file data test...
FAILED
REASON: Transmission error, unauthorized
```

mobizt, 2019-07-25:
Check your **FIREBASE_AUTH**.
What is the value return if you call `firebaseData.httpCode()` after `firebaseData.fileTransferError()` ?
And how about the normal Firbase get or set call?

komputerboy, 2019-07-25:
Silly me, i mistaken use Server Key instead of Firebase Secret on Firebase Auth. Now its working
Thanks

closed 2019-07-25 by komputerboy