mobizt/Firebase-ESP8266 #44: BUG Crash on multiple data while making use of firebasestream

JoshNC, 2019-08-03:
**Describe the bug**
I managed to get the firebase realtime database stream working without issue, altough right now when I edit the data in the database quicker (via an app or via browser) the ESP8266 (A Wemos D1 EPS-12E) crashes and reboots itself. The crash log is as follows: 
`Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000adf9 depc=0x00000000

ctx: cont
sp: 3ffffb60 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffd00:  3fffc278 40101fdc 3fffc200 00000022
3ffffd10:  3ffffd20 3ffffe28 53002200 4000050c
3ffffd20:  402191d2 00000030 00000010 ffffffff
3ffffd30:  40209220 3ffffe24 3fff5324 00000000
3ffffd40:  00000043 00000061 ff000000 0000007d
3ffffd50:  00000000 3ffffe42 000000ff 00000000
3ffffd60:  3fff518c 3fff5324 3ffffe50 3fff13cc
3ffffd70:  00000030 3ffef854 0000003d 401004e4
3ffffd80:  3fff1df1 7fffffff 0000003d 4020d154
3ffffd90:  00000000 00000074 00000030 4021964c
3ffffda0:  3fff1df1 00000000 00000000 40219752
3ffffdb0:  00000001 00000000 3fff1d14 402239d3
3ffffdc0:  00000000 00000003 00000030 4002297d
3ffffdd0:  c792f4cb 11b28dbf aff33bfe 3fff13cc
3ffffde0:  00000000 00000030 3ffef854 40219800
3ffffdf0:  3ffffe48 3a226874 3ffe8624 3fff1018
3ffffe00:  3fff13cc 00000030 3ffef854 402198a6
3ffffe10:  00000003 3fff100c 3fff1d14 40223ea8
3ffffe20:  64747570 3a617461 70227b20 3fff518c
3ffffe30:  3fff5324 7fffffff 3ffef854 402198d4
3ffffe40:  3fff13cc 3a226174 7d353532 ceb2a8c0
3ffffe50:  3fff5324 ffffffff 3ffef7a0 402198e8
3ffffe60:  3fff5324 7fffffff 3ffef7a0 4020bdf0
3ffffe70:  3ffefa48 00000000 3ffefac4 00000030
3ffffe80:  40211b40 00000000 00003a98 fffffffe
3ffffe90:  3fffff30 3fff17b4 3fff1804 3fff14f4
3ffffea0:  3ffefcdc 3ffef7a0 00000000 00000000
3ffffeb0:  3ffefcdc 00000000 00000000 fffffffe
3ffffec0:  3fff3fcc 3ffefa48 00000000 00000000
3ffffed0:  3fff13cc 0000000a 3ffef834 00000001
3ffffee0:  3ffef840 00000006 00000000 00000001
3ffffef0:  00000001 00000000 000930df 00000001
3fffff00:  3ffef850 00000000 3ffef7a0 402196a2
3fffff10:  3fffff30 00000032 00000032 00000032
3fffff20:  00000000 3ffefa48 3ffef7a0 4020c8e8
3fffff30:  40211b40 00000000 00003a98 3ffefac4
3fffff40:  3ffef8a4 3fff17b4 3fff1804 3fff14f4
3fffff50:  3ffefcdc 3ffefcdc 3fffff90 4020cd71
3fffff60:  3fff0544 40207e28 3ffefa98 3ffefac4
3fffff70:  3ffef7a0 3ffefa48 3ffefa30 4020c98b
3fffff80:  3fffdad0 00000000 3ffef7a0 40207668
3fffff90:  00000000 00000000 00000000 00000000
3fffffa0:  00000000 00000000 3ffefabc 402076c8
3fffffb0:  3fffdad0 00000000 3ffefabc 4020d350
3fffffc0:  feefeffe feefeffe 3ffe8624 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld`

With using the Exception decoder of Arduino I get the following decoded stack:
`Decoding stack results
0x402191d2: strstr at ../../../.././newlib/libc/string/strstr.c line 69
0x4021964c: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 619
0x40219752: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 479
0x402239d3: basic_read at ssl/tls1.c line 1374
0x40219800: std::string::_M_replace_safe(unsigned int, unsigned int, char const*, unsigned int) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 685
0x402198a6: std::string::assign(char const*, unsigned int) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 276
0x40223ea8: ssl_read at ssl/tls1.c line 314
0x402198d4: std::string::assign(char const*) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 1132
0x402198e8: std::string::operator=(char const*) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 555
0x402196a2: std::basic_string  , std::allocator  >::~basic_string() at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 539`

I don't know who igrokhotkov is but the exception decoder says the above.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the standard firebase stream
2. change 3 or more datatypes at the same time
3. crash

**Expected behavior**
I expect the ESP not to crash.

**IDE and its version:**
 - VSCODE, PlatformIO
 - Firebase ESP8266 client 2.1.2

mobizt, 2019-08-03:
First, update the library to latest version.
Then test and post your working code that expect to be issue.

JoshNC, 2019-08-03:
Just updated to version 2.1.6 as you said. Still crashes on me sadly..

`Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000bf21 depc=0x00000000

ctx: cont
sp: 3ffffb50 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffcf0:  3fffc278 40101fdc 3fffc200 00000022
3ffffd00:  3ffea770 00000076 ff000000 0000007d
3ffffd10:  4021b6c5 00000030 00000018 ffffffff
3ffffd20:  4020a76c 3ffffe13 3fff5454 00000061
3ffffd30:  00000065 00000070 ff000000 0000007d
3ffffd40:  00000000 3ffffe32 000000ff 00000000
3ffffd50:  3fff1ad4 3fff5454 3ffffe40 3fff14fc
3ffffd60:  0000002c 3ffef964 00000039 401004e4
3ffffd70:  3fff1ef9 7fffffff 00000039 4020f2bc
3ffffd80:  0000ff00 00ff0000 0000002c 4021bb48
3ffffd90:  3fff1ef9 00000000 00000000 4021bc4e
3ffffda0:  00000022 00000000 3fff1e1c 40225ec7
3ffffdb0:  00000000 00000003 0000002c f732bb2b
3ffffdc0:  16cf6584 a6f019d2 3ddc3ccf 3fff14fc
3ffffdd0:  00000000 0000002c 3ffef964 4021bcfc
3ffffde0:  3ffffe38 3a226874 3ffe865c 3fff5700
3ffffdf0:  3fff14fc 0000002c 3ffef964 4021bda2
3ffffe00:  00000003 3fff56f4 3fff1e1c 4022639c
3ffffe10:  64747570 3a617461 70227b20 3fff1ad4
3ffffe20:  3fff5454 7fffffff 3ffef964 4021bdd0
3ffffe30:  3fff14fc 3a226174 7d343731 3fffefb0
3ffffe40:  3fff5454 000000ff 3ffef8b0 4021bde4
3ffffe50:  3fff5454 7fffffff 3ffef8b0 4020d9b0
3ffffe60:  3fffc200 40106bac 00000020 40106bf4
3ffffe70:  40213fa8 00000000 00003a98 4000050c
3ffffe80:  3fffff20 3fff18b4 3fff1d9c 3fff18ec
3ffffe90:  40208fd0 00000001 4020f498 00000000
3ffffea0:  3ffefe0c 00000000 00000000 fffffffe
3ffffeb0:  3fff52bc 3ffefb58 00000001 3fff14fc
3ffffec0:  00000000 00000000 3ffef944 00000001
3ffffed0:  00000000 3ffef950 0000000c 00000001
3ffffee0:  00000001 000035c0 00000000 3ffef960
3ffffef0:  00000000 00000001 3ffef8b0 4021bb9e
3fffff00:  3fffff20 00000032 00000032 00000032
3fffff10:  00000000 3ffefb58 3ffef8b0 4020e4b0
3fffff20:  40213fa8 00000000 00003a98 4020266a
3fffff30:  3ffef9b4 3fff18b4 3fff1d9c 3fff18ec
3fffff40:  3ffefe0c 3ffefe0c 3ffe865c 3ffefc40
3fffff50:  3fff0674 402093b8 4020f42c 3ffefbec
3fffff60:  3ffef8b0 3ffefb58 3ffefb40 4020e553
3fffff70:  3fffdad0 3ffef8b0 3ffefbe4 40208b38
3fffff80:  00000000 00000000 00000000 00000000
3fffff90:  00000000 00000000 00000000 3ffefbec
3fffffa0:  3fffdad0 00000000 3ffefbe4 40208fd0
3fffffb0:  3fffdad0 00000000 3ffefbe4 4020f4b8
3fffffc0:  feefeffe feefeffe 3ffe865c 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld`

`0x4021b6c5: strstr at ../../../.././newlib/libc/string/strstr.c line 62
0x4021bb48: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 619
0x4021bc4e: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 479
0x40225ec7: basic_read at ssl/tls1.c line 1374
0x4021bcfc: std::string::_M_replace_safe(unsigned int, unsigned int, char const*, unsigned int) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 685
0x4021bda2: std::string::assign(char const*, unsigned int) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 276
0x4022639c: ssl_read at ssl/tls1.c line 314
0x4021bdd0: std::string::assign(char const*) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 1132
0x4021bde4: std::string::operator=(char const*) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 555
0x4021bb9e: std::basic_string  , std::allocator  >::~basic_string() at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 539`

mobizt, 2019-08-03:
Post your sketch.

JoshNC, 2019-08-03:
Sketch:
`#include <Arduino.h>
#include <string>
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

#define JSON_SIZE 323 // calculated the size based on the json output from Firebase https://arduinojson.org/v6/assistant/

#define RELAY_1 D9
#define RELAY_2 D8
#define RELAY_3 D7
#define RELAY_4 D6
#define RELAY_5 D5
#define RELAY_6 D4
#define RELAY_7 D3
#define RELAY_8 D2

#define RELAY_1_PATH "/relays/relay_1"
#define RELAY_2_PATH "/relays/relay_2"
#define RELAY_3_PATH "/relays/relay_3"
#define RELAY_4_PATH "/relays/relay_4"
#define RELAY_5_PATH "/relays/relay_5"
#define RELAY_6_PATH "/relays/relay_6"
#define RELAY_7_PATH "/relays/relay_7"
#define RELAY_8_PATH "/relays/relay_8"

#define LED_1_R "/ledstrip-1/r"
#define LED_1_G "/ledstrip-1/g"
#define LED_1_B "/ledstrip-1/b"
#define LED_1_BRIGHTNESS "/ledstrip-1/brightness"
#define LED_MAX_BRIGHTNESS 255

#define RELAY_ON HIGH
#define RELAY_OFF LOW

#define FIREBASE_HOST "cardozo-homenetwork.firebaseio.com"
#define FIREBASE_AUTH "KANdyvCh2wI6MAu4mgq0s3DlsVnIksKj81dxsHfP"
#define WIFI_SSID "Ziggo28FC5D2"
#define WIFI_PASSWORD "kvbCakjv5zyh"

StaticJsonDocument<JSON_SIZE> doc;

FirebaseData firebaseData;

struct Relay {
  int pin;

  String path;
};

struct LedStrip {
  int rPin;
  int gPin;
  int bPin;
  int brightnessValue;
  int rValue;
  int gValue;
  int bValue;
};

struct Relay relay1;
struct Relay relay2;
struct Relay relay3;
struct Relay relay4;
struct Relay relay5;
struct Relay relay6;
struct Relay relay7;
struct Relay relay8;

[... 211 lines cut ...]
  resetRelay(&relay4, RELAY_4_PATH, RELAY_4);
  resetRelay(&relay5, RELAY_5_PATH, RELAY_5);
  resetRelay(&relay6, RELAY_6_PATH, RELAY_6);
  resetRelay(&relay7, RELAY_7_PATH, RELAY_7);
  resetRelay(&relay8, RELAY_8_PATH, RELAY_8);

  relays[0] = relay1; 
  relays[1] = relay2; 
  relays[2] = relay3; 
  relays[3] = relay4; 
  relays[4] = relay5; 
  relays[5] = relay6; 
  relays[6] = relay7; 
  relays[7] = relay8; 

  for (struct Relay i : relays) {
    pinMode(i.pin, OUTPUT);
    delay(100);
  }
}

void resetRelay(struct Relay *r, String path, int pin) {
  r->path = path;
  r->pin = pin;
}`

mobizt, 2019-08-03:
It's better, if you pin point where is the possible lines of code that cause the issue.

JoshNC, 2019-08-03:
Just edited the loop for you:
 
`void loop() {
  handleDataStream();
}

void handleDataStream() {
  if (!Firebase.readStream(firebaseData)) {
    Serial.println("Stream error:");
    Serial.println(firebaseData.errorReason());
  }

  if (firebaseData.streamTimeout()) {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable()) {
    // CRASHES before it can print  "incomming data..."
    Serial.println("\nIncomming data...");
  }
}`

JoshNC, 2019-08-03:
If I change something by hand in firebase it works fine. But when I change 3 or more within half a second it crashes.

mobizt, 2019-08-03:
It's not caused by library by that basic usage.
You need to check free heap to make sure your no memory leak.

JoshNC, 2019-08-03:
It gives an exception(3) which could lead to a memory leak which is true... any advice on how to tackle that problem?

mobizt, 2019-08-03:
Where is the line that you "send messages"?
You need to print ESP.getFreeHeap() before and after some function call or anywhere you want.

JoshNC, 2019-08-03:
This ESP just listens to the firebase without sending messages to firebase. I will search around with the ESP.getFreeHeap() function and see if I can find any memory leaks.

Thanks for all the help you've given me!

mobizt, 2019-08-03:
Make sure you test long enough and under all possible cases.

mobizt, 2019-08-03:
Be careful you using Wemos D1 which it's pins assignment are redundant.
Check [this](https://github.com/esp8266/Arduino/blob/master/variants/d1/pins_arduino.h#L38).

For variable declaration

Use
```C++
 Relay relay1;
 Relay relay2;
 Relay relay3;
 Relay relay4;
 Relay relay5;
 Relay relay6;
 Relay relay7;
 Relay relay8;

 Relay relays[8];

 LedStrip ledStrip;
```

instead of
```C++
struct Relay relay1;
struct Relay relay2;
struct Relay relay3;
struct Relay relay4;
struct Relay relay5;
struct Relay relay6;
struct Relay relay7;
struct Relay relay8;

struct Relay relays[8];

struct LedStrip ledStrip;
```

JoshNC, 2019-08-03:
I was aware of that yes! I heard from a friend of mine it is possible to fix the issue with the pins but I have to search for it. If I cannot find it I'm probably going to use another ESP for the leds and seperate the leds from the relays. Thanks for the notice!

Is declaring variables in such a manner better for heap size? If I want to declare it like you said I should then I should typedef the struct of the relays and ledstrip right?

mobizt, 2019-08-03:
Not sure I understand your last paragraph.
It's not relate to heap but for valid syntax which prevent unexpected result when it compiled by compiler.

JoshNC, 2019-08-03:
Been testing around with ESP.getFreeHeap(). I start around 43528 bytes of free heap. After setting up the firebase I have 28384 bytes free. After making a ton of calls to the firebase with an interval of 2 seconds per call the free heap remains, +-100 bytes off, the same. The moment I make the interval between the calls quicker it starts to be unstable. At 1.5 seconds it crashes 10% of the time and it grows exponentionally after that. At 1 second per call it's around 30% of the time, 500ms it's around 70% of the calls and without interval it crashes 100% of the time.

mobizt, 2019-08-03:
The microcontroller programming is  different from PC which you can easily debug. 
You need to track by serial print at any check point. 

The decoded stacks are not show anything relate to the main tasks i.e. setup() and loop() which means it might not cause by program.

JoshNC, 2019-08-03:
Also found something else out. After testing and a lot of crashing it seems that the program crashes right before printing the heap size in the following piece of code:

`
if (firebaseData.streamAvailable()) {
    Serial.println("\nIncomming data...");
    Serial.println("Heapsize before receiving data: ");
    Serial.println(system_get_free_heap_size());
`

mobizt, 2019-08-03:
Same stack trace result?
Can you test your sketch with Ardduino IDE? and use [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder
) from me-no-dev.

The stack trace result might be different.

JoshNC, 2019-08-03:
Yes just done it. The Stack trace is way different actually.

Decoded stack:
```
0x402017aa: FirebaseESP8266::errorToString(int, std::string&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3704
0x4020c029: Print::write(char const*) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Print.h line 60
0x40202f24: FirebaseData::errorReason() at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 5040
0x4020c100: Print::println(char const*) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Print.cpp line 190
0x4020119c: loop() at C:\Users\joshu\Documents\Arduino\FirebaseStreamingWemosD1R2Mini/FirebaseStreamingWemosD1R2Mini.ino line 43
0x4020c7a1: esp_schedule() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020c830: loop_wrapper() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125```

mobizt, 2019-08-03:
Try to comment out firebaseData.errorReason() from your code.

JoshNC, 2019-08-03:
Did that the moment you posted it. Got a new stack now. 

Decoded stack:
```
0x4021e1ae: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4020ecc0: malloc(size_t) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x4021e1ae: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4021e3ac: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 621
0x4021e4b2: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 481
0x4022a2eb: basic_read at ssl/tls1.c line 1374
0x4021e560: std::string::_M_replace_safe(unsigned int, unsigned int, char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 687
0x4021e606: std::string::assign(char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 278
0x4021e634: std::string::assign(char const*) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 1134
0x4020c76c: esp_yield() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4021e648: std::string::operator=(char const*) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 557
0x40204dcb: FirebaseESP8266::getServerResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2248
0x40205d5f: ESP8266WiFiSTAClass::status() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 628
0x4021e402: std::basic_string  , std::allocator  >::~basic_string() at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 541
0x40205778: FirebaseESP8266::getServerStreamResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3063
0x4020bd78: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x4020bd78: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x40205d50: ESP8266WiFiSTAClass::status() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020581b: FirebaseESP8266::readStream(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1311
0x4020118b: loop() at C:\Users\joshu\Documents\Arduino\FirebaseStreamingWemosD1R2Mini/FirebaseStreamingWemosD1R2Mini.ino line 41
0x4020c78d: esp_schedule() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020c81c: loop_wrapper() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

Line 41 in the loop is the line which says: ```C++ if (!Firebase.readStream(firebaseData)) { ```

mobizt, 2019-08-03:
Found som improper use of basic string join with progmem chars array in my code that works with Core SDK v2.5.x and can cause error in lower version.

Try to edit file [FirebaseESP8266.cpp line 5042 and 5043](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.cpp#L5042) from

```C++
buff += ESP8266_FIREBASE_STR_132;
buff += ESP8266_FIREBASE_STR_6;
```
to
```C++
char *str = new char[100];
memset(str, 0, 100);
strcpy_P(str, EESP8266_FIREBASE_STR_132);
buff +=str;
memset(str, 0, 100);
strcpy_P(str, EESP8266_FIREBASE_STR_6);
buff +=str;
delete[] str;
```

JoshNC, 2019-08-03:
Alright, done that. It now looks like the following:
```C++
if (_firebaseError != "")
    {
        //buff += ESP8266_FIREBASE_STR_132;
        //buff += ESP8266_FIREBASE_STR_6;
		char *str = new char[100];
		memset(str, 0, 100);
		strcpy_P(str, ESP8266_FIREBASE_STR_132);
		buff +=str;
		memset(str, 0, 100);
		strcpy_P(str, ESP8266_FIREBASE_STR_6);
		buff +=str;
		delete[] str;
        buff += _firebaseError;
    }
```

I compiled the code successfully and uploaded it on the ESP. Once uploaded it throws an occaision stack trace.

decoded stack trace:
```
0x402017aa: FirebaseESP8266::errorToString(int, std::string&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3704
0x4020c065: Print::write(char const*) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Print.h line 60
0x40202f27: FirebaseData::errorReason() at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 5040
0x40205da0: ESP8266WiFiSTAClass::status() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020c0f0: Print::println() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Print.cpp line 178
0x4020119c: loop() at C:\Users\joshu\Documents\Arduino\FirebaseStreamingWemosD1R2Mini/FirebaseStreamingWemosD1R2Mini.ino line 43
0x4020c7dd: esp_schedule() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020c86c: loop_wrapper() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

Line 43 points to the:

```C++
Serial.println(firebaseData.errorReason());
```

Also when I send multiple data lines via the stream I get another stack trace, different one though.

Decoded stacktrace:
```
0x4021e1fe: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4020ed10: malloc(size_t) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x4021e1fe: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4021e3fc: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 621
0x4021e502: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 481
0x4022a33b: basic_read at ssl/tls1.c line 1374
0x4021e5b0: std::string::_M_replace_safe(unsigned int, unsigned int, char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 687
0x4021e656: std::string::assign(char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 278
0x4021e684: std::string::assign(char const*) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 1134
0x4020c7bc: esp_yield() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4021e698: std::string::operator=(char const*) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 557
0x40204e1b: FirebaseESP8266::getServerResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2248
0x4021e452: std::basic_string  , std::allocator  >::~basic_string() at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 541
0x402057c8: FirebaseESP8266::getServerStreamResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3063
0x4020bdc8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x4020bdc8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x40205da0: ESP8266WiFiSTAClass::status() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020586b: FirebaseESP8266::readStream(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1311
0x40201188: loop() at C:\Users\joshu\Documents\Arduino\FirebaseStreamingWemosD1R2Mini/FirebaseStreamingWemosD1R2Mini.ino line 41
0x4020c7dd: esp_schedule() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020c86c: loop_wrapper() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

mobizt, 2019-08-03:
At this time you can print firebaseData.httpCode() instead of error reason.

The value of above code described in this [header file from line 43 to 91](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266HTTPClient.h#L43).

What is ESP8266 Arduino Core SDK version you installed (the esp8266 installed via boards manager)?

JoshNC, 2019-08-03:
The errorcode I get from using firebaseData.httpCode() is: -4. According to the file you mentioned earlier this means it is the following error: HTTPC_ERROR_NOT_CONNECTED. Which is weird because when I update data on firebase it gets displayed in the monitor, so it should be connected?

The ESP8266 Arduino Core SDK version is: 2.5.2

JoshNC, 2019-08-03:
Also right now I get the following stacktrace when updating multiple lines in my firebase database:

```
0x40227fe8: mem_malloc at core/mem.c line 210
0x4020ed94: malloc(size_t) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x40220a98: lwip_cyclic_timer at core/timeouts.c line 233
0x40227fe8: mem_malloc at core/mem.c line 210
0x4010014c: millis() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 186
0x4021f3d8: do_memp_malloc_pool at core/memp.c line 254
0x4021f414: memp_malloc at core/memp.c line 356
0x40220a98: lwip_cyclic_timer at core/timeouts.c line 233
0x40220a3d: sys_timeout_abs at core/timeouts.c line 189
0x40220a98: lwip_cyclic_timer at core/timeouts.c line 233
0x40220acb: lwip_cyclic_timer at core/timeouts.c line 261
0x4021f43e: memp_free at core/memp.c line 447
0x40220c40: sys_check_timeouts at core/timeouts.c line 390
0x4020c7fb: loop_task(ETSEvent*) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 133
0x4022e058: bi_subtract at crypto/bigint.c line 331
0x4022e04a: bi_subtract at crypto/bigint.c line 324
0x4022dc92: bi_free at crypto/bigint.c line 219
0x4022dc92: bi_free at crypto/bigint.c line 219
0x40232cd1: SHA256_Final at crypto/sha256.c line 273
0x4022f094: hmac_sha256_v at crypto/hmac.c line 165
0x40232c61: SHA256_Final at crypto/sha256.c line 254
0x4022f094: hmac_sha256_v at crypto/hmac.c line 165
0x40232cd1: SHA256_Final at crypto/sha256.c line 273
0x4022f094: hmac_sha256_v at crypto/hmac.c line 165
0x40234094: add_hmac_digest at ssl/tls1.c line 793
0x4022a83c: basic_read at ssl/tls1.c line 1472
0x40201ad5: FirebaseESP8266::strpos(char const*, char const*, int) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4669
0x40201ad5: FirebaseESP8266::strpos(char const*, char const*, int) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4669
0x4020c840: esp_yield() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4020c840: esp_yield() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4020c922: __yield() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 100
0x40204717: FirebaseESP8266::getServerResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1958
0x4021e4da: std::basic_string  , std::allocator  >::~basic_string() at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 541
0x402057c4: FirebaseESP8266::getServerStreamResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3063
0x4020bdc4: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x4020bdc4: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x40205d9c: ESP8266WiFiSTAClass::status() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x40205867: FirebaseESP8266::readStream(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1311
0x40201189: loop() at C:\Users\joshu\Documents\Arduino\FirebaseStreamingWemosD1R2Mini/FirebaseStreamingWemosD1R2Mini.ino line 41
0x4020c861: esp_schedule() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020c8f0: loop_wrapper() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

mobizt, 2019-08-03:
Your original sketch, you use blocking code in loop (many delays for total time about 1.4 seconds) which will block the library to read the stream.

The return http code I was described [here](https://github.com/mobizt/Firebase-ESP8266/issues/42#issuecomment-514040413).

JoshNC, 2019-08-03:
Yes alright, but the stack trace I showed you in the comment above is from the sketch via arduino without blocking the stream. Also this is where I get the error code.

Code I used in the arduino IDE with above stack trace:
```C++
FirebaseData firebaseData;

void setup() {
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }

  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData, "stream-data")) {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop() {
  if (!Firebase.readStream(firebaseData)) {
    Serial.println("Stream error:");
//    Serial.println(firebaseData.errorReason());
    Serial.println(firebaseData.httpCode());
  }

  if (firebaseData.streamTimeout()) {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable()) {
    Serial.println("\nIncomming data...");
    Serial.println(firebaseData.dataType());
    Serial.println(firebaseData.dataPath());
  }
}
```

mobizt, 2019-08-03:
With your last basic sketch, how you know **from device** that the data was changed? Not from firebae console or any app.

JoshNC, 2019-08-03:
As you can see in the basic sketch code I have the following lines:
```C++
if (firebaseData.streamAvailable()) {
    Serial.println("\nIncomming data...");
    Serial.println(firebaseData.dataType());
    Serial.println(firebaseData.dataPath());
  }
```
When I update the firebase it shows me the type and the path in firebase.

mobizt, 2019-08-03:
And **updating multiple lines** means update multiple nodes (tree) in Firebase console ?

JoshNC, 2019-08-03:
Yes, pardon me. Updating multiple nodes in the firebase console. This gives me the following stack:

```
0x4021e286: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4020ed94: malloc(size_t) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x4021e286: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4021e484: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 621
0x4021e58a: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 481
0x4022a3c3: basic_read at ssl/tls1.c line 1374
0x4021e638: std::string::_M_replace_safe(unsigned int, unsigned int, char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 687
0x4021e6de: std::string::assign(char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 278
0x4021e70c: std::string::assign(char const*) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 1134
0x4020c840: esp_yield() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4021e720: std::string::operator=(char const*) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 557
0x40204e17: FirebaseESP8266::getServerResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2248
0x4021e4da: std::basic_string  , std::allocator  >::~basic_string() at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 541
0x402057c4: FirebaseESP8266::getServerStreamResponse(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3063
0x4020bdc4: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x4020ef78: free(void*) at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x40205d9c: ESP8266WiFiSTAClass::status() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x40205867: FirebaseESP8266::readStream(FirebaseData&) at C:\Users\joshu\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1311
0x40201189: loop() at C:\Users\joshu\Documents\Arduino\FirebaseStreamingWemosD1R2Mini/FirebaseStreamingWemosD1R2Mini.ino line 42
0x4020c861: esp_schedule() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020c8f0: loop_wrapper() at C:\Users\joshu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

In which line 42 is:
```C++
if (!Firebase.readStream(firebaseData)) { }
```

mobizt, 2019-08-03:
Try this

```C++
if (firebaseData.streamAvailable()) {
    Serial.println("\nIncomming data...");
    Serial.println(firebaseData.dataType());
    Serial.println(firebaseData.dataType());
    Serial.println(firebaseData.payload());
  }
```

JoshNC, 2019-08-03:
That prints the value which I fill in the node on Firebase. But I see after 3 times getting a stream error: -4 the whole connection stops and doesn't reconnect. Tried using the Firebase.setMaxRetry(firebaseData, 10); but that doesn't seem to work.

mobizt, 2019-08-03:
I suspected that the data may be truncated due to [this limit](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.h#L76).  Try to change to lager value as

`#define FIREBASE_RESPONSE_SIZE 1000`

JoshNC, 2019-08-03:
I will try that, thanks. What I'm doing is updating 3 nodes at the same time each of them has an integer value between 0 and 255. (RGB values).

JoshNC, 2019-08-03:
It also seems to happen when I rapidly change a value of a node in my realtime firebase database.

mobizt, 2019-08-03:
How can you updating 3 or more nodes at the same time? 
Is rapidly change mean you edit value in Firebase console and press enter?

JoshNC, 2019-08-03:
I made an android app which is running on my phone which writes data to nodes on firebase. I made a function which writes the RGB values of the selected color to the r, g, b nodes. Also to test out if it was just because I changed 3 at the same time I made a function, which doesnt affect the other function, which writes the RGB value of a selected color as one integer and then write it on a node. 

When quickly changing the second function crashes the ESP or when writing 3 values at the same it crashes.

mobizt, 2019-08-03:
Now I also get same strange error too (exception (3) error).  

May firebase update its response payload or relate to ESP8266 core SDK. I will find the root cause and let you know soon.

JoshNC, 2019-08-03:
Alright, glad that I'm not the only one with this error hahaha. I hope you'll find out the root of the problem soon!

mobizt, 2019-08-04:
Now I fix the issues and update the stream callback.
The issues are from improper use of flash string length which cause error in SDK v2.5.2.

JoshNC, 2019-08-04:
Ahh alright! Let me know when you fixed the issue! Thanks a lot!

closed 2019-08-04 by mobizt

JoshNC, 2019-08-04:
New version works via the Arduino IDE, quick last question: Do you know when does the new version update on platform io in VSCode?

mobizt, 2019-08-04:
Not more than 2 days