mobizt/Firebase-ESP8266 #47: problem when using FCM and real time database in same project

Sheimy, 2019-08-10:
when i try the following code 

/*
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2019 mobizt
 *
*/

//FirebaseESP8266.h must be included before ESP8266WiFi.h
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "gold-81cd8.firebaseio.com"
#define FIREBASE_AUTH "VRkzeI3SfKrsyMXWuCKkS3i062d0FhdW80DEkzlh"
#define FIREBASE_FCM_SERVER_KEY "AAAAyqlh9Qg:APA91bF65fdm5IOn3EhWFNL3Pg3knu02S6c-KwLgh-n55UCq1l6krPN1jytk6U8Us5uSJbBSHKJ9zF-Z60g2HOvuXVKbaJLq5ANzLMhNo6NpHNnGHpyDa8FSN-Yp_u9L0Yv9dkIHuAvv"

//#define WIFI_SSID "Yasmaina2"
//#define WIFI_PASSWORD "%@hmed$heimy2019#"

#define WIFI_SSID "Taha Shemi"
#define WIFI_PASSWORD "taha#shemi"

#define AC_Relay D5
//Define FirebaseESP8266 data object
FirebaseData firebaseData;
int ALM0,ACR;
String path = "/GOLD";

unsigned long lastTime = 0;

int count = 0;

void sendMessage();

void setup()
{

  Serial.begin(115200);
  pinMode(D5, OUTPUT);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  firebaseData.fcm.begin(FIREBASE_FCM_SERVER_KEY);
  
  //Set database read timeout to 1 minute (max 15 minutes)
  Firebase.setReadTimeout(firebaseData, 1000 * 60);
  //tiny, small, medium, large and unlimited.
  //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
  Firebase.setwriteSizeLimit(firebaseData, "tiny");
  
  /*
  This option allows get and delete functions (PUT and DELETE HTTP requests) works for device connected behind the
  Firewall that allows only GET and POST requests.
  
  Firebase.enableClassicRequest(firebaseData, true);
  */
    firebaseData.fcm.setPriority("high");

    firebaseData.fcm.setTimeToLive(1000);
  
  Firebase.setInt(firebaseData, path + "/ALARM/ALM1", 0);
  Firebase.setInt(firebaseData, path + "/CONTROL/AC", 0);
}

void loop()
{
    if (millis() - lastTime > 60 * 100)
    {
        lastTime = millis();

  
 if (Firebase.getInt(firebaseData, path + "/CONTROL/AC"))
    {
      Serial.println("PASSED");
      Serial.println("PATH: " + firebaseData.dataPath());
      Serial.println("TYPE: " + firebaseData.dataType());
      Serial.println("ETag: " + firebaseData.ETag());
      Serial.print("VALUE: ");
      ACR= firebaseData.intData();
      Serial.println(ACR);
 if (ACR==1) {
  Serial.println("LED ON");
  digitalWrite(D5,HIGH); 
  sendMessage();
  }
 else {
Serial.println("LED OFF");
digitalWrite(D5,LOW);  
}  
    }
    }
}

void sendMessage()
{

    Serial.println("------------------------------------");
    Serial.println("Send Firebase Cloud Messaging...");

    firebaseData.fcm.setNotifyMessage("Notification", "Hello World! " + String(count));

    firebaseData.fcm.setDataMessage("{\"myData\":" + String(count) + "}");

    firebaseData.fcm.setTopic("GOLD");
    //if (Firebase.broadcastMessage(firebaseData1))
    if (Firebase.sendTopic(firebaseData))
    //if (Firebase.sendMessage(firebaseData1, 0))//send message to recipient index 0
    {

        Serial.println("PASSED");
        Serial.println(firebaseData.fcm.getSendResult());
        Serial.println("------------------------------------");
        Serial.println();
    }
    else
    {
        Serial.println("FAILED");
        Serial.println("REASON: " + firebaseData.errorReason());
        Serial.println("------------------------------------");
        Serial.println();
    }

    count++;
}

it cant send FCM  it gives me FAILED  bad request 
this happen in 2.3 version but in 2.2 there is no problem 
also how to send data message only without notification message

mobizt, 2019-08-11:
Now the FCM connection refused was fixed and let you send data and notification messages separately and also able to send blank message without notification and data messages too.

Please update to the latest version, 2.3.1.

Sheimy, 2019-08-11:
the data messages is now ok  i can send data message  and notification also without any problem 
but the problem is that  when i try to send message like the following by mixing the basic and the cloud message example 

if (Firebase.getInt(firebaseData, path + "/CONTROL/AC"))
    {
      
      ACR= firebaseData.intData();
      Serial.println(ACR);
 if (ACR==1) {
  Serial.println("LED ON");
  digitalWrite(D5,HIGH); 
  sendMessage();
  }
 else {
Serial.println("LED OFF");
digitalWrite(D5,LOW);  
}  

i got the following
 Send Firebase Cloud Messaging...
00:38:22.828 -> FAILED
00:38:22.828 -> REASON: bad request
00:38:22.828 -> --------------------

mobizt, 2019-08-12:
Thank you for report this. The issue caused by some bypass code that I add during test and forgot to remove it when publish.

Please update library again, the version still be 2.3.1.

closed 2019-08-12 by mobizt

Sheimy, 2019-08-12:
perfect it is working good now
thanks