mobizt/Firebase-ESP8266 #48: PlatformIO build failed. v.2.3.1

Gizmodo, 2019-08-12:
**Describe the bug**

C:\Users\user\.platformio\lib\Firebase ESP8266 Client_ID6247\src\FirebaseESP8266.cpp: In member function 'void FirebaseESP8266::setClock()':
C:\Users\user\.platformio\lib\Firebase ESP8266 Client_ID6247\src\FirebaseESP8266.cpp:4885:30: error: 'time' was not declared in this scope
     time_t now = time(nullptr);
                              ^
C:\Users\user\.platformio\lib\Firebase ESP8266 Client_ID6247\src\FirebaseESP8266.cpp:4899:15: error: aggregate 'tm timeinfo' has incomplete type and cannot be defined
     struct tm timeinfo;
               ^
C:\Users\user\.platformio\lib\Firebase ESP8266 Client_ID6247\src\FirebaseESP8266.cpp:4900:29: error: 'gmtime_r' was not declared in this scope
     gmtime_r(&now, &timeinfo);
                             ^
*** [.pio\build\esp12e\lib573\Firebase ESP8266 Client_ID6247\FirebaseESP8266.cpp.o] Error 1

**To Reproduce**
Steps to reproduce the behavior:
1. Build any example

**Expected behavior**
Successful build

**IDE and its version:**
 - PlatformIO
 - Version 2.2.0

**ESP8266 Arduino Core SDK version**
 - Version 2.5.1

**Additional context**
If I downgrade to 2.3.0 - no build errors.

mobizt, 2019-08-12:
I fixed for missing time.h header file. Please update the library.

Gizmodo, 2019-08-12:
PlatformIO still see old version 2.3.1 (issue https://github.com/platformio/platformio-api/issues/23).
I tried to apply your commit manually, but the error still here.

mobizt, 2019-08-12:
Actually time.h is in the core library. Any third party libraries contained time.h?

Check and search third party lib for time.h, temporary remove it or update platformio again.

mobizt, 2019-08-12:
The code that you've got the error is just for get NTP server time.
This required for BearSSL SSL/TLS handshake process.
If you can't run the below code, the problem may come from time.h include file conflicts.
```C++
    configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");
    time_t now = time(nullptr);
    while (now < 8 * 3600 * 2)
    {
        delay(100);
        now = time(nullptr);
    }

    struct tm timeinfo;
    gmtime_r(&now, &timeinfo);
```

Gizmodo, 2019-08-12:
Thank you. The problem was with https://github.com/PaulStoffregen/Time library. It has Time.h.
I renamed it to TimeLib.h and build succeessfuly.
P.S. Arduino IDE does have this problem with 2.3.1.

mobizt, 2019-08-12:
I also don't understand why well known dev don't care about naming.

closed 2019-08-12 by mobizt