mobizt/Firebase-ESP8266 #49: HELP

kbssa, 2019-08-21:
Every time I put something into firebase from "inside the esp8266", the stream reads the value back.

I want that the stream only reads the payload only when the data is changed from "outside", from an app for example.

Is there a way of doing that ?

Thank you in advance

mobizt, 2019-08-21:
Any Firebase client (web app, mobile app, console and esp) makes the data at the node path that being monitor by other clients, those clients will be informed that change if they are still connected in stream mode.

Then you requirement is not suitable for Firebase Realtime Database system but you can add you own user identifier to the data under your stream path to distinguish who changed the data. When client get notify the stream changed, he will check the user identifier to process or ignore that data.

kbssa, 2019-08-21:
Ok, I will try something.

Thank you for your fast reply,

closed 2019-08-21 by kbssa