mobizt/Firebase-ESP8266 #51: Aleatory crashes when calling setString()

kbssa, 2019-09-06:
Hello mobizt,

My code is almost running stable, it is not yet, because sometimes it crashes with wdt reset.

The function that is suppose to be causing the crash is the setString().

I know that, because when I run my code with the function commented out it runs really stable even within a stress test.

Is there something I can do to turn my code really stable ?

Thanks in advance

mobizt, 2019-09-06:
Depending on your code, you should check free heap frequency somewhere on your code to debug for memory leak which can cause mcu to reset.

kbssa, 2019-09-06:
Understood, but only the line of code below causes the crash.

Firebase.setString(updateFirebaseData, path + _nodeName + deviceFuction + String(_channel), _value);

If I comment it out the code runs pretty good.

Anyway I will look for memory leak.

mobizt, 2019-09-06:
Every data must be convert to string internally for almost set function and send as payload to server then setString() is not the causes because it directly send.

ericpaepe, 2019-09-06:
Hello mobizt,

Running v 2.4.2 on 3 Esp8266's gave on all 5 to 10 "Wdt reset rst cause:4"/day. With v2.4.1 this seems OK.
I'm using : Firebase.pushJSON(firebaseData,"metingCV", message); in my code.
I will try the Generic ESP8266 Module , now I use NodeMCU 1.0(ESP-12E module).
Thanks for sharing your code.

Eric

mobizt, 2019-09-07:
@ericpaepe 
If you choose Generic ESP8266 Module, don't forget to choose Flash size.

I recommend to update to Core SDK to v2.5.2 because you will gain much more ram because BearSSL has better memory management which allow the library to reserve the known amount of ram for it.

closed 2019-09-07 by mobizt