mobizt/Firebase-ESP8266 #53: how to connect to a open database

poli44, 2019-09-17:
hi,
how to connect to a open database without providing a parameter "FIREBASE_AUTH". Like as Firebase.begin(FIREBASE_HOST);

{
  "rules": {
    ".read": "auth == null",
    ".write": "auth == null"
  }
}

Best regards

mobizt, 2019-09-17:
No you can't.

**Database secret** used in **FIREBASE_AUTH** is the special privilege's access method to Realtime Database which not affected by security rule (no matter what your database rules were configured all are by-pass with database secret).

Then you can full control the database with legacy Database secret without restriction by database rules.

There are several ways to access database i.e. through legacy and insecure **database secret** (used in this library and Google's AppScript), **custom access token** (for app or device that able to create, encrypt and sign the custom access token) and **OAuth2.0** authorization (for web or app that has can show user's consent screen).

poli44, 2019-09-17:
Thank you for the quick reply.
Is the recall date known legacy Database secret?
What is easier to implement custom access token or OAuth2.0 authorization?

mobizt, 2019-09-17:
Database secret still in use as long as it the only way for Google's AppScript to access Realtime Database. Then I think it set as obsolete for secure custom token and OAuth2.0 promotion but still support for some none UI platforms including embedded device.

For custom token, it's possible for ESP32 only (due to memory restriction) because of library to encrypt and sign token.

closed 2019-09-18 by mobizt