mobizt/Firebase-ESP8266 #54: BUG - Cannot read two streans simultaneously

JojoSalustiano, 2019-09-19:
**Describe the bug**
If I use two `readStream `in different `FirebaseData`, the `.streamAvailable()` stop working

**To Reproduce**
**1.** Compile e update this code
``` C++
// based in https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Stream/Stream.ino

//FirebaseESP8266.h must be included before ESP8266WiFi.h
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com"
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"
#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"

//Define FirebaseESP8266 data object
FirebaseData firebaseData0;
FirebaseData firebaseData1;

void printJsonObjectContent(FirebaseData &data);

unsigned long sendDataPrevMillis = 0;

String path0 = "/Stream_Test/0";
String path1 = "/Stream_Test/1";

void setup()
{

  Serial.begin(9600);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData0, path0))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData0.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (!Firebase.beginStream(firebaseData1, path1))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData1.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
  // data0
  if (!Firebase.readStream(firebaseData0)) // if comment this line, that firebaseData1 work
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData0.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData0.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData0.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData0.streamPath());
    Serial.println("EVENT PATH: " + firebaseData0.dataPath());
    Serial.println("DATA TYPE: " + firebaseData0.dataType());
    Serial.println("EVENT TYPE: " + firebaseData0.eventType());
    Serial.print("VALUE: ");
    if (firebaseData0.dataType() == "int")
      Serial.println(firebaseData0.intData());
    else if (firebaseData0.dataType() == "float")
      Serial.println(firebaseData0.floatData());
    else if (firebaseData0.dataType() == "boolean")
      Serial.println(firebaseData0.boolData() == 1 ? "true" : "false");
    else if (firebaseData0.dataType() == "string")
      Serial.println(firebaseData0.stringData());
    else if (firebaseData0.dataType() == "json")
      printJsonObjectContent(firebaseData0);
    Serial.println("------------------------------------");
    Serial.println();
  }

  // data1
  if (!Firebase.readStream(firebaseData1)) // if comment this line, that firebaseData0 work
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData1.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData1.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData1.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData1.streamPath());
    Serial.println("EVENT PATH: " + firebaseData1.dataPath());
    Serial.println("DATA TYPE: " + firebaseData1.dataType());
    Serial.println("EVENT TYPE: " + firebaseData1.eventType());
    Serial.print("VALUE: ");
    if (firebaseData1.dataType() == "int")
      Serial.println(firebaseData1.intData());
    else if (firebaseData1.dataType() == "float")
      Serial.println(firebaseData1.floatData());
    else if (firebaseData1.dataType() == "boolean")
      Serial.println(firebaseData1.boolData() == 1 ? "true" : "false");
    else if (firebaseData1.dataType() == "string")
      Serial.println(firebaseData1.stringData());
    else if (firebaseData1.dataType() == "json")
      printJsonObjectContent(firebaseData1);
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void printJsonObjectContent(FirebaseData &data){
  size_t tokenCount = data.jsonObject().parse(false).getJsonObjectIteratorCount();
  String key;
  String value;
  FirebaseJsonObject jsonParseResult;
  Serial.println();
  for (size_t i = 0; i < tokenCount; i++)
  {
    data.jsonObject().jsonObjectiterator(i,key,value);
    jsonParseResult = data.jsonObject().parseResult();
    Serial.print("KEY: ");
    Serial.print(key);
    Serial.print(", ");
    Serial.print("VALUE: ");
    Serial.print(value); 
    Serial.print(", ");
    Serial.print("TYPE: ");
    Serial.println(jsonParseResult.type);        

  }
}
```
**2.** Insert data in firebase
[image]

**Expected behavior**
That each stream work independently

**Screenshots**
With two `readStream()` don't work:
[image]

With only one `readStream()` work fine:
[image]
[image]

**IDE and its version:**
- IDE: PlatformIO [1.8.3] in VSCode [1.38]
- board: nodeMCU - ESP12E
- platform = espressif8266 [2.2.3]
- framework: arduino
 - Firebase ESP8266 Client [2.4.2]

mobizt, 2019-09-19:
Thanks for address the issue.

Due to the next update is nearly to finish and publish.

At this time, please change the code in library at **FirebaseESP8266.cpp**, at line no. [3056](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.cpp#L3056) from 
```C++
if (ml - dataObj._streamResetMillis > 50)
//https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.cpp#L3056
```
to

```C++
if (ml - dataObj._streamResetMillis > 100)
```
This value is the interval to routinely pause the stream reading and reset the FirebaseData to allow others mcu tasks for every 50 ms. If more FirebaseData streams are in sequential loop, this interval is timed up easily and quit the stream reading function immediately then it need more interval e.g. 100 for two FirebaseData streams or 150 for three FirebaseData streams.

The next update will add this fix and other bugs fix.

JojoSalustiano, 2019-09-19:
It's work perfect
Thank to the answer, support and awesome library

closed 2019-09-19 by JojoSalustiano