mobizt/Firebase-ESP8266 #55: How to parse many different values from a single JSON response

arlucio, 2019-09-25:
I have a stream response like this: 

```
STREAM PATH: 
EVENT PATH: /progA
DATA TYPE: json
EVENT TYPE: put
VALUE: {"endA":1569290460,"friA":false,"monA":false,"satA":false,"startA":1569243600,"sunA":true,"tempA":48,"thuA":false,"tueA":true,"wedA":true}
```

And I'm trying to read the values like this:

```
        json.clear().setJsonData(firebaseData.jsonData());

        startA = json.parse().get("progA").get("startA").parseResult().intValue;
        endA = json.parse().get("progA").get("endA").parseResult().intValue;
        sunA = json.parse().get("progA").get("sunA").parseResult().boolValue;
        monA = json.parse().get("progA").get("monA").parseResult().boolValue;
        tueA = json.parse().get("progA").get("tueA").parseResult().boolValue;
        wedA = json.parse().get("progA").get("wedA").parseResult().boolValue;
        thuA = json.parse().get("progA").get("thuA").parseResult().boolValue;
        friA = json.parse().get("progA").get("friA").parseResult().boolValue;
        satA = json.parse().get("progA").get("satA").parseResult().boolValue;
        tempA = json.parse().get("progA").get("tempA").parseResult().intValue;
```

I get to read startA, endA and tempA right, all are int, I'm not been able to read all the bool values. Am I doing something wrong? What would be the best way to do this with this new library version?

In the old version I was doing like this: 

```
  json = firebaseData.jsonData();
  deserializeJson(doc, json);
  tempA = doc["tempA"];
  startA = doc["startA"];
  endA = doc["endA"];
  sunA = doc["sunA"];
  monA = doc["monA"];
  tueA = doc["tueA"];
  wedA = doc["wedA"];
  thuA = doc["thuA"];
  friA = doc["friA"];
  satA = doc["satA"];
```
I've looked in the examples, couldn't find anything like this

Gizmodo, 2019-09-25:
FirebaseJson.cpp contains error.
Check line 496.

case JSMN_PRIMITIVE:
        strcpy_P(tmp, FirebaseJson_STR_7);
        if (strcmp(_jsonObj.stringValue.c_str(), tmp) == 0)
        {
            typeSet = true;
            **_jsonObj.boolValue = true;**
            strcpy_P(buf, FirebaseJson_STR_15);
        }
        else
        {
            memset(tmp, 0, 20);
            strcpy_P(tmp, FirebaseJson_STR_6);
            if (strcmp(_jsonObj.stringValue.c_str(), tmp) == 0)
            {
                typeSet = true;
                **_jsonObj.boolValue = false;**
                strcpy_P(buf, FirebaseJson_STR_15);
            }
        }

mobizt, 2019-09-28:
@Gizmodo, thanks for your contributions and pull request.
As I said that the library is nearly finish the update but from stress test, the library can occasionally cause the stack overflow on ESP8266 but works well on ESP32. Here is the [FirebaseJson](https://github.com/mobizt/FirebaseJson) library that is the part of this Firebase library. 

The next update was delay. You can pull the request again to fix the current issue.

Gizmodo, 2019-09-28:
@mobizt PR is ready.

mobizt, 2019-09-28:
Thanks, I merged it.

closed 2019-09-29 by mobizt

mobizt, 2019-10-28:
The new version of library is ready for update.

With the new version, your code can be
```C++
  FirebaseJson &json = firebaseData.jsonObject();
  FirebaseJsonData data;

  json.get(data, "startA");
  if (data.success && data.type == "int")
    startA = data.intValue;

  json.get(data, "endA");
  if (data.success && data.type == "int")
    endA = data.intValue;

  json.get(data, "sunA");
  if (data.success && data.type == "int")
    sunA = data.intValue;

  json.get(data, "monA");
  if (data.success && data.type == "int")
    monA = data.intValue;

  json.get(data, "tueA");
  if (data.success && data.type == "int")
    tueA = data.intValue;

  json.get(data, "wedA");
  if (data.success && data.type == "int")
    wedA = data.intValue;

  json.get(data, "thuA");
  if (data.success && data.type == "int")
    thuA = data.intValue;

  json.get(data, "friA");
  if (data.success && data.type == "int")
    friA = data.intValue;

  json.get(data, "satA");
  if (data.success && data.type == "int")
    satA = data.intValue;

  json.get(data, "tempA");
  if (data.success && data.type == "int")
    tempA = data.intValue;

```