mobizt/Firebase-ESP8266 #56: Crash when calling updateNodeSilent

kbssa, 2019-09-26:
I trying to use the update function updateNodeSilent to save some network data, but every time I call the function bellow the nodemcu crashes. 

It is working good if use the updateNode.

Is there any mistake on my code that I am not seeing ?

```
void updateFirebase(String& _nodeName, uint8_t _channel, uint8_t _value) {

	yield();

	String _path = FPSTR(path);
	_path += _nodeName;

	String key = "C";
	key += _channel;

	FirebaseJson json;
	json.addString(key, String(_value));
	Firebase.updateNodeSilent(updateFirebaseData, _path, json);

}
```

mobizt, 2019-09-28:
Please make sure path in **FPSTR(path)** is flash string (PROGMEM).

I was successfully tested it without any error.

closed 2019-09-28 by mobizt

kbssa, 2019-09-30:
I dont't know if it is bug in the library or in my code, but If I call the updateNodeSilent from inside any other library function the crash occurs. I have tested calling it from within of the ESPAsyncWebServer and painlessMesh, and both has crashed,  If I call it from the loop everything works good.

The updateNode works from anywhere I call it.

mobizt, 2019-09-30:
https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html

mobizt, 2019-10-01:
Use ESP32 instead because ESP8266 has limited stack and heap.