mobizt/Firebase-ESP8266 #59: I want to add Wifiwebserver in stream example

Sk787, 2019-10-03:
Hello, thanks a lot for the library. I was having a hard time to find a working library for firebase stream. This library works fine for me. I wanted to try adding wifiwebserver in my code. My webserver was working fine firebase stream was not working. It didn't respond to any change of firebase data base. 
Here is my code:

```
//FirebaseESP8266.h` must be included before ESP8266WiFi.h
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "HOST"
#define FIREBASE_AUTH "Secret"
#define WIFI_SSID "Wifi_name"
#define WIFI_PASSWORD "Pass"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;

void printJsonObjectContent(FirebaseData &data);

String path = "My-path";

uint16_t count = 0;

WiFiServer server(80);

void setup()
{
  Serial.begin(9600);
  pinMode(2, OUTPUT);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  server.begin();
}

void loop()
{
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: ");
    if (firebaseData.dataType() == "int")
      Serial.println(firebaseData.intData());
    else if (firebaseData.dataType() == "float")
      Serial.println(firebaseData.floatData());
    else if (firebaseData.dataType() == "boolean")
      Serial.println(firebaseData.boolData() == 1 ? "true" : "false");
    else if (firebaseData.dataType() == "string")
      Serial.println(firebaseData.stringData());
    else if (firebaseData.dataType() == "json")
      printJsonObjectContent(firebaseData);
    Serial.println("------------------------------------");
    Serial.println();
  }

      if (firebaseData.pauseFirebase(true))
    {
      WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  
  // Match the request
  int val;
  if (req.indexOf("/gpio/1") != -1)
    val = 0;
  else if (req.indexOf("/gpio/0") != -1)
    val = 1;
  else {
    Serial.println("invalid request");
    client.stop();
    return;
  }

  // Set GPIO2 according to the request
  digitalWrite(2, val);
  
  client.flush();
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\nGPIO is now ";
  s += (val)?"0":"1";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");  
  }
 
    else
    {
      Serial.println("------------------------------------");
      Serial.println("Can't pause the WiFi client...");
      Serial.println("------------------------------------");
      Serial.println();
    }
    //Unpause WiFi client from Firebase task
    firebaseData.pauseFirebase(false);
}

void printJsonObjectContent(FirebaseData &data){
  size_t tokenCount = data.jsonObject().parse(false).getJsonObjectIteratorCount();
  String key;
  String value;
  FirebaseJsonObject jsonParseResult;
  Serial.println();
  for (size_t i = 0; i < tokenCount; i++)
  {
    data.jsonObject().jsonObjectiterator(i,key,value);
    jsonParseResult = data.jsonObject().parseResult();
    Serial.print("KEY: ");
    Serial.print(key);
    Serial.print(", ");
    Serial.print("VALUE: ");
    Serial.print(value); 
    Serial.print(", ");
    Serial.print("TYPE: ");
    Serial.println(jsonParseResult.type);        

  }
}
```

mobizt, 2019-10-03:
Please remove pause/unpause firebase code and run your webserver directly in loop.

Sk787, 2019-10-04:
> Please remove pause/unpause firebase code and run your webserver directly in loop.

Thanks. It worked. But if i add delay in the loop, stream stops recieving updates.

mobizt, 2019-10-04:
Why delay? It delay everything.

Sk787, 2019-10-04:
> Why delay? It delays everything.

I need to run the device 24/7. Without the delay wont the CPU become hot?

mobizt, 2019-10-04:
CPU hot! No!

Use millis for Ticker for your schedule tasks avoid blocking process like delay.

Sk787, 2019-10-04:
Okay. Thanks a lot for your help.

closed 2019-10-04 by mobizt