mobizt/Firebase-ESP8266 #61: JSON parse

Gizmodo, 2019-10-25:
JSON Object:
{
  "LastOnline" : "15:49:39 24.10.2019",
  "Light" : {
    "Center" : {
      "enabled" : true,
      "off" : "21:45",
      "on" : "21:12",
      "pin" : 13,
      "position" : "Center",
      "state" : false
    }
}

Try to parse:
`
if (Firebase.getJSON(firebaseData, pathLight + ledPath)) {
            if (firebaseData.dataType() == "json") {
                Serial.println();

                FirebaseJson& json = firebaseData.jsonObject();
                FirebaseJsonData& jsonData = firebaseData.jsonData();

                Serial.println("Iterate JSON data:");
                Serial.println();
                size_t len = json.iteratorBegin();
                String key, value = "";
                int type = 0;
                for (size_t i1 = 0; i1 < len; i1++) {
                    json.iteratorGet(i1, type, key, value);
                  //  json.get(jsonData, key);
                    if (jsonData.success) {
                        Serial.print("jsonData[ ");
                        Serial.print("Key = ");
                        Serial.print(key);
                        Serial.print(" Type = ");
                        Serial.print(jsonData.type);
                        Serial.print(" Value = ");
                        Serial.print(jsonData.stringValue);
                        Serial.println(" ]");
                    }
                    Serial.print(i1);
                    Serial.print(", ");
                    Serial.print("Type: ");
                    Serial.print(type == JSON_OBJECT ? "object" : "array");
                    if (type == JSON_OBJECT) {
                        Serial.print(", Key: ");
                        Serial.print(key);
                    }
                    Serial.print(", Value: ");
                    Serial.println(value);
                }
                json.iteratorEnd();
            }
        }
`
Output:
Iterate JSON data:

0, Type: object, Key: enabled, Value: true
1, Type: object, Key: off, Value: 21:45
2, Type: object, Key: on, Value: 21:12
3, Type: object, Key: pin, Value: 13
4, Type: object, Key: position, Value: Center
5, Type: object, Key: state, Value: false

If uncomment line 
 //json.get(jsonData, key);

then the output became this:
Iterate JSON data:

jsonData[ Key = enabled Type = bool Value = true ]
0, Type: object, Key: enabled, Value: true
jsonData[ Key = enabled Type = bool Value = true ]
1, Type: object, Key: enabled, Value: true
jsonData[ Key = enabled Type = bool Value = true ]
2, Type: object, Key: enabled, Value: true
jsonData[ Key = enabled Type = bool Value = true ]
3, Type: object, Key: enabled, Value: true
jsonData[ Key = enabled Type = bool Value = true ]
4, Type: object, Key: enabled, Value: true
jsonData[ Key = enabled Type = bool Value = true ]
5, Type: object, Key: enabled, Value: true

Key is always the same.
How does it happen?

mobizt, 2019-10-25:
Because jsonData used for storing the result from parse function json.get(jsonData, key);

 jsonData is intend to be structed or class that holds tempolary info from parsing.

Gizmodo, 2019-10-25:
Thank you.
With current lib version my code often crashes: Panic core_esp8266_main.cpp:185 loop_task
I will use 2.4.2 for a while.

mobizt, 2019-10-25:
I recommended to update ESP8266 Core SDK (boards) to latest version.

Can you use ESP Exception Decoder and share the result?

How about your code in loop()?

Gizmodo, 2019-10-25:
I will share all info tomorrow

Gizmodo, 2019-10-26:
```
void writeTemperatureFirebase() {
    if (WiFi.isConnected()) {
        FirebaseData firebaseData;
        FirebaseJson json;
        Serial.printf_P(PSTR("%s"), "Save temperature to Firebase: \n");
        Serial.println(ESP.getFreeHeap());
        json.add("temp1",temp1).add("temp2",temp2).add("DateTime", String(clockRTC.dateFormat("H:i:s d.m.Y", clockRTC.getDateTime())));
        
        if (Firebase.setJSON(firebaseData, pathTemperatureOnline, json)) {
            Serial.printf_P(PSTR("\nDone\n"));
        } else {
            Serial.printf_P(PSTR("\nError writeTemperatureFirebase: %s\n"), firebaseData.errorReason().c_str());
        }
        firebaseData.clear();
    }
}
```
And stack trace:
```

Decoding stack results
0x402056cb: WiFiClient::available() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 263
0x4021ebc2: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 510
0x40238340: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 958
0x402382e8: br_ssl_hs_client_init_main at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 67
0x40237df6: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1305
0x4021ee08: BearSSL::WiFiClientSecure::_wait_for_handshake() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 547
0x4021eff4: BearSSL::WiFiClientSecure::_connectSSL(char const*) at c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/shared_ptr.h line 291
0x40215238: esp_yield() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x402160a7: delay(unsigned long) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x402060a5: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 136
0x4021f111: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 231
0x4020cc61: FirebaseHTTPClient::connect() at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266HTTPClient.cpp line 173
0x4020cc9c: FirebaseHTTPClient::sendRequest(char const*, char const*) at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266HTTPClient.cpp line 153
0x40205404: ESP8266WiFiSTAClass::status() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x40229032: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x402083e4: FirebaseESP8266::sendFirebaseRequest(FirebaseData&, char const*, unsigned char, unsigned char, char const*, char const*, unsigned int) at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4078
0x4020cba9: FirebaseHTTPClient::setRootCA(char const*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h line 70
0x40209f19: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2050
0x40227785: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 655
0x40205404: ESP8266WiFiSTAClass::status() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020a608: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2416
0x40205404: ESP8266WiFiSTAClass::status() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x40229032: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4020bb74: FirebaseESP8266::sendRequest(FirebaseData&, unsigned char, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&, std::string const&) at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2512
0x40205404: ESP8266WiFiSTAClass::status() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020c6a5: FirebaseESP8266::buildRequest(FirebaseData&, unsigned char, unsigned char, std::string const&, char const*, bool, std::string const&, std::string const&) at C:\Users\user\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 240
[... cut ...]
0x40204bfc: setup() at C:\Users\user\Documents\Arduino\sketch_aug25a/sketch_aug25a.ino line 614
0x402152e0: loop_wrapper() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 122
0x401000e8: app_entry() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 248
0x40240005: f255_mulgen at src/ec/ec_c25519_m15.c line 1115
0x40240055: f255_mulgen at src/ec/ec_c25519_m15.c line 1131
0x40232200: ip4_input at core/ipv4/ip4.c line 690

```

mobizt, 2019-10-26:
Look like it related to WiFiClientSecure library but I'm not sure that is accurate stack decoding or no info related to the library because it stop at some point in WiFiClient secure.

Are you using Arduino IDE or other IDE?
If not, use Arduino IDE and use ESP Exception Decoder Add on from me-no-dev.

Check the free heap. Share all your code because I can't see the problem in only just this portion of code,  recursive function call may cause stack overflow.

Gizmodo, 2019-10-26:
I use both IDE: Arduino with ESP Exeption Decoder and VSCode (PlatformIO)
Source Code:
[sketch_aug25a.zip](https://github.com/mobizt/Firebase-ESP8266/files/3774718/sketch_aug25a.zip)

mobizt, 2019-10-26:
After investigate your code.

I've found that you use FirebaseData as local object.  Due to the instance of WiFiClientSecure used inside the object is created, its resource may not completely free, or free later which mannages by WiFiClientSecure itself.

I recommended to use FirebaseData object as a global object to shared internal WiFi client.

Gizmodo, 2019-10-26:
I'll try it. Thanks a lot.

closed 2019-10-26 by Gizmodo