mobizt/Firebase-ESP8266 #63: How to set time out ? Firebase.setwriteSizeLimit()

Nursbg, 2019-11-02:
when i apply this code  in void setup().
```
...
    Firebase.setMaxRetry(firebaseData, 3);
    Firebase.setwriteSizeLimit(firebaseData, "tiny" );
    if ( Firebase.setString(firebaseData, path, "test" ) {
      
    } else {
      
    }
   digitalWrite (ledPin, HIGH);
...

```
it is work, but when network connection is error,  it is stoped in Firebase.setString(firebaseData, path, "test" ). and can not execute digitalWrite (ledPin, HIGH);

when i replace "tiny" to "tiny (1s)" ,
```
...
    Firebase.setMaxRetry(firebaseData, 3);
    Firebase.setwriteSizeLimit(firebaseData, "tiny (1s)" );
    if ( Firebase.setString(firebaseData, path, "test" ) {
      
    } else {
      
    }
   digitalWrite (ledPin, HIGH);
 ...
```
it can not write "test" to firebase, but can execute digitalWrite (ledPin, HIGH);
my goal is : how to set the time out, when network problem is occured. and can execute to next task (digitalWrite (ledPin, HIGH);).
`Thanks.`

mobizt, 2019-11-02:
The **writeSizeLimit** and **timeout** options are for Firebase Server operation and not for library or client operation. 

You can read the doc [here](https://firebase.google.com/docs/reference/rest/database#section-param-writesizelimit).

The string parameter' s values supported by function **writeSizeLimit** are **tiny**, **small**, **medium** and **large**, all other values e.g. tiny (1s) are not correct.

All Firebase calls operation and connection issues were handle by library and it provides the error reason related to the causes, then setting timeout is not necessary.

The default timeout of library for waiting the server response is 5 seconds. The library will quit the Firebase calls immediately when WiFi connection was disconnected and waits for 5 seconds if WiFi was connected and server was not response within 5 seconds.

Nursbg, 2019-11-02:
Thank you @mobizt ,

and now,
```
...
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
    Firebase.reconnectWiFi(true);
    //Firebase.setMaxRetry(firebaseData, 3);
    //Firebase.setwriteSizeLimit(firebaseData, "tiny" );
    if( Firebase.setString(firebaseData, path, "test" ) {
      
    } else {
     //this place never excecuted after 5 second time out.
    }
   digitalWrite (ledPin, HIGH);
...
```
Why  "digitalWrite (ledPin, HIGH);" never executed after 5 second time out. during wifi disconnected ?
please help me, Thanks.

mobizt, 2019-11-02:
Where is these code? In set up it runs once and before this you need to connect to connect to WiFi and it must be successful connected.

Nursbg, 2019-11-02:
Here is my simple code : 
```
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server(80);

#define FIREBASE_HOST "xxxxx.firebaseio.com"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxx"
#define WIFI_SSID "MySSID"
#define WIFI_PASSWORD "xxxxxx"

FirebaseData firebaseData;

const int ledPin = 2;
String path = "/user/test";

void setup()
{
  pinMode(ledPin, OUTPUT);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED)
  {
    digitalWrite(ledPin, 0);
    delay(200);
    digitalWrite(ledPin, 1);
    delay(200);
  }
  digitalWrite(ledPin, 0); //led on

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  server.on("/test", test);
  server.begin();
}

void loop()
{
  server.handleClient();
}

void test() {
  server.send(200, "text/plain", "OK");
  if ( Firebase.setString(firebaseData, path, "test") ) {

  } else {

  }
 // blink led
  digitalWrite(ledPin, 1);
  delay(200);
  digitalWrite(ledPin, 0);
  delay(200);
}

```
After connecting to router was complete and connected to internet.
client send request ("/test") to esp , after a few second led is blinking.
and then i try to disconnect router WAN from internet : 
-  for first test was success, after 5 second time out, led blinking.
-  but after 2nd test esp was crash and no respone.

thanks

mobizt, 2019-11-02:
Nothing error or crash from my test.

closed 2019-11-02 by mobizt