mobizt/Firebase-ESP8266 #64: FirebaseESP8266.h takes high memory, I need lighter version

reza-hadipour, 2019-11-06:
Hi, I have used FirebaseESP8266.h in my ESP-12F and I figured out it has a high memory usage.
Would you mind can we have a lighter library?

mobizt, 2019-11-06:
I can't shrink the size of flash mem because we think of available free memory (ram) first then every constant string will store as flash string to save ram usage. 

I recommend to use others ESP8266 chips or modules which the overall size is also small e.g. Wemos D1 mini pro.

reza-hadipour, 2019-11-06:
Yes, its true that ESP-12F is suitable for using the Firebase, But you may need to use other libraries in your code that don't have enough memory left and your esp module is keep going to restart.

I have try my firmware with wemos D1 mini pro.

mobizt, 2019-11-06:
You should debug rather blame on memory. You may make memory leak with your code or use unhandle memory/variable and leads to error.

The watchdog reset may happen when you 're running the blocking code.

Your request is not quite clear what you're talking between flash or ram. If you're new to esp, you need to learn how to debug and decode the stack trace.

mobizt, 2019-11-06:
Ok I just known the potential cause of low memory issue.

The issue is from the reserved Bearssl rx buffer is too large (16k), now I fixed it already.

You can update the library to v2.6.6 from Git version.

closed 2019-11-07 by mobizt

reza-hadipour, 2019-11-07:
Thanks for your considering
well for clear the issue I take a some screenshot of Sketch uses & Global variables use after verifying my arduino code.

1. my Arduino code without including "FirebaseESP8266.h"
==>>
Sketch uses 403788 bytes (38%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33088 bytes (40%) of dynamic memory, leaving 48832 bytes for local variables. Maximum is 81920 bytes.

2. my Arduino code after just including "FirebaseESP8266.h"
==>>
Sketch uses 423244 bytes (40%) of program storage space. Maximum is 1044464 bytes.
Global variables use 58984 bytes (72%) of dynamic memory, leaving 22936 bytes for local variables. Maximum is 81920 bytes.

3. after updating FirebaseESP8266 to version 2.6.6
==>>
Sketch uses 423244 bytes (40%) of program storage space. Maximum is 1044464 bytes.
Global variables use 58984 bytes (72%) of dynamic memory, leaving 22936 bytes for local variables. Maximum is 81920 bytes.

mobizt, 2019-11-07:
That is estimation.
You should check your free heap when running.

mobizt, 2019-11-07:
As I see the latest library cannot help you to gain more free memory.

This because you're using the old esp8266 core sdk version 2.4.x that use old WiFi client (axTLS) instead of BearSSL.

Please update your board (esp8266 core sdk) to the latest 2.5.2 to gain more memory.