mobizt/Firebase-ESP8266 #66: Stream responds very slow if there hasn't been new data for a long time

majorlennox, 2019-11-10:
Hello,
I am currently using ESP8266 to switch a light whenever I update a value in my firebase realtime databse. For that, I use stream, so that the ESP8266 reacts only when there's new data available. This works almost instantly BUT when I haven't updated the data for a day or so, it take's about 30 seconds to react to data changes and then returns to normal behaviour. Any ideas why?
Regards, MajorLennox

mobizt, 2019-11-10:
As long as Firebase.readStream and Firebase.streamDataAvailable are stayed in loop without blocking code, your stream will be responsed immediately unless stream timeout was occurred or your internet connection status and traffic (even WiFi was connected).

For better performance, readStream should run repeatedly as fast as possible, place inside millis or condition statement is not recommend.

majorlennox, 2019-11-10:
When does stream timeout trigger and for how long will it time out? WiFi connection was established, and internet connection aswell (at least with my phone, which uses the same).

Should readStream also run as fast as possible when there's no WiFi Connection?

mobizt, 2019-11-11:
When the device connected to Firebase in stream mode, server and client are open the connection for each other and server will send keep-alive message to client periodically (sometimes within 30 seconds or more) which depends on server operation and server-clent connection status. The library use this message to determine the timeout status, if there are absent of this message for 30 seconds, set the timeout status and begin to re-connect the stream again. The time used for begin the new stream connection is the same as other Firebase calls.

Every Firebase calls and stream read, WiFi connection was checkef and try to re-connect if WiFi disconnected without blocking operation or without waiting or delay and set the status and exit from that function immediately.

WiFi radio interference from other device with the same channel, your router or hotspot hang or other factors can happen.

For your case, you may need to check your internet connection periodically to prove the internet availability with ping.

The ping function is available or use library.

The readStream function is for data polling. If you delay or periodically run it with millis or delay then you will not get the data in real time.

You device will not sleep without set it to sleep.

majorlennox, 2019-11-11:
I see, thanks for explaining this to me. But if it's a problem related to internet, then why does the issue only come up if I haven't updated any data for a long time?

mobizt, 2019-11-11:
I don't know all your situation but from the client side, its operation is straight forward.  You may need to debug the WiFi operation from IDE to see what happens.

WiFi may unreliable when the board voltage are not stable, low memory or some blocking operation.

majorlennox, 2019-11-11:
Hm. I can't really narrow it down, but it happens only when I wasn't home for a couple of hours. I come back home, change the Firebase Data via an Android App -> nothing happens til a  minute or so has passed, then the ESP8266 recieves all changes and executes all of them. Could it be that the WiFi connection gets set to sleep since there hasn't been any new data for a while and waking it up takes a bit of time? If it was a connection issue, this problem would also occur when I use the esp8266 regularly, however the issue only occures when there hasn't been any new data for a long time.

mobizt, 2019-11-11:
Let try to remove yield() from your code.

majorlennox, 2019-11-11:
I've changed the code as follows, but the issue is still present:
/*#### Changelog ####
  TO-DO:

  (- add status indicator LED in case)
  v1.4: ..................to be tested
  - removed wifi connection check as well as 500ms timer
    - add feedback to indicate success of operation
  v1.3: ..................working
  - back to esp8266
  v1.2:
  - switched from constantly asking for new values to EventListener
  v1.1:
  - added 500ms delay for requests (lWaitMillis)
  v1.0:
  - switched to FirebaseArduino https://github.com/FirebaseExtended/firebase-arduino, hoping to prevent random restarts from happening
  v0.1:
  - added yield() after read to prevent watchdog from triggering due to delays in reading
*/
```

void setup() {

  WiFi.persistent(false);
  Serial.begin(115200);

  Serial.println(WiFi.macAddress());
  pinMode(WIFI_LED, OUTPUT);
  IPAddress ip(192, 168, 0, 250);
  IPAddress gateway(192, 168, 0, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress dns(8, 8, 4, 4);
  // WiFi.config(ip, gateway, subnet);
  wifi_set_macaddr(0, const_cast<uint8*>(mac));
  Serial.println(WiFi.macAddress());

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    digitalWrite(WIFI_LED, LOW);
    delay(100);
    digitalWrite(WIFI_LED, HIGH);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  digitalWrite(WIFI_LED, HIGH);
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Gateway address: ");
  Serial.println(WiFi.gatewayIP().toString());
  Serial.print("AP subnet mask: ");
  Serial.println(WiFi.subnetMask());

  delay(1000);
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Serial.println("Debug");
  delay(1000);
  if (Firebase.setString(firebaseData, PATH_LAMP, "1"))
  {
    Serial.println("PASSED");
  }
  else {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  stateint = state.toInt();  // initial setup
  while (!Firebase.beginStream(firebaseData, PATH_LAMP))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  Firebase.setReadTimeout(firebaseData, 300);

}

void loop() {
  int firebaseResult = firebaseGet(PATH_LAMP).toInt();
  if (firebaseResult != stateint) {
    if (firebaseResult == 1) {
      turnOn();
    }
    else if (firebaseResult == 0) {
      turnOff();
    }
  }

}
void turnOn() {
  digitalWrite(LAMPPIN, HIGH);
  Serial.println("ON");
  stateint = 1;
}
void turnOff() {
  digitalWrite(LAMPPIN, LOW);
  Serial.println("OFF");
  stateint = 0;
}
String firebaseGet(String s) {
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  if (firebaseData.streamAvailable())
  {
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    if (firebaseData.dataType() == "string") {
      returnDataTemp = firebaseData.stringData();
      if (returnDataTemp == "1" || returnDataTemp == "0") {
        returnData = returnDataTemp;
        Firebase.setString(firebaseData2, PATH_CHECK, "check");
      }
    }
    Serial.println("------------------------------------");
    Serial.println();
  }
  return (returnData);
}
```

mobizt, 2019-11-11:
You show me some of incompleted code.

You should debug the WiFi and check what's going on with your WiFi.
https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html

Your issue is not related to the library but your network then I will close this issue.

closed 2019-11-12 by mobizt

prakash50160, 2020-12-07:
Hi...folks... I had also same problem that sometimes the esp8266 wont trigger when firebase updated... So i tired of many things like removing extra blocks which not required making it so faster iteration. but nothing works. Finally i got with these just by adding a delay of 10 milliseconds at bottom of the code... 

Now it's totally reliable .. i had given 100 triggers and kept at closed small room where network is very too low.. and perfect i got all triggers. Thanks for the developer.. Here is my code below.

void setup()
{

  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN,HIGH);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop(){
  if (!Firebase.readStream(firebaseData)){
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData.streamTimeout()){
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  
  if (firebaseData.streamAvailable()){
    if((firebaseData.dataType()).equals("boolean")){
      if(firebaseData.boolData()){
        Serial.println("yes");
        digitalWrite(LED_BUILTIN,LOW);
      }else{
        Serial.println("no");
        digitalWrite(LED_BUILTIN,HIGH);
      }}}
  delay(10);  
}

mobizt, 2020-12-07:
@prakash50160 It's not because of just using the small delay you use. 

You should update the library and esp8266 core sdk to the latest.

You should check the internet connectivity from ISP to your device,  radio interference, stability of your power supply.

Hubert5111, 2023-01-07:
Dear friends, I have also a speed problem when reading _int_ values from Firebase.

This is what I want to do:
I have created an app with MIT AppInventor in order to control 2 electric doors via my smart phone. The app has 11 buttons (including 4 buttons for test purposes (see below _***** debug mode *****_)):

[image]

There is (also for test purposes) a counter which runs at the ESP8266. It is increased every progam cycle and is displayed in the app ("_from ESP8266: 56_") to check the data transfer from the ESP8266 to the app (it works, but also very slow).

 To get the state of the buttons from the app to the ESP8266, I first store them in Firebase and then read them from the ESP8266. When a button is pressed, an _int_ value is set to 1 until the button is released.

My problem is that it takes several seconds until the data are available at the ESP8266. Each time a button is pessesd in the app, the int LED at the ESP8266 is activated. The connection from the app to Firebase is not the issue, the values change without any recognizable delay when I press any button in the app.
One additional information: at the beginning I used only 2 buttons (and the counter) and it reacted much faster, but also with a recognizable delay (approx. 500ms).

Here is the code on the ESP8266:

<pre>

 <font color="#95a5a6">&#47;**********************************************************</font>
<font color="#95a5a6"> &nbsp;* author: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Hubert Unterberger &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*</font>
<font color="#95a5a6"> &nbsp;* created: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;05.01.2023 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*</font>
<font color="#95a5a6"> &nbsp;* last update: &nbsp;07.01.2023 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*</font>
<font color="#95a5a6"> &nbsp;* board: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Lolin(Wemos) D1 R2 &amp; mini &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*</font>
<font color="#95a5a6"> &nbsp;* ------------------------------------------------------ *</font>
<font color="#95a5a6"> &nbsp;* ESP8266 board 2.7.x, --&gt; NOT newer &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*</font>
<font color="#95a5a6"> &nbsp;* library: ArduinoJson version 5.13.5, --&gt; NOT newer &nbsp;&nbsp;&nbsp;&nbsp;*</font>
<font color="#95a5a6"> &nbsp;**********************************************************&#47;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">FirebaseESP8266</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> &nbsp;<font color="#434f54">&lt;</font><font color="#d35400">ESP8266WiFi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;credentials.h&#34;</font>

<font color="#5e6d03">#define</font> <font color="#000000">WIFI_SSID</font> <font color="#000000">SECRET_SSID</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; WIFI SSID</font>
<font color="#5e6d03">#define</font> <font color="#000000">WIFI_PASS</font> <font color="#000000">SECRET_PASS</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; WIFI password</font>

<font color="#5e6d03">#define</font> <font color="#000000">FIREBASE_HOST</font> <font color="#000000">SECRET_FIREBASE_HOST</font> &nbsp;<font color="#434f54">&#47;&#47; Firebase Host adress</font>
<font color="#5e6d03">#define</font> <font color="#000000">FIREBASE_AUTH</font> <font color="#000000">SECRET_FIREBASE_AUTH</font> &nbsp;<font color="#434f54">&#47;&#47; Firebase Authorization key</font>

<font color="#5e6d03">#define</font> <font color="#000000">intLED</font> <font color="#000000">2</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Pin of internal LED</font>

<font color="#000000">FirebaseData</font> <font color="#000000">firebaseData</font><font color="#000000">;</font>
<font color="#000000">FirebaseJson</font> <font color="#000000">json</font><font color="#000000">;</font>

<font color="#00979c">unsigned</font> <font color="#00979c">long</font> <font color="#000000">cycleTimer</font><font color="#000000">;</font>
<font color="#00979c">unsigned</font> <font color="#00979c">long</font> <font color="#000000">cycleTime</font> <font color="#434f54">=</font> <font color="#000000">5</font><font color="#000000">;</font>
<font color="#00979c">unsigned</font> <font color="#00979c">int</font> <font color="#000000">counter</font> <font color="#434f54">=</font> <font color="#000000">0</font><font color="#000000">;</font>

[... cut ...]
 &nbsp;<font color="#5e6d03">while</font> <font color="#000000">(</font><font color="#d35400">millis</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#434f54">&lt;</font> <font color="#000000">cycleTimer</font> <font color="#434f54">+</font> <font color="#000000">cycleTime</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; idle (for 5ms)</font>
 &nbsp;<font color="#000000">}</font>

<font color="#000000">}</font>

</pre>

What am I doing wrong. 
Please help me.

mobizt, 2023-01-08:
@Hubert5111 

Reading many database data in loop like that is inefficient due to multiple requests required the time for process the http request/response.

Firebase provides Realtime database which we can observe its changes in real-time using stream (SSE HTTP mode) feature.
Please see these [examples](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/DataChangesListener) for how to use stream.

Hubert5111, 2023-01-10:
@mobizt 
Thank you for your support und your suggestion to use stream.
For now I solved the problem by combining the status of all 11 buttons in the app to one value, and now I send just this single value instead of 11. 
It‘s not super fast, approx. 400ms until I receive the value, but it‘s ok for my purposes.
I‘m still wondering, if there is a super fast possibility ( e.g. 10ms) to get the info from the app in case a button is pressed.

Thanks again Mobizt,
Hubert

mobizt, 2023-01-11:
@Hubert5111 

It depends on how your codes block the stream.

In ESP8266, for Stream callback, the `readStream` function will be executed repeatedly in task scheduler.

Without stream callback, you need to place `readStream()` in main `loop()` to run repeatedly.

If some codes or tasks that use delay or something that blocking, run in the main `loop()`, the data received from stream will be delayed.

Normally, stream should be received in few milliseconds to tenths by its HTTP SSE technique.

Due to ESP8266 is single core MCU that all codes run serially, the stream may not fast as it should but it's not over 100 ms to received data. 

This example shows your code performance for stream. The database will be updated repeatedly 10000 times and your device should get all 10000 data changes events.

If you get some data lost from running test, your internet connection drop is a problem.
If you get all 10000 data changes but lagging, your device has some codes or tasks that block the stream.

You should use ESP32 in case you want Ideal Realtime response which the Stream (with callback) will run on separate FreeRTOS task which will no user codes can block it from running unless system tasks.

Even Raspberry Pi Pico that the Stream process runs on a single CPU core, the stream event will be received in Realtime as ESP32. 

If you update the library, this library is now supported Raspbery Pi Pico which Pico W device is ESP8266 replacement with the very cheap price.

Hubert5111, 2023-01-15:
@mobizt 

thank you again for your support.
So _stream_ is the solution to improve the speed of my data communication. 100ms max. response time is just perfect, I will try that!

Wish you a nice weekend and many thanks again,
Hubert