mobizt/Firebase-ESP8266 #68: HELP json.set() adds a "null" on which the code doesn't designate

sarkrui, 2019-11-11:
Hi @mobizt ,

This is might not an issue of the library, but it might be something wrong with my usage of using it. I cannot interpret where went wrong, so that's why I am here.

_The setup is, there're two devices (one named **Philips,** the other named **TUe.**)_ 
They both only **append** a byte number to each other's `colorQuue`(i.g. Philips device --> TUe device) when `tapped` is set to `false`; 
When `tapped` is `true`, then empties its own colorQueue.
```
{
    "philips": {
        "colorQueue": [
            2,
            3,
            4
        ],
        "tapped": false
    },
    "tue": {
        "colorQueue": [
            2,
            3,
            4
        ],
        "tapped": false
    }
}
```
The attached code runs as expected when it's set to `device_id :1`, however, when it's set to `device_id:2`, nothing else changed. The local device autonomously appends a `null` value to its own `colorQueue`, which is not coded at all. 
```
{
    "philips": {
        "colorQueue": [
            2,
            3,
            4,
            100,
            100
        ],
        "tapped": false
    },
    "tue": {
        "colorQueue": [
            2,
            3,
            4,
            null // the code should only append 100 to philips/colorQueue, and does nothing with the TUe one
        ],
        "tapped": true
    }
}
```
I can't understand why this happened, so please do me a favor!
Thanks! : D

[jsonSet_NULL.zip](https://github.com/mobizt/Firebase-ESP8266/files/3832316/jsonSet_NULL.zip)

mobizt, 2019-11-11:
Thanks for address this issue.

The strange result caused by error in **FirebaseJson** array parsing.  Now I fix it already.

Please update the library to resolve.

mobizt, 2019-11-12:
To prevent two or more clients write the data to the same path at the same time, I recommended to use Etag for each set or update.

closed 2019-11-12 by mobizt

sarkrui, 2019-11-13:
> The strange result caused by error in FirebaseJson array parsing. Now I fix it already.
Hi @mobizt , thanks for the prompt fix! The unwanted `NULL` issue is now gone.

However, another issue seems to emerge as I tried with `json.remove()`, it also empties the tree that I did not designate to. 

For example, hereby I only empty the `colorQueue` that it's under _/remote_device_ id/colorQueue_ when emoteTapped equals `true`. But in fact, it also empties the another colorQueue under `/device_id/`, which it's quite strange. I also tried with removing the branch either by `one-by-one removing` or `the complete branch`, unfortunately no luck. I hope you can take a look of my issue!
```
  if (remoteTapped == true) {
    Serial.println("Tapped");
    //empty queue and set from index:0
    
    //Method1: empty one by one
    //    for (size_t i = 0; i < queueArr.size(); i++) {
    //      remoteJson.remove(remote_device_id + "/colorQueue/[" + String(i) + "]");
    //    }
    
    //Method2: empty the whole branch
    //remoteJson.remove(remote_device_id + "/colorQueue");
    remoteJson.set(remote_device_id + "/colorQueue/[0]", 123);
  }
```

> To prevent two or more clients write the data to the same path at the same time, I recommended to use 
> Etag for each set or update.

Thanks for the hint! I'll check it later as I , at this moment, do not necessarily need to constantly write and read the data from the same path at same time. 

Thanks in adavance.

Yours
Sark

mobizt, 2019-11-14:
Thanks again, now I fixed it already.

Please update the library.

sarkrui, 2019-11-14:
Hi @mobizt 
Thanks for the quick fix! It worked. 

However, here's maybe another related issue.
```
//#define DEVICE_ID 1                      //id: 1 for Philips
#define DEVICE_ID 2                        //id: 2 for TU/e

#if (DEVICE_ID == 1)
const String device_id = "/philips";
const String remote_device_id = "/tue";
#endif

#if (DEVICE_ID == 2)
const String device_id = "/tue";
const String remote_device_id = "/philips";
#endif
```
Everything works as expected when `Device_ID` equals 1; However, when `Device_ID` equals 2,
it seems I have to include another line of `json.set("<unnecessary modifications of my data>");`
```
   remoteJson.set(remote_device_id + "/tapped", false);
```
after the 
```
  else if (remoteTapped == false) {
    Serial.println("Not Tapped");
    //append colorIndex to colorQueue
    Serial.println(remote_device_id);
    //Declare a firebasejsonArray for the remote device
    FirebaseJsonArray queueArr;
    remoteJson.get(remoteJsonData, remote_device_id + "/colorQueue");
    remoteJsonData.getArray(queueArr);
    Serial.println(queueArr.size());
    remoteJson.set(remote_device_id + "/colorQueue/[" + String(queueArr.size()) + "]", ColorIndex);
  }
```
Otherwise, it won't push `remoteJson` to Firebase. The jsonObject is properly serialized though. Do you have any idea why it went wrong?

Yours
Sark
[error_not_posting.ino.zip](https://github.com/mobizt/Firebase-ESP8266/files/3847327/error_not_posting.ino.zip)

mobizt, 2019-11-14:
I don't get your problem.

For Device_ID equals 2, which line of code does not work refer to your attach file?

sarkrui, 2019-11-14:
Sorry about the confusion. So, when `Device_ID` equals 2, it should append `ColorIndex` to `/remote_device_id/colorQueue` as scripted in 
> remoteJson.set(remote_device_id + "/colorQueue/[" + String(queueArr.size()) + "]", ColorIndex);

However, it did not. So I looked back to [Firebase_empting_.tree.zip](https://github.com/mobizt/Firebase-ESP8266/files/3843696/Firebase_empting_.tree.zip), I had an extra line of code

>localJson.set(device_id + "/tapped", true);

I added it to my [error_not_posting.zip](https://github.com/mobizt/Firebase-ESP8266/files/3847327/error_not_posting.ino.zip). (I meant a line of `FirebaseJson.set();` ), then it's working again. Without `FirebaseJson.set();` it doesn't; For `Device_ID`, then it's not necessary to include it.

Yours
Sark

mobizt, 2019-11-14:
I test Firebase.updateNode without  remoteJson.set and see some internal json serialize bugs and Firebase server reject it because of bad json or corrupt json data.

sarkrui, 2019-11-14:
That sounds strange to me. But for now, I just include a piece of non-meaning `FirebaseJson.set()` before `Firebase.updataNode` until you find a way to fix it! 
Thanks in advance!

mobizt, 2019-11-14:
Ok the issue was fixed, please update the library.

sarkrui, 2019-11-14:
You're awesome! Now it works like a charm!