mobizt/Firebase-ESP8266 #69: Bug When Using ESP8266 Time Alarm Library

komputerboy, 2019-11-13:
**Describe the bug**
I want to make ESP8266 has time alarm feature.  So for example if this time is at 17.00 i want to write data to firebase. The i found a library which is ESP8266TimeAlarm [here]. (https://github.com/jhagberg/ESP8266TimeAlarms). The library is working perfectly, it sync the current time with NTP. But when i adding this Firebase library program, the time not correct. So could you please resolve it? Thank You

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - Arduino
 - Version 1.8.9

**ESP8266 Arduino Core SDK version**
 - Version latest

**My Code**
```
#include "FirebaseESP8266.h"
#include <ESP8266TimeAlarms.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "testiotapp.firebaseio.com"
#define FIREBASE_AUTH "if8TsrodI1AJalJEx01yDJvsgqLnRqADIvI9Plyd"

#ifndef WIFI_CONFIG_H
#define YOUR_WIFI_SSID "mrm"
#define YOUR_WIFI_PASSWD "komputerboy"
#endif // !WIFI_CONFIG_H

FirebaseData firebaseData;
FirebaseJson json;
AlarmId id;

void configTimeZone() {
  configTime(0, 0, "0.se.pool.ntp.org");
  setenv("TZ", "WIB-7", 7);
  tzset();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  WiFi.mode(WIFI_STA);
  WiFi.begin(YOUR_WIFI_SSID, YOUR_WIFI_PASSWD);

  configTimeZone();

  Serial.print("Clock before sync: ");
  digitalClockDisplay();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.print("Clock after Wifi: ");
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  // create the alarms, to trigger at specific times
  Alarm.alarmRepeat(17, 4, 55, MorningAlarm); // 8:30am every day
  Alarm.alarmRepeat(23, 03, 0, EveningAlarm); // 5:45pm every day
}

void loop() {
  digitalClockDisplay();
  Alarm.delay(1000); // wait one second between clock display
}

// functions to be called when an alarm triggers:
void MorningAlarm() {
  Serial.println("Alarm: - turn lights off");
}

void EveningAlarm() {
  Serial.println("Alarm: - turn lights on");
}

void digitalClockDisplay() {
  time_t tnoww = time(nullptr);
  Serial.println(ctime(&tnoww));
}

void getData() {
  Firebase.getInt(firebaseData, "7sQIV2B3jvcqpiTuYdmPqbFnlbl1/lampu/30aea4044b21/status");
}
```

mobizt, 2019-11-13:
This is because the library also always set the system time (GMT+3 from NTP server) which required by  BearSSL client when root CA provided.
I will fix this conflicts and inform you soon.

mobizt, 2019-11-13:
Now the time bug was fixed.

Please update the library to latest version (v 2.6.8) from Github.

closed 2019-11-13 by mobizt