mobizt/Firebase-ESP8266 #70: Not work with static IP

Nursbg, 2019-11-19:
When I configure my ESP12F with static IP, the firebase didn't not work.
here is my simple code : 

```
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "xxxxxx.firebaseio.com"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
#define WIFI_SSID "xxxxxx"
#define WIFI_PASSWORD "xxxxxx"

IPAddress staticIP(192, 168, 2, 152);
IPAddress gateway(192, 168, 2, 1);
IPAddress subnet(255, 255, 255, 0);

FirebaseData firebaseData;
String path = "/test";

const int ledPin = 2;

void setup()
{
  pinMode(ledPin, OUTPUT);

  WiFi.config(staticIP, gateway, subnet); // static IP configuration.
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  
  while (WiFi.status() != WL_CONNECTED)
  {
    digitalWrite(ledPin, 0);
    delay(200);
    digitalWrite(ledPin, 1);
    delay(200);
  }
  
  digitalWrite(ledPin, 1); //led off

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Firebase.setString(firebaseData, path, "tost");
  
}

void loop()
{
    digitalWrite(ledPin, 0); 
}
```

**Expected behavior**
after connection to router was completed, esp able to send data to firebaase.
Firebase.setString(firebaseData, path, "tost"); 

**IDE and its version:**
 - Arduino
 - Version 1.8.10

**ESP8266 Arduino Core SDK version**
 - Version  2.5.2

mobizt, 2019-11-20:
It's not relate to the library because it's just software, the IP handles by another stack.

You need to test with ping or http client yourself.

closed 2019-11-20 by mobizt

Nursbg, 2019-11-20:
Now I test using : ESP8266WebServer.h : 

```
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server(80);

#define FIREBASE_HOST "xxxxxx.firebaseio.com"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
#define WIFI_SSID "xxxxxx"
#define WIFI_PASSWORD "xxxxxx"

IPAddress staticIP(192, 168, 2, 152);
IPAddress gateway(192, 168, 2, 1);
IPAddress subnet(255, 255, 255, 0);

FirebaseData firebaseData;
String path = "/test";

const int ledPin = 2;

void setup()
{
  pinMode(ledPin, OUTPUT);

  WiFi.config(staticIP, gateway, subnet);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  
  while (WiFi.status() != WL_CONNECTED)
  {
    digitalWrite(ledPin, 0);
    delay(200);
    digitalWrite(ledPin, 1);
    delay(200);
  }
  
  digitalWrite(ledPin, 1); //led off

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Firebase.setString(firebaseData, path, "tost"); // not work

  server.on("/test", blinkLed);
  server.begin();
  
}

void loop()
{
    digitalWrite(ledPin, 0); 
    server.handleClient();
}

void blinkLed() { // it is work
  server.send(200, "text/plain", "ok");
  digitalWrite(ledPin, 0);
  delay(200);
  digitalWrite(ledPin, 1);
  delay(200);
}

```
I test using http://192.168.2.152/test , it respones "ok", and led was blingking
But, Firebase.setString(firebaseData, path, "tost"); // it still not work.

mobizt, 2019-11-20:
You should test for internet not for local server.

mobizt, 2019-11-20:
Device can't access to the internet due to your network configuration problem.

Nursbg, 2019-11-20:
Thanks mobizt,
I 've tested using ping to internet, The result was false.
Any help related to this problem (static ip to connect to internet) ?

mobizt, 2019-11-20:
Look like you're using Belkin router.

Try reset or reboot it or change its ip to 192.168.1.1 or 192.168.0.1.

Nursbg, 2019-11-20:
Ok mobizt, thank you very much.

mobizt, 2019-11-20:
If you can change router ip, use ipconfig in Windows cmd to see the changes.

mobizt, 2019-11-20:
You can change the router IP address and disable DHCP see [this on step7](https://www.yourictmagazine.com/howtos/436-steps-to-configure-a-belkin-wireless-router-as-internet-gateway-for-your-home-or-office.html).