mobizt/Firebase-ESP8266 #72: Crash and restart when readstream is called

majorlennox, 2019-11-25:
I have a program that streams data from firebase. It runs for like 2 seconds and then crashes when readStream() is called. When I comment the readStream() part out it runs without issues.
Here's the code:
```
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "sxx"
#define FIREBASE_AUTH "xx"
#define WIFI_SSID "xx"
#define WIFI_PASSWORD "xx"
#define WIFI_LED D5
#define STARTUP_STATE 1
#define PATH_LAMP "/tripod/status"
#define PATH_CHECK "/tripod/check"
#define LAMPPIN D2
String returnData = "1";
String returnDataTemp = "1";
long lWaitMillis;
int stateint;
//Call to ESpressif SDK
extern "C" {
#include <user_interface.h>
}

FirebaseData firebaseData;
FirebaseData firebaseData2;
uint8_t mac[6] {0xA8, 0xD9, 0xB3, 0x0D, 0xAA, 0xAF};
void setup() {

  pinMode(LAMPPIN, OUTPUT);
  //Lampe Startmodus:
  String state = "0";
  if (STARTUP_STATE == 1) {
    turnOn();
    state = "1";
  }

  else if (STARTUP_STATE == 0) {
    turnOff();
    state = "0";
  }
  WiFi.persistent(false);
  Serial.begin(115200);

  Serial.println(WiFi.macAddress());
  pinMode(WIFI_LED, OUTPUT);
  IPAddress ip(192, 168, 0, 250);
  IPAddress gateway(192, 168, 0, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress dns(8, 8, 4, 4);
  // WiFi.config(ip, gateway, subnet);
  wifi_set_macaddr(0, const_cast<uint8*>(mac));
  Serial.println(WiFi.macAddress());

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    digitalWrite(WIFI_LED, LOW);
    delay(100);
    digitalWrite(WIFI_LED, HIGH);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  digitalWrite(WIFI_LED, HIGH);
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Gateway address: ");
  Serial.println(WiFi.gatewayIP().toString());
  Serial.print("AP subnet mask: ");
  Serial.println(WiFi.subnetMask());

  delay(1000);
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Serial.println("Debug");
  delay(1000);
  if (Firebase.setString(firebaseData, PATH_LAMP, "1"))
  {
    Serial.println("PASSED");
  }
  else {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  stateint = state.toInt();  // initial setup
  while (!Firebase.beginStream(firebaseData, PATH_LAMP))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  Firebase.setReadTimeout(firebaseData, 300);

}

void loop() {
  Serial.println("Debug1");
  int firebaseResult = firebaseGet(PATH_LAMP).toInt();
  Serial.println("Succ");
  if (firebaseResult != stateint) {
    if (firebaseResult == 1) {
      turnOn();
    }
    else if (firebaseResult == 0) {
      turnOff();
    }
  }
}
void turnOn() {
  digitalWrite(LAMPPIN, HIGH);
  Serial.println("ON");
  stateint = 1;
}
void turnOff() {
  digitalWrite(LAMPPIN, LOW);
  Serial.println("OFF");
  stateint = 0;
}
String firebaseGet(String s) {
  Serial.println("Reading");
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  if (firebaseData.streamAvailable())
  {
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    if (firebaseData.dataType() == "string") {
      returnDataTemp = firebaseData.stringData();
      if (returnDataTemp == "1" || returnDataTemp == "0") {
        returnData = returnDataTemp;
        Firebase.setString(firebaseData2, PATH_CHECK, "check");
      }
    }
    Serial.println("------------------------------------");
    Serial.println();
  }
  else{
    Serial.println("No data");
  }
  return (returnData);
}
```

Here's the decoded stack dump:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4000e1e0
EXCVADDR: 0x00000000

Decoding stack results
0x4021c238: BearSSL::WiFiClientSecure::read() at C:\Users\Finn\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 374
0x40206856: FirebaseESP8266::getServerResponse(FirebaseData&) at C:\Users\Finn\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2872
0x40212cf2: __yield() at C:\Users\Finn\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266\core_esp8266_main.cpp line 100
0x40226f06: std::basic_string  , std::allocator  >::~basic_string() at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 541
0x402078a0: FirebaseESP8266::getServerStreamResponse(FirebaseData&) at C:\Users\Finn\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3751
0x40201d41: FirebaseESP8266::reconnect(FirebaseData&) at C:\Users\Finn\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4457
0x40207948: FirebaseESP8266::readStream(FirebaseData&) at C:\Users\Finn\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2004
0x402015f1: firebaseGet(String) at C:\Users\Finn\Documents\SmartLight\SmartLightESP8266Back/SmartLightESP8266Back.ino line 140
0x40223040: strtol at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/strtol.c line 224
0x4021261c: String::copy(char const*, unsigned int) at C:\Users\Finn\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266\WString.cpp line 214
0x402018d8: loop() at C:\Users\Finn\Documents\SmartLight\SmartLightESP8266Back/SmartLightESP8266Back.ino line 118
0x40212cc9: esp_schedule() at C:\Users\Finn\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266\core_esp8266_main.cpp line 112
0x40212dd8: loop_wrapper() at C:\Users\Finn\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266\core_esp8266_main.cpp line 179
```

I use the newest version of all libraries.

mobizt, 2019-11-25:
Did you know you should not print to serial to often in loop?
Because mcu has no  idle time to do other tasks then wdt reset.

If you intend to do, use delay(0) or yield() in loop.

mobizt, 2019-11-25:
Here are all lines of code you need to comment it out.
```C++
...
Serial.println("Reading");
...
Serial.println("No data");
...
Serial.println("Debug1");
...
Serial.println("Succ");
```

majorlennox, 2019-11-25:
Yes, I added all the Serial prints to debug. The problem still occurs when I print nothing at all.

mobizt, 2019-11-25:
Nothing error for my test

majorlennox, 2019-11-25:
What version of the ESP8266 board library are you using for your tests?

majorlennox, 2019-11-25:
Commenting these lines out did not help for me..

mobizt, 2019-11-25:
Ok I found it crashes in latest ESP8266 v 2.6.1.
Now I debug it and I'll inform you the update soon.

majorlennox, 2019-11-25:
Alright, thank you in advance.

mobizt, 2019-11-25:
Now bug was fixed already.
You can update the library from Github version.

majorlennox, 2019-11-25:
Alright, I will try. How can I make the IDE use the Github version instead of the one in the library manager?

mobizt, 2019-11-25:
At the library's first GitHub page, select download zip file and in Arduino IDE, choose Sketch > Include Library > Add .ZIP library... and choose the downloaded zip file.

You may need to delete the old library from libraries folder.

majorlennox, 2019-11-25:
ITS WORKING
Thank you so much!!

closed 2019-11-25 by mobizt