mobizt/Firebase-ESP8266 #74: FATAL ERROR: BSSL stack overflow + WeMos D1 mini

winkimania, 2019-12-09:
Firebase.sendMessage(...) generates a FATAL ERROR: BSSL stack overflow when esp8266 Core SDK > 2.5.2. 
Core SDK 2.5.2 is working 

Example:
`  
  ...

  //if (Firebase.broadcastMessage(firebaseData1))
  //if (Firebase.sendTopic(firebaseData1))
  if (Firebase.sendMessage(firebaseData1, 0))
  {
      Serial.println("PASSED");
      Serial.println(firebaseData1.fcm.getSendResult());        
  }
  else
  {      
      Serial.println("FAILED");
      Serial.println("REASON: " + firebaseData1.errorReason());
  }

   ...
`

Result:

SDK:2.2.2-dev(5ab15d1)/Core:2.6.1=20601000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
scandone
Connecting to Wi-Fi..........scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with winkinet 1, channel 11
dhcp client start...
ip:192.168.178.22,mask:255.255.255.0,gw:192.168.178.1

Connected with IP: 192.168.178.22

------------------------------------
Send Firebase Cloud Messaging...
BSSL:_connectSSL: start connection
BSSL:CERT: 30 82 06 7d 30 82 05 65 a0 03 02 01 02 02 10 26 fd e5 dc 3f f8 f6 5a 08 00 00 00 00 1d 8a fe 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 42 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 1e 30 1c 06 03 55 04 0a 13 15 47 6f 6f 67 6c 65 20 54 72 75 73 74 20 53 65 72 76 69 63 65 73 31 13 30 11 06 03 55 04 03 13 0a 47 54 53 20 43 41 20 31 4f 31 30 1e 17 0d 31 39 31 31 30 35 30 37 35 32 35 37 5a 17 0d 32 30 30 31 32 38 30 37 35 32 35 37 5a 30 71 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 13 30 11 06 03 55 04 08 13 0a 43 61 6c 69 66 6f 72 6e 69 61 31 16 30 14 06 03 55 04 07 13 0d 4d 6f 75 6e 74 61 69 6e 20 56 69 65 77 31 13 30 11 06 03 55 04 0a 13 0a 47 6f 6f 67 6c 65 20 4c 4c 43 31 20 30 1e 06 03 55 04 03 13 17 65 64 67 65 63 65 72 74 2e 67 6f 6f 67 6c 65 61 70 69 
BSSL:CERT: 73 2e 63 6f 6d 30 59 30 13 06 07 2a 86 48 ce 3d 02 01 06 08 2a 86 48 ce 3d 03 01 07 03 42 00 04 8f f8 72 9b 93 c2 0b 60 8e c5 b2 b8 ef ce 62 2e 66 60 09 92 5b d2 7a c6 34 d7 c4 d9 c2 1e f0 73 91 ea 71 07 7b 1c be 25 77 c2 51 b8 34 e6 d8 1c 67 b6 31 42 df 13 69 71 be d3 13 bd 66 6b 5e 00 a3 82 04 09 30 82 04 05 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 07 80 30 13 06 03 55 1d 25 04 0c 30 0a 06 08 2b 06 01 05 05 07 03 01 30 0c 06 03 55 1d 13 01 01 ff 04 02 30 00 30 1d 06 03 55 1d 0e 04 16 04 14 94 c6 87 78 f5 1f 06 e6 9b a1 12 ee fa e6 20 bd 19 77 9a d2 30 1f 06 03 55 1d 23 04 18 30 16 80 14 98 d1 f8 6e 10 eb cf 9b ec 60 9f 18 90 1b a0 eb 7d 09 fd 2b 30 64 06 08 2b 06 01 05 05 07 01 01 04 58 30 56 30 27 06 08 2b 06 01 05 05 07 30 01 86 1b 68 74 74 70 3a 2f 2f 
BSSL:CERT: 6f 63 73 70 2e 70 6b 69 2e 67 6f 6f 67 2f 67 74 73 31 6f 31 30 2b 06 08 2b 06 01 05 05 07 30 02 86 1f 68 74 74 70 3a 2f 2f 70 6b 69 2e 67 6f 6f 67 2f 67 73 72 32 2f 47 54 53 31 4f 31 2e 63 72 74 30 82 01 cc 06 03 55 1d 11 04 82 01 c3 30 82 01 bf 82 17 65 64 67 65 63 65 72 74 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 16 61 6e 64 72 6f 69 64 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 17 62 69 67 74 61 62 6c 65 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 22 63 6c 6f 75 64 65 72 72 6f 72 72 65 70 6f 72 74 69 6e 67 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 19 63 6c 6f 75 64 74 72 61 63 65 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 18 63 6f 6e 74 61 69 6e 65 72 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 17 64 61 74 61 66 6c 6f 77 2e 
BSSL:CERT: 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 18 64 61 74 61 73 74 6f 72 65 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 12 66 63 6d 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 18 66 69 72 65 73 74 6f 72 65 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 16 6c 6f 67 67 69 6e 67 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 19 6d 6f 6e 69 74 6f 72 69 6e 67 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 15 70 75 62 73 75 62 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 1c 72 75 6e 74 69 6d 65 63 6f 6e 66 69 67 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 1d 73 65 72 76 69 63 65 63 6f 6e 74 72 6f 6c 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 16 73 70 61 6e 6e 65 72 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 82 1a 73 74 61 63 6b 64 72 69 76 
BSSL:CERT: 65 72 2e 67 6f 6f 67 6c 65 61 70 69 73 2e 63 6f 6d 30 21 06 03 55 1d 20 04 1a 30 18 30 08 06 06 67 81 0c 01 02 02 30 0c 06 0a 2b 06 01 04 01 d6 79 02 05 03 30 2f 06 03 55 1d 1f 04 28 30 26 30 24 a0 22 a0 20 86 1e 68 74 74 70 3a 2f 2f 63 72 6c 2e 70 6b 69 2e 67 6f 6f 67 2f 47 54 53 31 4f 31 2e 63 72 6c 30 82 01 04 06 0a 2b 06 01 04 01 d6 79 02 04 02 04 81 f5 04 81 f2 00 f0 00 76 00 b2 1e 05 cc 8b a2 cd 8a 20 4e 87 66 f9 2b b9 8a 25 20 67 6b da fa 70 e7 b2 49 53 2d ef 8b 90 5e 00 00 01 6e 3a c4 62 34 00 00 04 03 00 47 30 45 02 20 7c ec de 4a e3 a4 ad f1 ec 50 ed 15 31 1c 52 dc 36 2c 3f e4 4d 79 eb 86 bf d4 b6 7d 67 51 a4 e5 02 21 00 bc 55 38 da aa 4b ce 50 5a c8 36 ba b5 62 34 ff b6 d3 76 0e 69 a6 9f 64 bf c1 b9 f4 d8 ca 8a be 00 76 00 5e a7 73 f9 df 56 c0 e7 
BSSL:CERT: b5 36 48 7d d0 49 e0 32 7a 91 9a 0c 84 a1 12 12 84 18 75 96 81 71 45 58 00 00 01 6e 3a c4 62 5a 00 00 04 03 00 47 30 45 02 20 03 da 95 3e e6 c6 93 84 53 c4 da 07 78 db f8 e4 7d 4a 27 fc 5b 4a 31 b3 2a 81 d9 94 ca c4 6c e5 02 21 00 d6 1f 14 9b c6 c2 a9 9d a7 d4 e2 90 b6 e4 1c 59 81 36 56 7e 9d 49 cc fb bb 08 cb f8 09 30 a7 37 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 03 82 01 01 00 bf 61 43 ef 54 7b 59 da 36 cf ac ff a9 33 1d d9 f2 62 df 41 aa 15 98 d0 5d 9f 9a 68 38 e5 6d 32 74 75 30 84 a1 59 a3 3e db 74 8e e4 53 e8 43 25 76 98 7c 7a ab 46 9f 02 98 4f e9 1d 7c bc 00 49 2f 25 21 fc 45 d6 72 8c 2e af 4b 7b 90 5a b2 5d b1 82 9e d9 d1 99 46 52 62 7e b9 82 05 25 d2 06 67 fb e0 4b df 40 af cb a3 9b 78 a0 fe 0a f1 53 c4 04 b2 2f aa ee b6 4b d7 27 be a5 20 15 f2 
BSSL:CERT: 45 a7 2a a5 d7 87 0d 89 0f 47 31 c5 28 2e e6 99 b0 40 16 27 a6 36 37 f9 35 7d 51 9e 8b ae f5 15 65 42 5e 96 3b a3 23 e0 02 68 6a c6 80 44 57 c9 df a2 07 70 20 43 0f 91 bb a6 e8 08 33 50 ab b0 04 0d c9 37 9e fc c8 d4 14 5c 73 44 bd 71 c2 21 ec aa b7 e2 5c 42 3b 30 bf f6 07 4b 5e a4 df 5e eb 32 0c ee 79 57 a6 82 1d 73 83 54 94 91 3b 5d 2d c9 bd eb f7 55 dd bf 5e de 0e 82 31 17 f5 c3 5b 
FATAL ERROR: BSSL stack overflow

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: bearssl
sp: 3fff18ec end: 3fff196c offset: 0000
3fff18ec:  3ffe9ac4 00000046 00000004 deadbeef  
3fff18fc:  000000fe 00000000 00000000 00000000  
3fff190c:  00000000 00000000 00000000 3fff193c  
3fff191c:  00000004 00000090 00000004 3ffe9ac4  
3fff192c:  00003a98 00000000 3fff021c 4020df0e  
3fff193c:  3ffe897b 00000000 402302d0 4020df24  
3fff194c:  3fff3c81 3fff33b4 deadbeef 4020d265  
3fff195c:  deadbeef 00000000 3fff021c 40208ac4  

ctx: cont
[... 32 lines cut ...]
3ffffed0:  4025874e 3fff295c 3ffeea60 40203818  
3ffffee0:  3fff295c 000000d0 00000190 40100400  
3ffffef0:  3ffef914 40209d5c 3ffeea60 3ffe891d  
3fffff00:  3fff27bc 3fff295c 3ffeede0 40202648  
3fffff10:  3ffeea60 3ffee9f8 3fff2124 40201df2  
3fffff20:  3fff2418 3ffee9f8 00000000 00000000  
3fffff30:  3ffeea60 3ffee9f8 00000000 402027a2  
3fffff40:  3ffeee1c 3ffeef14 3ffe8936 3ffeefb8  
3fffff50:  00000000 3ffee9f8 3ffeee1c 3ffeefb8  
3fffff60:  00000000 3ffeee1c 3ffee9f8 40202820  
3fffff70:  3ffeee1c 3ffee9f8 3ffeef14 40201195  
3fffff80:  79646f00 00980000 80efeffe 6c746900  
3fffff90:  00230065 80efeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffeef78 4020db68  
3fffffb0:  feefeffe feefeffe 3ffe8520 401010c9  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld

mobizt, 2019-12-10:
Thank you for addressing this issue.

The issue due to bugs [#6811](https://github.com/esp8266/Arduino/issues/6742) in BearSSL in ESP8266 Arduino Core SDK version 2.6.1.

Please update the Library to latest version, 2.7.2 and also update the ESP8266 Arduino Core SDK to newer version (v 2.6.2).

noppakow1993, 2019-12-10:
Dear K Suwatchai and mobizt

Sorry to raise up again ....
I have already followed as your recommendation (to update your library and
update esp8266 to ver. 2.6.2)
but it still not working.
result is the same . could you please kindly advise for this ..
Thank you in advance.

PS : file.txt is already collected  in my SPIFFS .

BS,
Noppakow S.

[image: image.png]

mobizt, 2019-12-10:
@noppakow1993 

Actually the updates should work.

Please post your code.

closed 2019-12-11 by mobizt

noppakow1993, 2019-12-11:
Dear K Suwatchai and mobizt

please see my code per attachment .
Thank you so much.

Best Regard,
Noppakow S.

mobizt, 2019-12-12:
@noppakow1993

Are you kidding?

If you receive the Github Email, please go to Github and post. You can't attach file and reply to notification Email.